MichaIng/DietPi #151: DietPi-Software | Add MATE desktop installation option

Fourdee, 2015-12-23:
> http://fuzon.co.uk/phpbb/viewtopic.php?f=12&t=278&p=1178#p1176

xenfomation, 2016-05-17:
As a fan of mate, no complaints. Will manually test this week and report findings/level of effort

xenfomation, 2016-05-18:
Fresh DietPi install of v118 and command-line testing of installing MATE (complete):

`apt-get install mate-desktop-environment`

This failed due to non-core packages, such as DRM (digital rights management, etc).  I will install the core first, or:

`apt-get install mate-desktop-environment-core`

From there - I will post my results!

xenfomation, 2016-05-27:
Working on this, literally now, for addition to installation options.

xenfomation, 2016-05-29:
Update on pre-pull request
- Git clone of TESTING
- Added under Software Additional as it is not optimized
- Added MATE (full installation + extras) - 1.1GB of space
- Installs XServer/Xauth, Iceweasel, p7zip, and ALSA

Menu Screenshot:
[image]

Wrapping up testing before commit/pull request for @Fourdee and his review.

xenfomation, 2016-05-29:
Installation from raw v118 to MATE Desktop, IceWeasel, ALSA, etc good
[image]

Fourdee, 2016-06-04:
Additional implementation and testing for reference: https://github.com/Fourdee/DietPi/pull/364

Fourdee, 2016-06-08:
Documentation to finish: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=60#p2073

Fourdee, 2016-06-08:
Done.

closed 2016-06-08 by Fourdee