MichaIng/DietPi #152: mmc spi driver patch

atlury, 2015-12-25:
Can the following patches be applied for the next release of alpine linux for rpi.

https://github.com/raspberrypi/linux/issues/911
https://github.com/raspberrypi/linux/pull/1135

Error is with mmc_spi with the latest kernels. You always get error messages like this:
mmc1: error -110 whilst initialising SD card

This happens with sdcard is connected to SPI interface.

Note1: LC-Studio SD-Card Adapter was used.

Seems the latest suggested patch is:
https://github.com/msperl/linux-rpi/commit/aef50a19f335ae3d2eb43ac5496c7f03930b394f

Fourdee, 2015-12-26:
@atlury 
Hi, thanks for the information on this.

Ideally, if this patch is stable, it would gradually make its way to the master branch where I assume rpi-update would be used to apply it.
As this is not the case with the mmc spi patch, I'am not prepared to roll it out for the DietPi Rpi images, as this could cause more issues than it's worth.

I would recommend you compile the kernel with the required patch for your system, or, wait until this patch makes it to the master/stable branch.

Please reopen if required.

closed 2015-12-26 by Fourdee

atlury, 2015-12-30:
Thanks Fourdee!