MichaIng/DietPi #153: Odroid c1 ethernet/wifi connection fails

migueLoureiro, 2015-12-28:
Cannot connect to internet on odroid c1 (original c1, not c1+)
I have tried with both the ethernet cable and with a wifi dongle:
TPLINK TL-WN725N, link: http://www.tp-link.com/lk/products/details/cat-11_TL-WN725N.html
I can see my network, connect to it (in theory) but then the internet connection test always fails.
With the ethernet cable, it just doesnt connect.
Both the cable and wifi dongle work with the odroid c1 (linux distro and openelec) so the problem is not hardware.
I reported this here because i couldn't create an account on your foruns, i create an account but on login, it says:
"The specified username is currently inactive. If you have problems activating your account, please contact a board administrator."

Fourdee, 2015-12-28:
@migueLoureiro 
Hi, thanks for the report.

I'll need to get some info on your installation, setup, hardware. Could you please run the following:
```
dietpi-bugreport
```
Follow the onscreen instructions. As your system is offline, the bugreport will be created on the SD card. Once completed, please upload the bug report and link here.
The bug report will contain lots of files that should allow me to resolve this issue for you.

Yep apologies for the forums, I manually activate accounts due to spambot. Creating a git ticket is prefered, but if you can let me know the username you signed up with, I can activate your accout.

migueLoureiro, 2015-12-28:
Thanks for the quick reply! :)
My username on the forum is: *"lasy"*
I will generate the bug report and place it here.

Thanks for the support.

Best regards,
José Miguel Loureiro

Fourdee, 2015-12-28:
@migueLoureiro 
I've enabled your account on the forums. Just give us a shout when you have the bugreport.

Fourdee, 2016-01-13:
No response, please reopen if required.

closed 2016-01-13 by Fourdee