MichaIng/DietPi #155: Squeezebox server | squeezezeboxserver_init

Fourdee, 2016-01-02:
> http://fuzon.co.uk/phpbb/viewtopic.php?f=9&t=173#p1186
line 18: 1257 illegal instruction 
74da385b58ad-0

Fourdee, 2016-01-03:
Appears the 7.9 nightly builds have dropped support for ARMv6 architecture (RPi v1).
v7.8 supports ARMv6, can't be launched from cli, same error as https://github.com/Fourdee/DietPi/issues/123 :
```
The following modules failed to load: DBI EV XML::Parser::Expat HTML::Parser JSON::XS Digest::SHA1 YAML::XS Sub::Name
```
Works from /etc/init.d/logitechmediaserver service. Change to service based?

Fourdee, 2016-01-03:
7.8 /etc/init.d/logitechmediaserver:
Fails to start on ARMv6, fine on ARMv7. Seems the service is not the issue.

Both 7.8 & 7.9 ignores SLIMUSER= and always runs as user squeezeboxserver:
[image]
The following has no effect:
```
sed -i "/SLIMUSER=/c\SLIMUSER=root" /etc/default/logitechmediaserver
sed -i "/SLIMUSER=/c\SLIMUSER=root" /etc/init.d/logitechmediaserver

```

Fourdee, 2016-01-04:
Note to self:
http://forums.slimdevices.com/showthread.php?84680-Installation-failure-to-load-Perl-Modules

Fourdee, 2016-01-06:
Disabled in DietPi-Software for ARMv6, until I (or anyone else) can find a workaround.

masterxilef, 2016-01-11:
Seems like I got it working on a Raspberry Pi B 512MB, DietPi Jessie v105, using this guide as a reference:
http://forums.slimdevices.com/showthread.php?104515-Squeezebox-server-stopped-working-after-upgrade-to-Jessie-on-raspberrypi&p=834976&viewfull=1#post834976

But got 100% CPU usage, so after reading this:
https://github.com/Fourdee/DietPi/issues/65

So this seems to do the job:

wget http://dietpi.com/downloads/binaries/all/logitechmediaserver_7.9.0_all.deb
dpkg -i logitechmediaserver_7.9.0_all.deb
service logitechmediaserver stop
wget http://download1642.mediafire.com/p50pznj996dg/153ccfm16lb5ka2/logitechmediaserver_7.9.0_CPAN_5.20_armv6hf.tar.gz
tar xvzf logitechmediaserver_7.9.0_CPAN_5.20_armv6hf.tar.gz -C /
update-rc.d logitechmediaserver remove
rm /etc/init.d/logitechmediaserver
rm /etc/default/logitechmediaserver
cp /DietPi/dietpi/conf/squeezeboxserver_init /etc/squeezeboxserver_init
chmod +x /etc/squeezeboxserver_init

Plays mp3 fine, didn't try other codecs, nor in Wheezy.

Hope it helps.

Fourdee, 2016-01-11:
@masterxilef :+1: 
I will give this a spin tomorrow on my Zero. If all goes well, I'll update the installation code and enable for v106.
Thanks for sharing!

Fourdee, 2016-01-12:
@masterxilef You are a legend :+1: . Working perfectly!
I've updated and re-enabled the code. Will be available for install (again :)) in next release (v106)

masterxilef, 2016-01-12:
Glad I could help.
Did you tried on Wheezy?

Fourdee, 2016-01-13:
@masterxilef 
Yep, working on Wheezy :+1: :
[image]

closed 2016-01-13 by Fourdee