MichaIng/DietPi #156: Explain this

prov3it, 2016-01-02:
Hi,

My DietPI Installation (Debian Wheezy) went from version 104 to version 68. dietpi-backup doesnt even exist anymore.

I dont know how, but it happened.

prov3it, 2016-01-02:
I rebooted and its version 104 again. Before reboot the /Dietpi/dietpi/ folder contained:
boot,dietpi-apt-get_update, dietpi-boot_order, dietpi-config, dietpi-external_drive_setup, dietpi-logclear, dietpi-ramdisk, dietpi-software, dietpi-uninstall, finalise, login, conf, dietpi-banner, dietpi-bugreport, dietpi-cpu_set,	dietpi-funtime, dietpi-obtain_hw_model, dietpi-services, dietpi-survey, dietpi-update, func, misc.

After reboot it contained:
boot, dietpi-autostart, dietpi-bugreport, dietpi-cpuinfo, dietpi-external_drive_setup, dietpi-logclear, dietpi-ramdisk, dietpi-survey, finalise,conf, dietpi-backup, dietpi-cleaner, dietpi-cpu_info, dietpi-funtime, dietpi-morsecode, dietpi-ramlog, dietpi-sync, func, dietpi-affinity, dietpi-banner, dietpi-cloudshell, dietpi-cpu_set, dietpi-launcher, dietpi-nice, dietpi-services, dietpi-uninstall, login, dietpi-apt-get_update, dietpi-boot_order, dietpi-config, dietpi-cron, dietpi-letsencrypt, dietpi-obtain_hw_model, dietpi-software, dietpi-update, misc.

When i tried to run any of them before reboot it gave me the following error:
-bash: /DietPi/dietpi/dietpi-launcher: No such file or directory

Does anyone know what happened here?

Fourdee, 2016-01-02:
>>My DietPI Installation (Debian Wheezy) went from version 104 to version 68. dietpi-backup doesnt even exist anymore.
I dont know how, but it happened.

@prov3it Hi,
Unsure myself, i've never seen this behaviour. Have you modified any of the DietPi scripts, or, made customizations to any linux configuration files?

I need to see your system configuration/settings, please run the following to send a report
```
dietpi-bugreport
```
No need to fill out the text document, i only need the reference code so please reply with that.

Fourdee, 2016-01-13:
No response, please reopen if required.

closed 2016-01-13 by Fourdee