MichaIng/DietPi #161: Shairport Sync [feature request]

nickrttn, 2016-01-10:
Integration of https://github.com/mikebrady/shairport-sync would be awesome. The installation process for this isn't too bad to do manually but I'll take automation any day. 

It makes the RPi a great lightweight AirPlay streaming device. I'm running it headless, connected to my stereo, without `libsox` and it's fine to be honest. Haven't tried the libsox functionality because I'm on a B+ and it's supposed to eat a lot of CPU.

Fourdee, 2016-01-10:
@nickrttn 
Thanks for the request :+1: , looks interesting. I'll compile on my RPiv2 and take it for a spin. Hopefully we can make a .deb package for the installation to make it quicker.

Fourdee, 2016-01-10:
Looking good. Running on Orange Pi PC:
[image]

Will be available in next release.
Most likely a .deb install. Just need to check if RPi 1 (ARMv6) needs its own compile.

nickrttn, 2016-01-10:
Wow! that was :zap: fast! Thanks.

Fourdee, 2016-01-11:
@nickrttn 

Ok done some testing with soxr interpolation:
```
RPi v2 + soxr + resync_threshold=2250 (default) | Choppy, it can't keep up.
RPi v2 + soxr + resync_threshold=0 | Fine, but defeats the purpose of sync.
RPi v2 + soxr + resync_threshold = 3500 | Fine
```
So it looks like soxr is possible if you are willing to lapse the resync_threshold somewhat.

For the dietpi-software installation, I will leave interpolation on basic (disabling soxr) for the moment.

Fourdee, 2016-01-11:
Update, using dietpi-nice to set nice -20 for shairport-sync
```
RPi v2 + soxr + resync_threshold=2250 (default) + nice -20 (dietpi-nice) | Fine
```

Fourdee, 2016-01-11:
> http://fuzon.co.uk/phpbb/viewtopic.php?f=8&t=5&p=1221#p1221

Fourdee, 2016-01-11:
Completed. Available in next release (v106) from dietpi-software.

Hopefully get v106 release in a next few days.

closed 2016-01-11 by Fourdee

nickrttn, 2016-01-18:
Just wanted to let you know that I've been running it for a couple of days now, and it works perfectly. Thanks.

Fourdee, 2016-01-18:
@nickrttn Excellent, no worries, good to hear :+1: