MichaIng/DietPi #164: Automatic update during initial boot not skipping

EzTargetUK, 2016-01-13:
I'm not sure if this is me setting the options wrong, or a small logic error in the dietpi-software script.

In my dietpi.txt file I've set the following:
AUTO_DietpiSoftware_SkipUpdateRebootPrompt=1

When setting up a new install of the OS on a Raspberry Pi I keep getting the message saying that the DietPi has been updated to the latest version.

I've copied the code from line 3682 to the current test RPi, and the result is a 1, not the 0 that's on the line:

 # cat /DietPi/dietpi.txt | grep -ci -m1 '^AUTO_DietpiSoftware_SkipUpdateRebootPrompt=1'
1
 # cat /DietPi/dietpi.txt | grep -ci -m1 '^AUTO_DietpiSoftware_SkipUpdateRebootPrompt=0'
0

So I'm wondering if line 3682 in dietpi-software should be:

if (( $(cat /DietPi/dietpi.txt | grep -ci -m1 '^AUTO_DietpiSoftware_SkipUpdateRebootPrompt=1') == 1 )); then

Fourdee, 2016-01-13:
@EzTargetUK 
```
So I'm wondering if line 3682 in dietpi-software should be:

if (( $(cat /DietPi/dietpi.txt | grep -ci -m1 '^AUTO_DietpiSoftware_SkipUpdateRebootPrompt=1') == 1 )); then
```
There was a logic/ordering error in the dietpi-software script which prevented this flag from being read, before the if statement.

This has now been resolved: https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L3741-L3744

```if (( $(cat /DietPi/dietpi.txt | grep -ci -m1 '^AUTO_DietpiSoftware_SkipUpdateRebootPrompt=1') == 0 )); then``` If AUTO_DietpiSoftware_SkipUpdateRebootPrompt=1, the -ci result will be 1, which will skip that section of code (the prompt).

Although this issue is now resolved https://github.com/Fourdee/DietPi/issues/158 , you will need to use the updated v105 image. This is because the dietpi-software script does not change/update until the script is terminated (eg: reboot) as it runs in memory.

I'll get the Jessie image uploaded today and provide you with the URL.

Fourdee, 2016-01-13:
All DIetPi images have been updated to v106 which resolves the issue:
http://dietpi.com/downloads/images/

closed 2016-01-13 by Fourdee