MichaIng/DietPi #165: OrangePi-PC not work swap file

supercoollbob, 2016-01-13:
kswapd0 100%  cpu load.

root@DietPi02:/mnt# free
             total       used       free     shared    buffers     cached
Mem:       1026964     286724     740240      15476      20320     166268
-/+ buffers/cache:     100136     926828
Swap:      1023996          0    1023996

Fourdee, 2016-01-13:
Please send a bug report and reply with reference Code once completed:
```
dietpi-bugreport
```

supercoollbob, 2016-01-14:
40a5ef09e1ac-1

judging by the comments people are a problem present in all cores from Loboris. Need patch .

Fourdee, 2016-01-14:
Looks like an issue with the Kernel swap process:
```
root        36 17.9  0.0      0     0 ?        S    01:03 127:44 [kswapd0]
```
Which is strange because swapiness is set to 1 and you still have Ram available.

How big is your swapfile?
```
cat /etc/dphys-swapfile
```

Can you try disabling dphys swapfile:
```
/DietPi/dietpi/func/dietpi-set_dphys-swapfile 0
reboot
```

Fourdee, 2016-01-14:
```judging by the comments people are a problem present in all cores from Loboris. Need patch ```

I believe its more a case of lack of development and support for the H3 in general.

supercoollbob, 2016-01-14:
Bug report Sent. Thank you.
 Your reference code: 40a5ef09e1ac-2

KiB Mem:   1026964 total,   860824 used,   166140 free,     4932 buffers
KiB Swap:   524284 total,        0 used,   524284 free.   757688 cached Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
   36 root      20   0       0      0      0 R  99.6  0.0   0:20.12 kswapd0
 1296 root      20   0   30496   5648   3844 S  13.6  0.5   0:57.83 smbd

root@DietPi02:~# free
             total       used       free     shared    buffers     cached
Mem:       1026964     868580     158384       6124       4972     765072
-/+ buffers/cache:      98536     928428
Swap:       524284          0     524284

Fourdee, 2016-01-14:
Can you clear the memory cache:
```
free; echo 3 > /proc/sys/vm/drop_caches
```

And post free result:
```
free
```

To print just CPU usage of kswap:
```
ps -aux | grep '[k]swap' | awk '{print $3}'
```

supercoollbob, 2016-01-14:
root@DietPi02:~# free
             total       used       free     shared    buffers     cached
Mem:       1026964     495268     531696       6164       1172     370964
-/+ buffers/cache:     123132     903832
Swap:       524284          0     524284

root@DietPi02:~# ps -aux | grep '[k]swap' | awk '{print $3}'
5.8

supercoollbob, 2016-01-14:
It is a half-measure. need works swap.

Fourdee, 2016-01-14:
Looks like a possible issue with the 3.4.39-01-lobo kernel, or, unstable RAM. 
It appears Loboris hasn't updated the kernel for a while, possibly stopped maintaining it: http://loboris.eu/opi_kernel.tar.gz

I've been unable to get on the OPi forums for a few days, seems their webserver is overloaded, or China throttles the UK: http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=342

supercoollbob, 2016-01-14:
Use torbrowser, work fine.

Fourdee, 2016-07-23:
Marking as closed as we use the ARMbian kernel now.

closed 2016-07-23 by Fourdee