MichaIng/DietPi #168: DietPi-Software |  Add Seafile CE

supercoollbob, 2016-01-14:
Seafile e.g. Dropbox 
https://forum.seafile-server.org/t/tutorial-for-seafile-ce-nginx-on-arm/3023

Fourdee, 2016-01-14:
:+1:  Seafile brings back some memories :)

Zone22, 2016-05-09:
Posting to show my support for Seafile. Seems a little more robust OwnCloud. Not to mention client-side encryption.

Fourdee, 2016-05-10:
@Zone22 Yep, i agree. I'll mark this for v119. Once v118 is wrapped up and released, I'll take a look.

Zone22, 2016-05-10:
@Fourdee 
The default install is highly insecure for production environment or home use as noted by the developers thus the creation of an install script. I would recommend using or at least referencing when constructing your own script [_Auto install Seafile Server_](https://github.com/Seafile/seafile-server-installer) as it offer a quick and easy way to set up a production ready Seafile Server using MariaDB, Memcached and NGINX as a reverse proxy in under 5 minutes. 

**Take note** as reported by flames.
There is a issue is in the database setup portion of the script. The password parameter is not passed to the mysqladmin/mysql cli tools, so we get access denied on user/db/table creation. W/o correct db setup seafile/seahub services do not start. Result is, there is nothing to listen on for Nginx = Bad Gateway.

After setting up db manually, seafile started and works perfectly.

Fourdee, 2016-05-10:
@Zone22 
Thank you for the info, appreciate it :+1: 

> After setting up db manually, seafile started and works perfectly.

Excellent, i'll look at generating the db manually (we normally do for software with MySQL db's). Script : https://github.com/Fourdee/DietPi/blob/master/dietpi/func/create_mysql_db

Just need to get v118 wrapped up 1st, turned out to be a big one (yet again) lol :)

perlian, 2016-05-19:
Will Seafile on DietPi have SSL access ?

if yes:
Self signed certificate or generated via lets encrypt ?

Fourdee, 2016-05-19:
@perlian 

SSL, not sure yet, will get back to you.

> So i installed my last Seafile server with the debian script https://github.com/seafile/seafile-server-installer/blob/master/seafile_v5_debian

Thanks for the info :+1: , i'll take a look.

perlian, 2016-05-21:
This could also be helpful
describes installing Seafile on ARM platforms
Did it yesterday on Odroid XU4 and it works

[https://forum.seafile.de/t/tutorial-for-seafile-ce-nginx-dynamic-dns-on-arm/3023](url)

Zone22, 2016-07-07:
Slight poke. 
Soon I'll be testing
https://github.com/seafile/seafile-server-installer/issues/81

Fourdee, 2016-07-08:
@Zone22 

> Slight poke. 

Poke away :+1: 

I've been through my notes, this is a big one to implement as I'am basically having to rewrite their automated installation script line by line, then convert it to DietPi.
I'll add this to v123, but it may be nudged to v124 depending on how good my notes are lol :)

Fourdee, 2016-07-14:
Sorry guys, going to have to move this one out of v123 for now.  I just havn't had any quality time to sit down, get in the zone and implement this without RL distractions. Theres a lot of code to siv through and I want to do it justice.

Zone22, 2016-07-31:
so what does it mean that this ticket has moved to prototyping?

Fourdee, 2016-08-01:
@Zone22 

https://github.com/Fourdee/DietPi/milestone/11
> For all tickets that are: 
- Open discussions. 
- Partially started and need work completing. 
- Not assigned to any version milestone.

At the moment, its in progress, but on hold.

Zone22, 2016-08-09:
What's making this difficult to work on either from a time or technical aspect?

Fourdee, 2016-08-09:
> What's making this difficult to work on either from a time or technical aspect?

Both real life distractions and time at the moment. 
As I explained previously, I'd need to go through their entire sourcecode line by line for their current automated installation. As this is Nginx only, i'd also need to implement support Lighttpd and Apache2 installations for our users.
Not only is this a big task, we already have Owncloud and Pydio available in DietPi. Another "variant" is low priority over the rest of DietPi.

For me, at this moment, and possibly for some time in the foreseeable future, its not a task I can not dedicate to. I'll let you know if that changes. Either that, or we get more users who can contribute to DietPi :+1:

Zone22, 2016-09-01:
I'll continue to make crack at it you can leave me what you have done thus far...
Really I see the script is something maybe to be used beyond Seafile CE. It could be used as a standard set up server applications behind a reverse proxy with secure settings per application. This would obviously grow as we have time to incorporate best practice and server applications.

While I can work on puzzling out I think it will be worth your time to create a guide on how to add software/or customization so that others can learn contribute. do you think it's worthwhile for a ticket to be created?

Fourdee, 2016-09-01:
@Zone22 

> I'll continue to make crack at it you can leave me what you have done thus far...
Really I see the script is something maybe to be used beyond Seafile CE. It could be used as a standard set up server applications behind a reverse proxy with secure settings per application. This would obviously grow as we have time to incorporate best practice and server applications.

I have no idea what state my notes are in. So it would probably be best to start from scratch, but heres my notes:
```
#Seafile CE -----------------------------------------------------------------------
# https://github.com/seafile/seafile-server-installer/blob/master/seafile_v5_debian

Requires L*MP stack

# - postfix
debconf-set-selections << EOF
postfix postfix/root_address    string
postfix postfix/rfc1035_violation       boolean false
postfix postfix/mydomain_warning        boolean
postfix postfix/mynetworks      string  127.0.0.0/8 [::ffff:127.0.0.0]/104 [::1]/128
postfix postfix/mailname        string  $(hostname -f)
postfix postfix/tlsmgr_upgrade_warning  boolean
postfix postfix/recipient_delim string  +
postfix postfix/main_mailer_type        select  Internet Site
postfix postfix/destinations    string  $(hostname -f), localhost.$(hostname -d)
postfix postfix/retry_upgrade_warning   boolean
# Install postfix despite an unsupported kernel?
postfix postfix/kernel_version_warning  boolean
postfix postfix/not_configured  error
postfix postfix/sqlite_warning  boolean
postfix postfix/mailbox_limit   string  0
postfix postfix/relayhost       string
postfix postfix/procmail        boolean false
postfix postfix/bad_recipient_delimiter error
postfix postfix/protocols       select  all
postfix postfix/chattr  boolean false
EOF
apt-get install postfix -y
dpkg-reconfigure postfix

# - Seafile prereqs
apt-get install -y python-setuptools python-simplejson python-imaging python-mysqldb memcached python-memcache pwgen curl openssl
# + Pro
apt-get install -y python3-uno libpython2.7 openjdk-7-jre libreoffice libreoffice-script-provider-python ttf-wqy-microhei ttf-wqy-zenhei xfonts-wqy poppler-utils python-urllib3

# - Seafile services
cat << _EOF_ > /etc/systemd/system/seafile.service
[Unit]
Description=Seafile Server
After=network.target dietpi-service mysql.service
[Service]
Type=oneshot
ExecStart=/opt/seafile/seafile-server/seafile.sh start
ExecStop=/opt/seafile/seafile-server/seafile.sh stop
RemainAfterExit=yes
User=root
Group=nogroup
[Install]
WantedBy=multi-user.target
_EOF_
systemctl enable seafile
cat << _EOF_ > /etc/systemd/system/seahub.service
[Unit]
Description=Seafile Seahub
After=network.target dietpi-service seafile.service
[Service]
ExecStart=/opt/seafile/seafile-server/seahub.sh start-fastcgi 8000
ExecStop=/opt/seafile/seafile-server/seahub.sh stop
User=root
Group=nogroup
Type=oneshot
RemainAfterExit=yes
[... 79 lines cut ...]
CLOUD_MODE                          = False
FILE_PREVIEW_MAX_SIZE               = 30 * 1024 * 1024
SESSION_COOKIE_AGE                  = 60 * 60 * 24 * 7 * 2
SESSION_SAVE_EVERY_REQUEST          = False
SESSION_EXPIRE_AT_BROWSER_CLOSE     = False
FILE_SERVER_ROOT                    = 'https://${HOSTNAME}/seafhttp'
REPO_PASSWORD_MIN_LENGTH            = 8
USER_PASSWORD_MIN_LENGTH            = 6
USER_PASSWORD_STRENGTH_LEVEL        = 3
USER_STRONG_PASSWORD_REQUIRED       = True
ENABLE_MAKE_GROUP_PUBLIC            = False
ENABLE_THUMBNAIL                    = True
THUMBNAIL_ROOT                      = '/opt/seafile/seafile-server/seahub-data/thumbnail/thumb/'
REPO_PASSWORD_MIN_LENGTH            = 8
USER_PASSWORD_MIN_LENGTH            = 8
USER_PASSWORD_STRENGTH_LEVEL        = 3
USER_STRONG_PASSWORD_REQUIRED       = True
_EOF_

#Seafile CE -----------------------------------------------------------------------
```

> While I can work on puzzling out I think it will be worth your time to create a guide on how to add software/or customization so that others can learn contribute. Do you think it's worthwhile for a ticket to be created?

Excellent idea, i'll sort that out :+1:

closed 2017-12-30 by Fourdee