MichaIng/DietPi #169: DietPi-Software |  Add FreePBX&Asterisk

supercoollbob, 2016-01-14:
http://www.raspberry-asterisk.org/
This combination will greatly enhance the attractiveness of the distribution DietPi.

Fourdee, 2016-01-14:
> http://www.asterisk.org/get-started
 Asterisk turns an ordinary computer into a communications server.

Looks interesting :+1: 

I'd imagine this would only be used in a business/commercial environment, instead of personal use. I'll look into it when I can, but it will have to be low priority for the moment.

ivanobom, 2017-07-29:
I run Asterisk at home as my personal PBX. Using an ATA adapter I can handle my analog PSTN line and also I connect to any other VoIP technology like Google Voice and SIP carriers. It'd be great porting Asterisk to an optimized version on DitePi!

WolfganP, 2017-07-29:
@ivanobom sounds like a nice setup. Care to point to more details on your setup? 
Which ATA adapter are you using? (most I saw were very expensive indeed, but it seems some went down in price recently ie https://www.lifewire.com/top-phone-adapters-3426340)

ivanobom, 2017-08-01:
@WolfganP I use a Grandstream HT503 (https://www.amazon.com/GrandStream-HT503-1-FXS-Analog-Telephone/dp/B002H29TGA/ref=sr_1_1?ie=UTF8&qid=1501603219&sr=8-1&keywords=grandstream+ht503) to be an interface between my analog PSTN phone line and convert it into SIP and register on my Asterisk running on my PI device (I've successfully tested on RaspberryPi 2, OrangePi PC and Odroid-XU4). 

I'm using Debian Stretch (9.1) and Installed the Debian's default Asterisk version (13.14). Also, I had to compile G729 codec (instructions from http://www.adeleda.com/compiling-g729-codecs-for-arm-on-raspberry-pi-b-with-bcg729-en.html however I used the most current version of BCG729 and Asterisk-G729 packages). 

As a plus, I'm also using **googlecontacts** plugin (https://github.com/PhilRW/gcontacts-asterisk/blob/master/googlecontacts.py) to populate a local database with my google contacts in order to update the incoming callerID with its name (if the caller is in my contacts list). 

About google voice, it's a pretty simple configuration on Asterisk (motif and xmpp) that add it as another VoIP trunk (just like SIP) and I can send and receive calls thru my Google Voice phone number.

Fourdee, 2017-12-30:
General Cleanup:
Marking all software requests as closed. Once interest peaks, we'll reopen once work starts.

closed 2017-12-30 by Fourdee

kodorevi8ulis, 2018-01-16:
i want that too :)
i also have a ht503 with raspbx (pi 1 b+) 
(use for home,  to avoid "spamm" calls and others)

the problem is that you have 2 options
either use the image (and you must check if something changes permissions of files etc -it's easy to "break" the asterisk/freepbx service) 
either use the step-by-step instructions and install manually the asterisk / freepbx / others which it's a long procedure and it takes a lot of time (at least in pi 1 / pi zero)

@ivanobom you can also use (in freepbx gui) the superfecta module for googlecontacs and other things

cakristof, 2021-11-08:
> 
> 
> > http://www.asterisk.org/get-started
> > Asterisk turns an ordinary computer into a communications server.
> 
> Looks interesting 👍
> 
> I'd imagine this would only be used in a business/commercial environment, instead of personal use. I'll look into it when I can, but it will have to be low priority for the moment.

Virtually everyone who runs FreePBX on a Pi / SBC is a home user.  Business and commercial users use appliances, servers, or SaS products.

I've used FreePBX to power my home landlines for many years.  It's s huge cost saving over paying $25-30 a month for a "real" landline.

Adding FreePBX shouldn't be too difficult as you can copy most of what you need from RasPBX: http://www.raspberry-asterisk.org/