MichaIng/DietPi #170: Waiting for DietPi-RAMDISK to finish copying DietPi to RAM

weakiwi, 2016-01-15:
I login the dietpi by ssh service and it show this reaptedlly.
![此处输入图片的描述][1]
If I Ctrl+C to break out and execute a shell command,which will lead to Read-only filesystem.
![此处输入图片的描述][2]

  [1]: http://ww3.sinaimg.cn/mw1024/68566be7gw1f00fxou0aoj20nd0enakv.jpg
  [2]: http://ww3.sinaimg.cn/mw1024/68566be7gw1f00fybvdufj20nd0endmk.jpg

Fourdee, 2016-01-15:
@weakiwi 
Hi, is this on a fresh install of DietPi? 

Also, please provide the following:
- PSU model/brand/voltage/amp (eg: Official RPi 5v/2a)
- SD card model/brand/size (eg: Sandisk Extreme 16gb uhs3)

weakiwi, 2016-01-16:
5v/1a normal mobile phone charger
Team 16gb sdhc
nothing

Fourdee, 2016-01-16:
@weakiwi 
Mobile phone chargers are not recommended. I've owned a few 5v/2a from Samsung which dont provide a stable voltage.

I've never heard or tested a Team SD card, If you have other SD cards (eg: Sandisk), please try one of those.

It appears the issue is caused by an IO error on the fileystem, hence why root FS is falling back to read only as a failsafe.
I would suggest you try rewriting the DietPi image and use alternative PSU/SD card combination where possible.

Fourdee, 2016-01-26:
Assumed resolved. Please reopen if required.

closed 2016-01-26 by Fourdee