MichaIng/DietPi #174: DietPi-Automation | AUTO_Locale AUTO_Timezone

Fourdee, 2016-01-17:
>b827ebe82d26-0

>AUTO_Locale=de_DE
Timezone and Keyboard Setting work.

>perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "de_DE"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
/usr/bin/locale: Cannot set LC_CTYPE to default locale: No such file or directory
/usr/bin/locale: Cannot set LC_MESSAGES to default locale: No such file or directory
/usr/bin/locale: Cannot set LC_ALL to default locale: No such file or directory

>FIX:
Set Language via diet-config manually

Fourdee, 2016-04-18:
As per https://github.com/Fourdee/DietPi/issues/291#issuecomment-211495754:
Ensure ```en_GB.UTF-8``` is not removed when automation user locale is set.

xenfomation, 2016-05-17:
I think it may be worth adding a locale check: enforcing en_GB.UTF-8 for scripts.

- check current locale; save in variable original_locale
- set locale to GB utf-8
- set locale back to user preferred

k-plan, 2016-08-03:
@Fourdee 

http://dietpi.com/phpbb/viewtopic.php?f=11&t=566

Fourdee, 2016-08-03:
@k-plan  :+1: 

Thanks. I'll take a look, working on automation at the moment.

Fourdee, 2016-08-05:
Locale is fixed:

Starting from en_GB:
```
AUTOINSTALL_LANGUAGE=en_US;sed -i "/$AUTOINSTALL_LANGUAGE.UTF-8/c $AUTOINSTALL_LANGUAGE.UTF-8 UTF-8" /etc/locale.gen;echo -e "LANG=$AUTOINSTALL_LANGUAGE.UTF-8" > /etc/default/locale;locale-gen;localectl set-locale LANG="$AUTOINSTALL_LANGUAGE.UTF-8"
```

After reboot:
```
root@DietPi:~# locale
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=
```

Also updated dietpi.txt docs to explain ```AUTO_Locale=``` will only support 1 entry.

Fourdee, 2016-08-05:
Fixed for v127. Requires v127 image which i'll create prior to v127 release.

closed 2016-08-05 by Fourdee