MichaIng/DietPi #176: DietPi-Software | Add rTorrent

Fourdee, 2016-01-20:
> any chance rtorrent will be added ?

Ardakilic, 2016-01-27:
I don't know how this will be, but if there would be an option to to choose version I'd like to, because of this:

https://github.com/rakshasa/rtorrent/issues/191

For lower models such as Rpi1 Model B/B+ etc, 0.9.3/0.13.3 will be way less resource-consuming.

Fourdee, 2016-01-27:
@Ardakilic Thanks for the info. Will look into this when work begins on rTorrent.

Fourdee, 2016-08-12:
0.9.3 isn't available as a release, we'll try 0.9.4 https://github.com/rakshasa/rtorrent/releases

Ardakilic, 2016-08-12:
@Fourdee Try these:

```
http://pkgs.fedoraproject.org/repo/pkgs/libtorrent/libtorrent-0.13.3.tar.gz/e94f6c590bb02aaf4d58618f738a85f2/libtorrent-0.13.3.tar.gz
http://pkgs.fedoraproject.org/repo/pkgs/rtorrent/rtorrent-0.9.3.tar.gz/0bf2f262faa8c8c8d3b11ce286ea2bf2/rtorrent-0.9.3.tar.gz
```

Old libtorrent website was wiped so the older releases are unavailable at the moment. But these mirror works as it should and hashes match.

Fourdee, 2016-08-13:
@Ardakilic 
rtorrent packages available in apt-get is version 0.9.2. So I'll start with that 1st.

Is there any major improvements/fixes in 0.9.3 that make it worth compiling?

Ardakilic, 2016-08-13:
@Fourdee Well I always compile because of xmlrpc-c support which is important to make it connect to other applications to manage it. Most how-to guides cover it.

Before compiling, I first install dependencies:

```
apt-get install autoconf build-essential ca-certificates comerr-dev libcloog-ppl-dev libcppunit-dev libcurl3 libcurl4-openssl-dev libncurses5-dev ncurses-base ncurses-term libterm-readline-gnu-perl libsigc++-2.0-dev libssl-dev libtool libxml2-dev ntp openssl patch libperl-dev php7.0-fpm php7.0-cli php7.0-curl php7.0-geoip php7.0-mcrypt php7.0-xmlrpc pkg-config python-scgi dtach ssl-cert subversion zlib1g-dev pkg-config unzip irssi curl cfv unrar nginx
```

Then  then compile xmlrpc-c (git master or superstable release):
https://sourceforge.net/projects/xmlrpc-c

```
cd xmlrpc-c
./configure --prefix=/usr --enable-libxml2-backend --disable-libwww-client --disable-wininet-client --disable-abyss-server --disable-cgi-server
make
sudo make install
```

Then while compiling rtorrent, I add `--with-xmlrpc-c` flag to `./configure` line.

After that I install [ruTorrent](https://github.com/Novik/ruTorrent) WebUI (php, addons such as xmlrpc-c and nginx is dependency of it) but it's another subject I think.

I know that 0.9.4 is buggy and banned on various websites (trackers) because of wrong announces. So please ignore that release for the time being.

Well, if you add rtorrent 0.9.2/0.13.2 directly from the repositories, It will only be usable from cli, but people rarely (almost never) use the cli. People wouldn't be able to connect it with 3rd party clients including webui (which most people use) to manage.

So I'd suggest you to compile.

Fourdee, 2016-08-13:
@Ardakilic 

Thanks. :+1: 

> xmlrpc-c support which is important to make it connect to other applications and WebUI to manage it. Most how-to guides cover it.

I have rtorrent 0.9.2 from apt, and rutorrent 3.7 running well on lighttpd:
[image]

So i'd assume the apt rtorrent package has been compiled with xmlrpc-c?

> If required, I can give you my full nginx/apache2 setup step by step if it'd ease your process.

I'd appreciate it :+1: 

> So I'd suggest you to compile.

Yep, just going to finish off draft installation code, then i'll give it a whirl.

Ardakilic, 2016-08-13:
Well last time I checked it wasn't. Or maybe I did something wrong and rutorrent doesn't need xmlrpc-c, I don't know to be honest. That was the main the first reason I'm always compiling. For the time being for most people it's usable then :+1: 

Also, just as a double check, if you have an android device, could you please try to install https://github.com/erickok/transdroid apk file and test if you can connect to the rtorrent with it ? You just set a new server, type rtorrent and credentials, that's it. If you can see debian iso there, at this stage, everything is good to go :tada:

Fourdee, 2016-08-13:
>  That was the main reason I'm always compiling.

Yep, i'am all for compiling. But, if we compile, we have to compile and support for:
- ARM64
- AMD64
- ARMv6
- ARMv7+ (optional, but we generally do it)

So if we can avoid having to do it, it saves alot of time lol. I believe 0.9.2 is fine for now, as our target was 0.9.3.

> Also, just as a double check, if you have an android device, could you please try to install https://github.com/erickok/transdroid apk file and test if you can connect to the rtorrent with it ? You [...]

Ah, i see. So if i'am correct, the web configuration is needed to:
- Create a username/password for ip/rutorrent web login
- Allow for direct access to rtorrent cli through port 80
?

> nginx config

Please, so I can get a test system running with the installation we are after.

Fourdee, 2016-08-13:
User/Pw for webservers:
:u55b6: Lighttpd
:u5272: apache2
:u5272: nginx

Ardakilic, 2016-08-13:
Hello,

This is my nginx config:

```
server {
  listen 80;
  server_name localhost my.rutorrent.com;

  root /srv/rutorrent;
  index index.html index.htm index.php;

  auth_basic "Restricted";
  auth_basic_user_file /etc/nginx/rutorrent-htpasswd; #http://www.htaccesstools.com/htpasswd-generator/ #because I don't want to install apache tools for this.

  location / {
    try_files $uri $uri/ =404;
  }

  location /RPC2 {
    include scgi_params;
    #scgi_pass unix:/home/user/.rtorrent-scgi.sock;
    scgi_pass 127.0.0.1:5000;
  }

  location ~ \.php$ {
    try_files $uri =404;
    fastcgi_split_path_info ^(.+\.php)(/.+)$;
    fastcgi_pass unix:/var/run/php/php7.0-fpm.sock; #change for php5 accordingly.
    fastcgi_index index.php;
    fastcgi_param SCRIPT_FILENAME $document_root$fastcgi_script_name;
    include fastcgi_params;
  }
}
````

I also have  `system.umask.set = 002` on my .rtorrent.rc for other users in same group to interact properly with the files.

Fourdee, 2016-08-14:
@Ardakilic 
Many thanks for the config and info, appricate it.

Having a little issue with it at the moment, heres the Nginx specific configuration code, based on yours:
```
			# - Nginx
			elif (( ${aSOFTWARE_INSTALL_STATE[85]} >= 1 )); then

				cat << _EOF_ > /etc/nginx/sites-available/rutorrent
server {
    listen 80;
    root /var/www/rutorrent;

    error_log /var/log/nginx/rutorrent_error.log;
    access_log /var/log/nginx/rutorrent_access.log;

    auth_basic "Restricted";
    auth_basic_user_file /etc/nginx/rutorrent-htpasswd;

    location / {
        try_files $uri $uri/ =404;
    }

    location /RPC2 {
        include scgi_params;
        scgi_pass 127.0.0.1:5000;
    }

    location ~ \.php$ {
        try_files $uri =404;
        fastcgi_split_path_info ^(.+\.php)(/.+)$;
        fastcgi_pass unix:///var/run/php5-fpm.sock;
        include fastcgi_params;
        fastcgi_param SCRIPT_FILENAME $request_filename;
        fastcgi_read_timeout 300;
    }
}
_EOF_

				ln -sf /etc/nginx/sites-available/rutorrent /etc/nginx/sites-enabled/rutorrent

				#htpasswd -cb /etc/nginx/.htpasswd root dietpi
				echo "root:$(openssl passwd -crypt dietpi)" > /etc/nginx/rutorrent-htpasswd

			fi
```

Log:
```
root@DietPi:~# cat /var/log/nginx/error.log
2016/08/14 11:02:00 [error] 930#0: *2 FastCGI sent in stderr: "PHP message: PHP Warning:  fopen(/var/www/rutorrent/share/settings/uisettings.json): failed to open stream: No such file or directory in /var/www/rutorrent/php/getsettings.php on line 7" while reading response header from upstream, client: 192.168.0.14, server: dietpi, request: "POST /rutorrent/php/getsettings.php HTTP/1.1", upstream: "fastcgi://unix:/run/php5-fpm.sock:", host: "192.168.0.112", referrer: "http://192.168.0.112/rutorrent/"
2016/08/14 11:09:44 [emerg] 3903#0: invalid number of arguments in "try_files" directive in /etc/nginx/sites-enabled/rutorrent:21
2016/08/14 11:10:49 [emerg] 4525#0: invalid number of arguments in "try_files" directive in /etc/nginx/sites-enabled/rutorrent:21
2016/08/14 11:12:06 [emerg] 5115#0: invalid number of arguments in "try_files" directive in /etc/nginx/sites-enabled/rutorrent:21
```

Fourdee, 2016-08-14:
Fixed ```*2 FastCGI sent in stderr``` by adding document_root:
```
fastcgi_param SCRIPT_FILENAME $document_root$fastcgi_script_name;
```

Just this one
```
root@DietPi:~# cat /var/log/nginx/error.log
0
2016/08/14 11:24:37 [emerg] 9985#0: invalid number of arguments in "try_files" directive in /etc/nginx/sites-enabled/rutorrent:21
```

Ardakilic, 2016-08-14:
@Fourdee Hmm maybe your script identifies $ in the script as variable. How about escaping them ?

Could you please dump the /etc/nginx/sites-enabled/rutorrent file for me ?

Meanwhile, try this:

```
location / {
    try_files $uri $uri/ /index.html;
  }
```

Fourdee, 2016-08-14:
> Hmm maybe your script identifies $ in the script as variabl

Yep, I completely forgot to escape those entries lol

Fourdee, 2016-08-16:
@Ardakilic 
I can not get this to work with Nginx. No password prompt.

Simple default config ```/etc/nginx/sites-available/default
```, following instructions from https://www.digitalocean.com/community/tutorials/how-to-set-up-password-authentication-with-nginx-on-ubuntu-14-04 and https://www.howtoforge.com/tutorial/basic-http-authentication-with-nginx/ and http://kbeezie.com/protecting-folders-with-nginx/:
```
#D I E T - P I
# /etc/nginx/sites-available/default

server {
	listen 80 default_server;

	root /var/www;
	index index.php index.html index.htm;

	server_name "$hostname";

	location / {
		try_files $uri $uri/ =404;
	}

	#RUTORRENT
	location /rutorrent {
		auth_basic "Restricted Content";
		auth_basic_user_file /etc/nginx/.htpasswd;
	}

	location /RPC2 {
		include scgi_params;
		scgi_pass 127.0.0.1:5000;
	}
	#RUTORRENT

	error_page 404 /404.html;
	error_page 500 502 503 504 /50x.html;
	location = /50x.html {
		root /var/www;
	}

	location ~ \.php(?:$|/) {
		fastcgi_split_path_info ^(.+\.php)(/.+)$;
		fastcgi_param PATH_INFO $fastcgi_path_info;
		fastcgi_pass unix:/run/php5-fpm.sock;
		fastcgi_index index.php;
		fastcgi_param SCRIPT_FILENAME $document_root$fastcgi_script_name;
		include fastcgi_params;
	}

	include /etc/nginx/sites-dietpi/*.config;

}
```

Ardakilic, 2016-08-16:
@Fourdee Could you please dump the output of "sudo nginx -t" ?

So it works and no password prompt given, or does it not work at all ?

Fourdee, 2016-08-16:
@Ardakilic No password prompt, rutorrent web works fine, not tested android client yet

Ardakilic, 2016-08-16:
Does `sudo nginx -t` give any errors ?

What's the owner and group of the /etc/nginx/.htpasswd file ?

Assuming your nginx user is `www-data` , could you please type and see if can be accessed by web user?

`sudo -u www-data cat /etc/nginx/.htpasswd`

Fourdee, 2016-08-16:
```
root@DietPi:~# sudo nginx -t
nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
nginx: configuration file /etc/nginx/nginx.conf test is successful
```

```
root@DietPi:~# ls -lha /etc/nginx/.htpasswd
-rw-r--r-- 1 root root 43 Aug 16 07:26 /etc/nginx/.htpasswd
```

```
root@DietPi:~# sudo -u www-data cat /etc/nginx/.htpasswd
root:$apr1$WjCblfYg$Ib1xkLlAIjiWcM99XYZl2.
```

===

Fixed it, replaced ```htpasswd -cb /etc/nginx/.htpasswd root dietpi``` with your method:
```
echo "root:$(openssl passwd -crypt dietpi)" > /etc/nginx/.htpasswd
```

Ardakilic, 2016-08-16:
@Fourdee  oh, good to know :) to be honest It's been a while since I've made a htpasswd file with htpasswd command. I use openssl command or that online site depending on my laziness 😄

Fourdee, 2016-08-16:
@Ardakilic 
Transdroid working with rtorrent :+1:

Fourdee, 2016-08-16:
Ok, i'll wrap up the code for Nginx and send the commit. You can then test the installation if you like.

Ardakilic, 2016-08-16:
@Fourdee Awesome! Great to know it's working! 🎉 

Of course! I can test on ODROID-XU4 and Raspberry Pi 2. Just let me know when to download.

Fourdee, 2016-08-16:
@Ardakilic 
Ok, Nginx code is in there, simply follow this guide to install from testing branch: https://github.com/Fourdee/DietPi/blob/master/TESTING-BRANCH.md

Have 2 issues outstanding:
```
[16.08.2016 13:35:48] rss: Some functionality will be unavailable. Webserver user can't access external program (curl).
[16.08.2016 13:35:50] Unpack plugin: rTorrent user can't access 'unrar' program.
```
I'll look at Apache2 implementation and testing.

Ardakilic, 2016-08-16:
@Fourdee just install `curl` and `unrar` from repository and restart services. Both should be fixed.

Ok then, I'm currently at work. I'll try to test this evening 👍

Fourdee, 2016-08-16:
@Ardakilic 

Strange, curl is already installed, i'll take a look.
```
root@DietPi:~# apt-get install curl unrar
Reading package lists... Done
Building dependency tree
Reading state information... Done
curl is already the newest version.
The following NEW packages will be installed:
  unrar
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 120 kB of archives.
After this operation, 327 kB of additional disk space will be used.
```

Ardakilic, 2016-08-16:
Try installing php5-curl or php7.0-curl then depending your PHP version.

Fourdee, 2016-08-16:
Raspbian:
```
root@DietPi:~# apt-get install screen mediainfo unrar php5-curl
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package unrar is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'unrar' has no installation candidate
```

Ardakilic, 2016-08-16:
Hmm you may need to get pre-compiled deb file or get the source and compile yourself then

```
echo "deb-src http://mirrordirector.raspbian.org/raspbian/ jessie main contrib non-free rpi" | sudo tee -a /etc/apt/sources.list

sudo apt-get update
sudo apt-get build-dep unrar-nonfree
sudo apt-get source -b unrar-nonfree
sudo dpkg -i unrar*.deb
```

But the better way is to download from [rarlab](http://www.rarlab.com/rar_add.htm) directly:

```
wget http://www.rarlab.com/rar/unrar-5.3.7-arm.gz
gunzip unrar-5.3.7-arm.gz #or gzip -d unrar-5.3.7-arm.gz
chmod +x unrar
mv unrar /usr/local/bin
which unrar #to test
```

So this way rar 5.x format will be supported and you won't be adding sources repository which will bring older unrar or won't be needing to install dependencies..

Though these should be correct, I'm not sure, I'd appreciate if you could test.

Fourdee, 2016-08-16:
status:

Lighttpd
Web PW =  :u6307: 
Transdroid = :u55b6: 

Nginx
Web PW = :u6307: 
Transdroid =  :u6307: 

Apache2
Web PW = :u5408: 
Transdroid = :u5408:

Ardakilic, 2016-08-16:
Oh and, mediainfo may be overkill if you'd ask me. That will possibly bring tons of dependencies and since I don't care so I never install. So un-dietish.

You don't have to favor dependencies for every rutorrent plugin. Instead, I'd prefer you echoing steps to install them.

Fourdee, 2016-08-16:
@Ardakilic 

> Oh and, mediainfo may be overkill if you'd ask me. That will possibly bring tons of dependencies and since I don't care so I never install. So un-dietish.

FFmpeg is the worst offender :)

I'll leave them in for now, testing, possibly take them out prior to release.

Ardakilic, 2016-08-16:
@Fourdee Exactly :) Them being left out at the release would be great then.

Fourdee, 2016-08-18:
@Ardakilic 
All done, many thanks for your help on this one, wouldnt of been possible without it, great work :+1: 

Docs: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=2603#p2603

Marking this as closed.

closed 2016-08-18 by Fourdee

Ardakilic, 2016-08-18:
@Fourdee Awesome! Glad to hear it's implemented, thank you for your hard work! 🎉 💃 I'll try as soon as possible and let you know!