MichaIng/DietPi #180: DietPi-specific apps are not executing directly when using ZSH (and possibly any other shell)

Ardakilic, 2016-01-27:
Hello,

I'm using DietPi with ODROID-XU4, loving it so far! :smile: 

I've a request, I hope this is considered.

I prefer using [zsh](http://www.zsh.org/) ( + [oh-my-zsh](https://github.com/robbyrussell/oh-my-zsh) ) over bash, however when I install it, DietPi-specific commands such as `dietpi-config`, `cpu` etc. are not executed. I had to dig the source code to find that they are actually `/Dietpi/dietpi/dietpi-config`, `/Dietpi/dietpi/dietpi-update` etc.

I would appreciate if these executables are also available when using `zsh` or any other shell without the need of using the whole path.

Thanks in advance,

Fourdee, 2016-01-27:
@Ardakilic 
DietPi uses bash alias to create the shortcut names. ```/etc/bash.bashrc```

You will need to ensure zsh loads/applies the bash aliases when connecting. I'am unsure as I've never used zsh before.

You could also run ```/DietPi/dietpi/dietpi-launcher``` which executes the full filepath of your selected DietPi script/program.

Ardakilic, 2016-01-27:
@Fourdee I've tried sourcing it manually but it breaks another stuff of oh-my-zsh:

```
➜  ~  source /etc/bash.bashrc
/etc/bash.bashrc:11: command not found: shopt
\u@\h:\w$ ls  # <- see, the arrow is now weird characters.
```

Since you've put it in the proper paths I'll be closing this and try to fix it in the zsh's end. 

Thank you again,

closed 2016-01-27 by Ardakilic

msongz, 2018-02-12:
@Ardakilic add this 3 lines at the bottom of .zshrc

/DietPi/dietpi/login
. /DietPi/dietpi/func/dietpi-globals
export PATH=/usr/bin:/bin:/usr/sbin:/sbin:$PATH

guilsa, 2019-11-25:
FYI if you uninstall oh-my-zsh and go back to bash you can avoid this issue.

MichaIng, 2019-11-25:
The problem is indeed that official support means that DietPi-Globals would need to be proven zsh compatible, and this required quite some testing or a detailed study of bash vs zsh differences which I do not find time for currently.

FWeissenb, 2019-11-26:
If I use the solution by @msongz it works 4 me. (I only tried dietpi-update :D )
Would it be possible to leave this ticket open and add scripts that make trouble (that are not zsh compatible)?

MichaIng, 2019-11-26:
@FWeissenb 
Yes the aliases will work, that's true. As well all our scripts will work, since the have a bash shebang, so will always run in bash, even if called from zsh shell.

The issue is all the other functions and inits that are done within: https://github.com/MichaIng/DietPi/blob/dev/dietpi/func/dietpi-globals
Especially that arrays start with index 1 in zsh instead of index 0 makes it incompatible, other minor flows due to missing ${var^} ${var,} support and probably more critical the much larger palette of file expansion syntax makes it even dangerous, e.g. we use a string that have no special meaning in bash and are involved in a removal or move task, and in zsh suddenly it has a syntax meaning and is expanded to a local file that is then removed or overwritten.

The only viable solution I can think of is to create a zsh-only login script, e.g. `/etc/zshrc.d/dietpi.zsh` should work, and then add only safe features, like the aliases and the dietpi-login script+banner (which are external scripts called as bash) to autologin/start and such. Further parts/features can be added as needed, like the G_AG* command set, when carefully checked or made zsh compatible, at a later time.

JPUnD, 2020-03-08:
> @Ardakilic add this 2 lines at the bottom of .zshrc
> 
> ```
> /DietPi/dietpi/login
> . /DietPi/dietpi/func/dietpi-globals
> ```
> 
> add this line at the top of .zshrc
> 
> ```
> export PATH=/usr/bin:/bin:/usr/sbin:/sbin:$PATH
> ```

Actually now instead of having `/DietPi/dietpi/login` you have to have `/DietPi/dietpi/dietpi-login`. 

Sorry for replying to a closed thread but i thought this might help, it seems to be working on my system (raspberry pi 4) as expected.

I would suggest adding a feature in dietpi-software that would allow the user to migrate between default shells and setup would be automatic, it doesn't seem that complicated and i would imagine most new users would rather use zsh or fish instead of bash.

MichaIng, 2020-03-08:
@JPUnD 
The issue is still present, that bash and zsh (and others) are not 100% compatible when it comes to scripting. And `dietpi-globals` includes quite a lot of code where e.g. wrong array index counting and others result in bugs and visual flakes at best.

But I have the idea to separate things a bid:
- `dietpi-login` and the program aliases do not require `dietpi-globals`, hence those can be loaded regardless of shell.
- The few environment checks we do, can be made cross-compatible easily.
- `dietpi-globals` would then only be loaded on bash for now, which includes G_* functions and variables, which I assume are rarely used directly by users anyway,

Only question I have is if zsh allows to add custom system-wide scripts via e.g. `/etc/zshrc.d/myscript.sh`?

JPUnD, 2020-03-14:
@MichaIng 

I have no idea how most of these things work, all i did was use that guy solution and changed it's path. If all the scripts have the correct shebang then they still will be run with bash, so having zsh should have no effect on the output or syntax of the script.

I have no idea if zsh allows custom system-wide scripts, but i've added custom paths before to zsh and calling the scripts worked as expected.

MichaIng, 2020-03-14:
@JPUnD 
> If all the scripts have the correct shebang then they still will be run with bash

That is not true. `/etc/bashrc.d/dietpi.bash` is not "executed" but "sourced" (leading dot `.`), which means that it is parsed like if you would type those commands right into your zsh session (regardless of shebang). And it again sources `/DietPi/dietpi/func/dietpi-globals`, which is hence as well parsed in zsh. `/DietPi/dietpi/login` and all the aliases dietpi-software, dietpi-config etc are executed, hence for those it is true that the originating shell does not matter.

JPUnD, 2020-03-14:
@MichaIng 

Ah alright then, but so far i've been using the scripts with zsh in my RPI4 and haven't encountered any issue yet.

MichaIng, 2020-03-15:
@JPUnD 
Jep as said, as long as you don't use G_* commands from command line, everything else should well. It is a fine workaround, but there above separations are required to not add buggy/failing G_* functions to zsh shells.

MichaIng, 2021-10-13:
Good to know. The contained global functions sometimes contain arrays starting with index 0, which AFAIK on zsh starts with index 1. E.g. does this work fine after sourcing the script via `emulate bash -c '. /etc/bashrc.d/dietpi.bash'`:
```sh
G_EXEC_OUTPUT=1 G_EXEC echo success
```

scorgn, 2021-10-13:
Yup, that's what was causing issues for me sourcing directly with ZSH.

Here is an example from my terminal using ZSH.

```
$ cat ~/test.bash
FOOBAR () {
   TEST_VAR[0]="Zero"
   TEST_VAR[1]="One"
   TEST_VAR[2]="Two"

   echo ${TEST_VAR[0]}
}
$ source ~/test.bash
$ FOOBAR
FOOBAR:1: TEST_VAR: assignment to invalid subscript range
$ emulate bash -c '. ~/test.bash'
$ FOOBAR
Zero
```

freshgiammi, 2021-11-28:
> Be changing `. $i` to `emulate bash -c '. "$i"'` you're able to source those files and use anything defined within them. So the one liner to put in your `.zshrc` would be
> 
> ```
> for i in /etc/bashrc.d/*.sh /etc/bashrc.d/*.bash; do [ -r "$i" ] && emulate bash -c '. "$i"'; done; unset -v i
> ```

Just a pointer to those bumping into this, it seems like this doesn't work anymore (at least on DietPi v7.8.2), as it triggers
```
/boot/dietpi/func/dietpi-globals:828: parse error near `('
[FAILED] DietPi-Login | Failed to load DietPi-Globals. Skipping DietPi login scripts...
```
with 828 being [this line](https://github.com/MichaIng/DietPi/blob/561f7a7724e51623742a016f12c03bef66c235f3/dietpi/func/dietpi-globals#L828), so everyone running something newer than [v7.7](https://github.com/MichaIng/DietPi/commit/b725fc43568693ea1400845c68a4bc2783495956) is probably affected.

This does not seem to be a omz related issue since a clean zsh suffers from this too. Commenting the line is enough to get everything (at least, it seems like it) loaded correctly. I'd fix it by trying something else than commenting entirely, but my bash/zsh knowledge is fairly limited.

MichaIng, 2021-11-28:
ZSH doesn't support extended regex?? It cannot be the array truncate as this is done in line 1511 as well.

The line will go with the first release next year, as we then drop support for Debian Stretch, but we use extended regex in quite a lot of places and will surely add it into global functions as well, when it fits.

I'd report this to ZSH: If they do have a bash support mode, then extended regex support in bash conditionals is simply mandatory. It is one of the main reasons why one may choose bash over bourne shell.

freshgiammi, 2021-11-28:
FWIW, it looks like zsh's emulate doesn't actually emulate bash (see [here](https://zsh.sourceforge.io/Doc/Release/Shell-Builtin-Commands.html) and [here](https://zsh.sourceforge.io/Doc/Release/Invocation.html#Compatibility)).

> Zsh tries to emulate sh or ksh when it is invoked as sh or ksh respectively; more precisely, it looks at the first letter of the name by which it was invoked, excluding any initial ‘r’ (assumed to stand for ‘restricted’), and if that is ‘b’, ‘s’ or ‘k’ it will emulate sh or ksh. 

and falls back to 'sh' if 'bash' is given. So yeah, I guess that it just isn't supported.

MichaIng, 2021-11-28:
Hmm, if it emulates `sh`, then why are e.g. `bash` conditionals (`[[ condition ]]`) and arrays in general not an issue? Is it `zsh` syntax as well and in emulation mode it allows `zsh` syntax + `sh` syntax or so? Strange is only that all these shells are bourne shell `sh` compatible anyway, so `sh` emulation doesn't make any sense/difference as long the syntax is not restricted to bourne shell only 🤔.

freshgiammi, 2021-11-28:
What if we use BRE instead of ERE ([see here](https://en.wikibooks.org/wiki/Regular_Expressions/POSIX-Extended_Regular_Expressions))? Escaping characters and writing the regex like:
```
[[ $G_DISTRO -le 4 && ${acommand[*]::3} =~ ^'systemctl '\(en\|dis\)'able --now'$ && $(dpkg-query -Wf '${version}' systemd) == '232'* ]] || return
```
seems to let zsh load `dietpi-globals` correctly. Again, my bash/zsh knowledge is fairly limited so I don't know if this can be the best workaround.

MichaIng, 2021-11-28:
This breaks it at least in bash, I'm pretty sure it also doesn't work as expected in zsh, while only the syntax is correct: `=~` in a bash conditional means to interpret extended regex without escaping (what one might know from `grep` when not using the `-E` flag. Adding the backslash makes those characters being interpreted literally, so that it doesn't do what it's supposed to do.

j-lakeman, 2023-11-06:
> @Ardakilic add this line at the bottom of .zshrc
> 
> ```
> /boot/dietpi/dietpi-login
> ```
> 
> 2021 update for the latest commit
> 
> add this line at the top of .zshrc
> 
> ```
> export PATH=/usr/bin:/boot/dietpi:/bin:/usr/sbin:/sbin:$PATH
> ```

… which would be for fish shell:

`fish_add_path /boot/dietpi` in the login section
`/boot/dietpi/dietpi-login` in the interactive section