MichaIng/DietPi #181: DietPi-Software | Add Cozy installation option

Fourdee, 2016-01-28:
## Donator request (high priority).

Cozy (repo) https://docs.cozy.io/en/host/install/install-on-debian.html

Cozy (lite) https://github.com/cozy-labs/cozy-light

Fourdee, 2016-01-31:
Moved to v109

Fourdee, 2016-02-05:
```
The following packages have unmet dependencies:
 cozy : PreDepends: couchdb but it is not installable
```
Build couchdb source

Fourdee, 2016-02-06:
```
#REQUIRES LAMP

apt-get install -y ca-certificates apt-transport-https
apt-get install -y nodejs nodejs-legacy npm && npm install -g npm@latest-2

#Cozy repo/key
wget -O - https://debian.cozycloud.cc/cozy.gpg.key 2>/dev/null | apt-key add -
echo -e "deb https://debian.cozycloud.cc/debian jessie main" > /etc/apt/sources.list.d/cozy.list

apt-get update

apt-get install -y erlang-base libjs-jquery-form erlang-crypto erlang-eunit erlang-inets erlang-os-mon erlang-public-key erlang-ssl erlang-syntax-tools erlang-tools erlang-xmerl libmozjs185-1.0 libnspr4 libnspr4-0d

#couchdb arm
wget dietpi.com -O package.deb
dpkg -i package.deb
rm package.deb

apt-get install -y cozy
```

Fourdee, 2016-02-08:
Test installation of cozy took over 5 hours to install on a OrangePi PC. Calendar and file module errors. This is taking much longer than expected to implement into DietPi, and will be delayed.

Fourdee, 2016-09-28:
> https://cozy-labs.github.io/cozy-light/

```
npm install cozy-light -g
```

Fourdee, 2016-09-28:
```
root@DietPi:~# cozy-light install-distro personal-cloud
(node:10271) fs: re-evaluating native module sources is not supported. If you are using the graceful-fs module, please update it to a more recent version.
info  - Start distribution installation for personal-cloud
info  - Installing plugins...
info  - Fetching data...
cozy-labs/cozy-light-simple-dashboard/master/package.json
info  - Installing plugin cozy-light-simple-dashboard...
(node:10271) fs: re-evaluating native module sources is not supported. If you are using the graceful-fs module, please update it to a more recent version.
(node:10271) fs: re-evaluating native module sources is not supported. If you are using the graceful-fs module, please update it to a more recent version.
cozy-light-simple-dashboard@0.1.0 node_modules/cozy-light-simple-dashboard
└── serve-static@1.11.1 (parseurl@1.3.1, escape-html@1.0.3, encodeurl@1.0.1, send@0.14.1)
info  - cozy-light-simple-dashboard installed. Enjoy!
info  - Fetching data...
cozy-labs/cozy-light-auth/master/package.json
info  - Installing plugin cozy-light-auth...
info  - Configuration changed...
cozy-light-auth@0.1.0 node_modules/cozy-light-auth
├── connect-flash@0.1.1
├── cookie-parser@1.4.3 (cookie-signature@1.0.6, cookie@0.3.1)
├── read@1.0.7 (mute-stream@0.0.6)
├── passport-local@1.0.0 (passport-strategy@1.0.0)
├── bcryptjs@2.3.0
├── randomstring@1.1.5 (array-uniq@1.0.2)
├── passport@0.2.2 (passport-strategy@1.0.0, pause@0.0.1)
├── cookie-session@1.2.0 (on-headers@1.0.1, debug@2.2.0, cookies@0.5.0)
└── body-parser@1.15.2 (content-type@1.0.2, bytes@2.4.0, depd@1.1.0, qs@6.2.0, on-finished@2.3.0, http-errors@1.5.0, debug@2.2.0, raw-body@2.1.7, iconv-lite@0.4.13, type-is@1.6.13)
info  - cozy-light-auth installed. Enjoy!
info  - Installing apps...
info  - Fetching data...
cozy-labs/tasky/master/package.json
info  - Configuration changed...
info  - Installing app cozy-tasky...
npm WARN deprecated jade@1.8.2: Jade has been renamed to pug, please install the latest version of pug instead of jade
npm WARN deprecated transformers@2.1.0: Deprecated, use jstransformer
npm WARN peerDependencies The peer dependency bn.js@^0.15.0 included from miller-rabin will no
npm WARN peerDependencies longer be automatically installed to fulfill the peerDependency
npm WARN peerDependencies in npm 3+. Your application will need to depend on it explicitly.
npm WARN deprecated tough-cookie@0.9.15: ReDoS vulnerability parsing Set-Cookie https://nodesecurity.io/advisories/130
npm WARN deprecated object-keys@0.2.0: Please update to the latest object-keys
|
> leveldown@0.10.6 install /root/.cozy-light/node_modules/cozy-tasky/node_modules/americano-cozy-pouchdb/node_modules/jugglingdb-pouchdb-adapter/node_modules/pouchdb/node_modules/leveldown
> node-gyp rebuild

make: Entering directory '/root/.cozy-light/node_modules/cozy-tasky/node_modules/americano-cozy-pouchdb/node_modules/jugglingdb-pouchdb-adapter/node_modules/pouchdb/node_modules/leveldown/build'
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/db/builder.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/db/db_impl.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/db/db_iter.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/db/filename.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/db/dbformat.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/db/log_reader.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/db/log_writer.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/db/memtable.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/db/repair.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/db/table_cache.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/db/version_edit.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/db/version_set.o
../deps/leveldb/leveldb-1.14.0/db/version_set.cc:58:13: warning: ‘std::string leveldb::{anonymous}::IntSetToString(const std::set<long long unsigned int>&)’ defined but not used [-Wunused-function]
 std::string IntSetToString(const std::set<uint64_t>& s) {
             ^
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/db/write_batch.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/helpers/memenv/memenv.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/table/block.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/table/block_builder.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/table/filter_block.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/table/format.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/table/iterator.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/table/merger.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/table/table.o
  CXX(target) Release/obj.target/leveldb/deps/leveldb/leveldb-1.14.0/table/table_builder.o
[... 463 lines cut ...]
gyp ERR! cwd /root/.cozy-light/node_modules/cozy-calendar/node_modules/pouchdb/node_modules/leveldown
gyp ERR! node -v v6.7.0
gyp ERR! node-gyp -v v3.0.1
gyp ERR! not ok
npm WARN optional dep failed, continuing leveldown@0.10.6
{ Error: time@0.11.1 install: `node-gyp rebuild`
Exit status 1
[... cut ...]
    at ChildProcess.<anonymous> (/usr/local/lib/node_modules/cozy-light/node_modules/npm/lib/utils/spawn.js:24:14)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:877:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
  code: 'ELIFECYCLE',
  pkgid: 'time@0.11.1',
  stage: 'install',
  script: 'node-gyp rebuild',
  pkgname: 'time' }
error - cozy-calendar installation failed.
time@0.11.1 install: `node-gyp rebuild`
Exit status 1
error - An error occured while installing your distribution
```

Fourdee, 2016-09-29:
Repo install:

No support for ARM64.

```
apt-get install -y ucf wget ca-certificates apt-transport-https
wget -O - https://debian.cozycloud.cc/cozy.gpg.key 2>/dev/null | apt-key add -
echo 'deb https://debian.cozycloud.cc/debian jessie cozy' \
 > /etc/apt/sources.list.d/cozy.list

apt-get update
apt-get install python-cozy-management
cozy_management install_cozy

```

Fourdee, 2017-12-30:
General Cleanup:
Marking all software requests as closed. Once interest peaks, we'll reopen once work starts.

closed 2017-12-30 by Fourdee

studioai, 2019-12-23:
https://docs.cozy.io/en/tutorials/selfhost-debian/

according to the document, it sounds like there's arm64 build.

MichaIng, 2019-12-23:
Indeed, their repo seems to ship everything required. Consider adding it to our FeatHub page and link back from there: https://feathub.com/MichaIng/DietPi