MichaIng/DietPi #185: XU4 | Automation | User input required for kernel update

Fourdee, 2016-02-02:
http://fuzon.co.uk/phpbb/viewtopic.php?f=11&t=304#p1305

Fourdee, 2016-02-03:
Also applies to C1

Fourdee, 2016-02-03:
Resolved for v109:
```
DEBIAN_FRONTEND='noninteractive' apt-get upgrade -y
```

rhkean, 2016-02-11:
it's important to note that this will not update the linux kernel packages during the automated install and you will need to perform the following when all is done:

apt-get install linux-headers-armhf-odroid-xu3 linux-image-armhf-odroid-xu3

(since this fix causes the apt-get upgrade command to skip these packages)

Fourdee, 2016-02-11:
@rhkean 
:+1: I'll run some tests and verify.

rhkean, 2016-02-11:
I tacked that line at the end of my Automation_Custom_Script.sh...
ran beautifully and don't get any "package is held back" messages

Fourdee, 2016-02-11:
Ok, I just tested the C1, the kernel packages are installed and updated correctly.
[image]

Now to test the XU4 :)

rhkean, 2016-02-11:
Looks like, in the script, it will definitely have to be $HW_MODEL dependent.  Doesn't look like the virtual packages (linux-headers and linux-image) resolve to definitive packages.

it's either
    apt-get -y install linux-headers-armhf-odroid-c1 and linux-image-armhf-odroid-c1
or
    apt-get -y install linux-headers-armhf-odroid-XU3 and linux-image-armhf-odroid-XU3

would be a more generalized/universal fix if you could just call
    apt-get -y install linux-headers linux-image

Fourdee, 2016-02-11:
@rhkean 
The specific kernel/header packages for each device is pre-installed on the DietPi images for Odroids. 

The Odroid kernel/header updates and packages are a little different.
The kernel updates are provided by meveric and using the mdjir Odroid repo. So when the user runs ```apt-get upgrade```, this also pulls in the kernel/header updates for the device.

I just checked the XU4, seems its upgrading the kernel/headers fine.
[image]

Fourdee, 2016-02-11:
> Looks like, in the script, it will definitely have to be $HW_MODEL dependent. Doesn't look like the virtual packages (linux-headers and linux-image) resolve to definitive packages.

Yep that would work. Although I think we might be able to use debconf-set-selections. Looking into it.

rhkean, 2016-02-11:
doh!!!! now you're out of my depth.... LOL

Fourdee, 2016-02-11:
Looks like debconf is linked to the kernel version number. As this will change, it won't resolve future updates.
```
root@DietPi:~# debconf-get-selections | grep linux-
linux-image-3.10.92+    linux-image-3.10.92+/preinst/already-running-this-3.10.92+      note
linux-image-3.10.92+    linux-image-3.10.92+/preinst/failed-to-move-modules-3.10.92+    note
linux-image-3.10.92+    linux-image-3.10.92+/prerm/removing-running-kernel-3.10.92+     boolean true
linux-image-3.10.92+    linux-image-3.10.92+/postinst/kimage-is-a-directory     note
linux-image-3.10.92+    linux-image-3.10.92+/postinst/depmod-error-initrd-3.10.92+      boolean false
linux-image-3.10.92+    linux-image-3.10.92+/preinst/overwriting-modules-3.10.92+       boolean false
linux-image-3.10.92+    linux-image-3.10.92+/preinst/abort-overwrite-3.10.92+   note
linux-image-3.10.92+    linux-image-3.10.92+/postinst/depmod-error-3.10.92+     boolean false
linux-image-3.10.92+    linux-image-3.10.92+/postinst/old-dir-initrd-link-3.10.92+      boolean true
linux-image-3.10.92+    linux-image-3.10.92+/postinst/old-system-map-link-3.10.92+      boolean true
```

Ok. 
The following code has been placed into the patch file. This will update all XU3/4 installations.
```
DEBIAN_FRONTEND='noninteractive' apt-get install -y linux-headers-armhf-odroid-xu3 linux-image-armhf-odroid-xu3
```
When the next kernel update is available, I will check if ```DEBIAN_FRONTEND='noninteractive' apt-get upgrade``` installs it. If the packages are still held back, I will also add the code above to ```DietPi-Software``` when installation is started for XU3/4.

rhkean, 2016-02-11:
so, did you put this in dietpi-software or dietpi-update?
and, have you checked it into github?  I'll try and test tonight.

Fourdee, 2016-02-11:
@rhkean 
Exisiting XU4 installations will update the kernel when the user updates DietPi: https://github.com/Fourdee/DietPi/blob/testing/dietpi/patch_file#L324-L329

Yes i just tried it as I'am currently updating the XU4 image.  The new image will have the latest kernel, but I will need to monitor if the packages are being held back in the future. If they are, I will then add this to dietpi-software:
```
DEBIAN_FRONTEND='noninteractive' apt-get install -y linux-headers-armhf-odroid-xu3 linux-image-armhf-odroid-xu3
```

> 'll try and test tonight.

The patch won't be available until v109 is released. So if you want to try it, you will need to change ```/DietPi/dietpi.txt``` at the bottom ```gitbranch=testing``` then run ```dietpi-update``` to test.

rhkean, 2016-02-11:
yeah... I switched to the testing branch last night to test the curl fix.  that's how I found out that the kernel packages were held back with the "noninteractive" fix.

I'll use the v106 image I have... verify that it updates from v106 and runs to completion with updated kernel packages without need for a any human intervention.

Fourdee, 2016-02-11:
@rhkean 

> I'll use the v106 image I have... verify that it updates from v106 and runs to completion with updated kernel packages without need for a any human intervention.

Excellent, thanks.

rhkean, 2016-02-12:
the start-from v106 image did not work.  it ran the entire install without requiring a keyboard intervention and installed the 3.10.96+ kernel packages, HOWEVER
it still boots to 3.10.92+
it is showing an error on boot about unable to load kernel modules
and, worse yet, I can seem to convince it to upgrade to the 3.10.96+ kernel now, b/c it says it's already installed.

I'll try the v109 image next

Fourdee, 2016-02-12:
@rhkean 
Bizzare, I'll retest the v106 > v109 patch and see whats causing it. 

The v109 image is upgraded to .96+:
```
root@DietPi-XU4:~# dpkg -l | grep linux-
ii  linux-headers-3.10.96+         3.10.96-20160205-XU3             armhf        Header files related to Linux kernel, specifically,
ii  linux-headers-armhf-odroid-xu3 3.10.96-1                        armhf        Linux Image/Headers Meta Package for ODROIDs
ii  linux-image-3.10.96+           3.10.96-20160205-XU3             armhf        Linux kernel binary image for version 3.10.96+
ii  linux-image-armhf-odroid-xu3   3.10.96-1                        armhf        Linux Image/Headers Meta Package for ODROIDs
```

rhkean, 2016-02-12:
yeah... the new v109 image is working perfectly.  I just reimaged my XU4 this again this morning with it.  It processed all the stock dietpi.txt settings and my entire auto_custom_script flawlessly.  :)

The v106 to v109, however... :(
it's installing the packages, I just couldn't tell why is wasn't applying them.  But, I've never worked with debconf before, so I was stumbling around in unknown territory. 
oh, I also noticed that it wasn't loading modules like the bluetooth.... I'm guessing that it purged the 92+ modules, but since is wasn't booting the 96+ kernel, it was stuck.  (BT worked fine on the new image, btw)

Fourdee, 2016-02-12:
> The v106 to v109, however... :(
> it's installing the packages, I just couldn't tell why is wasn't applying them. But, I've never worked with debconf before, so I was stumbling around in unknown [...]

@rhkean Thanks for testing. 
It appears we have no current way of automating the XU3/4 kernel upgrade without causing damage. So for now, i'll remove it from the patch code.

The v109 image has the latest kernel.

To prevent future kernel upgrades causing issues with automation, I have put the kernel packages on hold. Users can then upgrade the kernel manually with ```apt-get install linux-headers-armhf-odroid-xu3 linux-image-armhf-odroid-xu3```

rhkean, 2016-02-12:
@Fourdee I'm inclined to agree.

That method works for sure.  In fact, adding it as the last line in my Auto_Custom_Script.sh with a "-y" worked perfectly since there is an almost immediate reboot from that point.

Can you add that as a comment in the release notes?  (I doubt many people will read through this issue log)  ROFL

Fourdee, 2016-02-12:
> Can you add that as a comment in the release notes? (I doubt many people will read through this issue log) ROFL

Hehe :+1:  :)

I've updated the release notes. There may be some spelling and grammar errors (i've had a few tonight ;)):
https://github.com/Fourdee/DietPi/blob/testing/CHANGELOG.txt#L6

Planning to get v109 released before Sunday, that is, if the remaining RPi automation tests are successful.

rhkean, 2016-02-12:
ROFL

Well, you enjoy your evening.  I'll verify the RPi tests tonight.  (it's only 4:30pm here)  So, hopefully I'll be able to give you some answers before you recover tomorrow.  :+1:

Fourdee, 2016-02-12:
@rhkean 
lol, Will do! 
There's no rush on the tests. Make sure you enjoy your evening also :)

Fourdee, 2016-02-13:
Resolved for v109.

closed 2016-02-13 by Fourdee