MichaIng/DietPi #187: Pi-hole web-interface not showing stats

fairyfeller, 2016-02-04:
Made a clean fresh install of DietPi (v 108) with Pi-hole - seems like pi-hole working (i mean blocking ads), but there's some elements missing in web-interface and no stats.. 
[image]

Fourdee, 2016-02-04:
@fairyfeller 
Looks like they moved the logfile parsing code out of index.php to another shell script: https://github.com/pi-hole/AdminLTE/blob/master/index.php#L2
https://github.com/pi-hole/pi-hole/blob/master/advanced/Scripts/chronometer.sh

Run the following:
```
wget https://raw.githubusercontent.com/pi-hole/pi-hole/master/advanced/Scripts/chronometer.sh -O /usr/local/bin/chronometer.sh
chmod +x /usr/local/bin/chronometer.sh
```

Then refresh the webpage (ctrl+f5).

Fourdee, 2016-02-04:
And BC (an additional package) is now required for PiHole to work out the percentage.
https://github.com/pi-hole/pi-hole/blob/master/advanced/Scripts/chronometer.sh#L55

```
apt-get install bc
```

Fourdee, 2016-02-04:
[image]

fairyfeller, 2016-02-04:
Yep, after getting new chronometer.sh and BC - it started to show some stats..
Still missing some menus on the left side as Update, Whitelist, and Reboot

Fourdee, 2016-02-04:
Resolved for v109. Will also be applied as a patch during update.

PiHole on DietPi will now be installed from a separate/stable fork of the project.
https://github.com/Fourdee/pi-hole
https://github.com/Fourdee/AdminLTE

This will allow us to merge/update the PiHole project in a testing environment, before being rolled out to DietPi users.

Fourdee, 2016-02-04:
@fairyfeller 

> Yep, after getting new chronometer.sh and BC - it started to show some stats..
Still missing some menus on the left side as Update, Whitelist, and Reboot

Will take a look.

Updates for PiHole are rolled out as patches with dietpi-update.

Fourdee, 2016-02-04:
@fairyfeller 
I cant see any references to blacklist/whitelist/reboot in the sourcecode of the webpage: https://github.com/Fourdee/AdminLTE/blob/master/index.php

Are you sure this has been implemented by Pi-Hole?

fairyfeller, 2016-02-05:
@Fourdee , i expected something like this [http://jacobsalmela.com/pi-hole-2-0-web-interface-dashboard/](url)
Thought it's already released :)

Fourdee, 2016-02-05:
Ah, good find. So it did exist but was removed.

As this resolves all outstanding issues, i'll close this ticket. If you experience any further issues or questions, please reopen.

closed 2016-02-05 by Fourdee