MichaIng/DietPi #191: DietPi-Software | Add xrdp installation option

starstuff, 2016-02-09:
Please add xrdp app for windows remote desktop support, an alternative to access lxde desktop.

`sudo apt-get install xrdp`

xenfomation, 2016-05-13:
@Fourdee -- is this still active as I have no issues taking care of this as I relentlessly use xrdp for my Debian-based distributions!

Nice, @starstuff !!  Thanks!

Fourdee, 2016-05-13:
@xenfomation 
Yep :+1: , all open tickets are up for the taking.

xenfomation, 2016-05-17:
Double checking deps based on recent Deb work with xrdp.  I used no machine, but will post findings.

xenfomation, 2016-05-18:
Update as, of course, xrdp works and requires < 13MiB of space!

Screenshots attached to show the result of `apt-get install xrdp`:

[image]
[image]
[image]

### Linux Software
Along with No Machine, XRDP is great -- maybe this will dictate a REMOTE SERVER option under Linux Software

I will work on a prototype.

Thanks!

xenfomation, 2016-05-27:
Working on re-testing and incorporation under "Software Linux" -> "Remote Desktop".  Seems logical to me as each package will have the description to state "Server" or "Client".

xenfomation, 2016-05-29:
Tested and preparing to commit.

[image]

Fourdee, 2016-06-04:
Additional implementation and testing for reference: https://github.com/Fourdee/DietPi/pull/364

Fourdee, 2016-06-08:
Documentation to finish: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=70#p2074

Fourdee, 2016-06-08:
@xenfomation 
Is there a way to disable applications that are not installed, being listed?
[image]

Fourdee, 2016-06-08:
Done.

closed 2016-06-08 by Fourdee