MichaIng/DietPi #192: Add DietPi to Raspberry Pi Third Party OS Images

starstuff, 2016-02-09:
Please add DietPi to the  Raspberry Pi Third Party OS Images to be more accessible to the general public

[https://www.raspberrypi.org/downloads/](url)

Im a new linux user, and find DietPi to be super easy to use!

Fourdee, 2016-02-09:
@starstuff 
Thanks for all the requests.

> Please add DietPi to the Raspberry Pi Third Party OS Images to be more accessible to the general public
I've tried on various occasions to make this happen with forum posts and a few tweets https://twitter.com/DietPi_/status/693493797427232770

Unfortunately, there are no FAQ, rules, requirements, or contact information available for users who wish to add other distros.

I think the only way we can achieve this is to promote, discuss and spread the word on DietPi. Maybe once we have enough "popularity", the Raspberry Pi staff may take notice :)

bhaveshgohel, 2016-02-10:
+1 DietPi Rulzz... :fire:

closed 2016-10-29 by Fourdee