MichaIng/DietPi #195: DietPi-Software | Add QuiteRSS installation option

starstuff, 2016-02-09:
Please add QuiteRSS to dietpi,

> QuiteRSS is a open-source cross-platform RSS/Atom news feeds reader

[https://quiterss.org/](url)
[https://quiterss.org/quiterss-for-other-os](url)
[https://packages.debian.org/search?keywords=quiterss](ur

xenfomation, 2016-05-10:
Heads up -- went to install this package... 

381 packages, 489MB of space required.

IMHO, as much as I love RSS -- does this seem a little steep?
[image]

Fourdee, 2016-05-11:
@xenfomation 
It looks like its installing Xserver aswell. So that could be installed by ```XSERVERXORG``` in ```dietpi-software```.

Have you tried ```apt-get install xxxx --no-install-recommends```? Usually gives you the bare minimal.

xenfomation, 2016-05-11:
Hey, @Fourdee  -- long, long two days -- so, pardon me not running that apt-get flag!

1.  It having X deps is cool by me, but thanks for pointing out that is where most of the space comes from.  I was about to write us "diet-atomrss" - which I may do - for the CLI or GTK+

2.  Back on track, here is the output of apt-get install xxxxx --no-install-recommends:

```
root@DietPi:~# apt-get install quiterss --no-install-recommends
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following extra packages will be installed:
  fontconfig fontconfig-config fonts-dejavu-core fonts-freefont-ttf iso-codes
  liba52-0.7.4 libaa1 libasound2 libasound2-data libass5 libasyncns0 libaudio2
  libavahi-client3 libavahi-common-data libavahi-common3 libavc1394-0
  libavcodec56 libavformat56 libavresample2 libavutil54
  libbasicusageenvironment0 libbluray1 libcaca0 libcddb2 libcdio13
  libchromaprint0 libdc1394-22 libdca0 libdirectfb-1.2-9 libdvbpsi9 libdvdnav4
  libdvdread4 libebml4 libegl1-mesa libenca0 libfaad2 libflac8 libfontconfig1
  libfreerdp-cache1.1 libfreerdp-client1.1 libfreerdp-codec1.1
  libfreerdp-common1.1.0 libfreerdp-core1.1 libfreerdp-crypto1.1
  libfreerdp-gdi1.1 libfreerdp-locale1.1 libfreerdp-primitives1.1
  libfreerdp-rail1.1 libfreerdp-utils1.1 libfreetype6 libfribidi0 libgbm1
  libgl1-mesa-glx libglapi-mesa libgles1-mesa libgles2-mesa libgpm2
  libgraphite2-3 libgroupsock1 libgsm1 libgstreamer-plugins-base1.0-0
  libgstreamer1.0-0 libharfbuzz0b libice6 libiso9660-8 libjbig0
  libjpeg62-turbo libjpeg8 libjson-c2 libkate1 liblcms2-2 liblircclient0
  liblivemedia23 liblua5.2-0 libmad0 libmatroska6 libmng1 libmodplug1
  libmp3lame0 libmpcdec6 libmpeg2-4 libmtp-common libmtp9 libogg0 libopenjpeg5
  libopus0 liborc-0.4-0 libphonon4 libpostproc52 libproxy-tools libproxy1
  libpulse-mainloop-glib0 libpulse0 libqt4-dbus libqt4-network libqt4-sql
  libqt4-sql-sqlite libqt4-xml libqt4-xmlpatterns libqtcore4 libqtdbus4
  libqtgui4 libqtwebkit4 libraw1394-11 libresid-builder0c2a libsamplerate0
  libschroedinger-1.0-0 libsdl-image1.2 libsdl1.2debian libshine3 libshout3
  libsidplay2 libsm6 libsndfile1 libspeex1 libspeexdsp1 libsqlite3-0
  libswscale3 libtag1-vanilla libtag1c2a libtheora0 libtiff5 libtwolame0
  libupnp6 libusageenvironment1 libva-drm1 libva-x11-1 libva1 libvcdinfo0
  libvlc5 libvlccore8 libvncclient0 libvorbis0a libvorbisenc2 libvpx1
  libwayland-client0 libwayland-server0 libwebp5 libwinpr-crt0.1
  libwinpr-crypto0.1 libwinpr-dsparse0.1 libwinpr-environment0.1
  libwinpr-file0.1 libwinpr-handle0.1 libwinpr-heap0.1 libwinpr-input0.1
  libwinpr-interlocked0.1 libwinpr-library0.1 libwinpr-path0.1
  libwinpr-pool0.1 libwinpr-registry0.1 libwinpr-rpc0.1 libwinpr-sspi0.1
  libwinpr-synch0.1 libwinpr-sysinfo0.1 libwinpr-thread0.1 libwinpr-utils0.1
  libwrap0 libx11-6 libx11-data libx11-xcb1 libx264-142 libxau6
  libxcb-composite0 libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-keysyms1
  libxcb-present0 libxcb-randr0 libxcb-render0 libxcb-shape0 libxcb-shm0
  libxcb-sync1 libxcb-xfixes0 libxcb-xv0 libxcb1 libxdamage1 libxdmcp6
  libxext6 libxfixes3 libxi6 libxinerama1 libxkbfile1 libxml2 libxpm4
  libxrender1 libxshmfence1 libxt6 libxtst6 libxvidcore4 libxxf86vm1
  libzvbi-common libzvbi0 phonon phonon-backend-vlc qdbus qtchooser
  qtcore4-l10n ucf vlc vlc-data vlc-nox vlc-plugin-pulse x11-common
Suggested packages:
  isoquery libasound2-plugins alsa-utils nas libbluray-bdj
  libchromaprint-tools python-acoustid libdvdcss2 freerdp-x11 gpm
  libvisual-0.4-plugins gstreamer-codec-install gnome-codec-install
  gstreamer1.0-tools liblcms2-utils lirc opus-tools pulseaudio libqt4-dev
  libthai0 qt4-qtconfig libraw1394-doc speex phonon-backend-gstreamer
  phonon-backend-mplayer videolan-doc
Recommended packages:
  libaacs0 libgl1-mesa-dri gstreamer1.0-plugins-base libmtp-runtime libcups2
  va-driver-all va-driver tcpd xml-core vlc-plugin-notify vlc-plugin-samba
  xdg-utils
The following NEW packages will be installed:
  fontconfig fontconfig-config fonts-dejavu-core fonts-freefont-ttf iso-codes
  liba52-0.7.4 libaa1 libasound2 libasound2-data libass5 libasyncns0 libaudio2
  libavahi-client3 libavahi-common-data libavahi-common3 libavc1394-0
  libavcodec56 libavformat56 libavresample2 libavutil54
  libbasicusageenvironment0 libbluray1 libcaca0 libcddb2 libcdio13
  libchromaprint0 libdc1394-22 libdca0 libdirectfb-1.2-9 libdvbpsi9 libdvdnav4
[... 21 lines cut ...]
  libwinpr-crypto0.1 libwinpr-dsparse0.1 libwinpr-environment0.1
  libwinpr-file0.1 libwinpr-handle0.1 libwinpr-heap0.1 libwinpr-input0.1
  libwinpr-interlocked0.1 libwinpr-library0.1 libwinpr-path0.1
  libwinpr-pool0.1 libwinpr-registry0.1 libwinpr-rpc0.1 libwinpr-sspi0.1
  libwinpr-synch0.1 libwinpr-sysinfo0.1 libwinpr-thread0.1 libwinpr-utils0.1
  libwrap0 libx11-6 libx11-data libx11-xcb1 libx264-142 libxau6
  libxcb-composite0 libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-keysyms1
  libxcb-present0 libxcb-randr0 libxcb-render0 libxcb-shape0 libxcb-shm0
  libxcb-sync1 libxcb-xfixes0 libxcb-xv0 libxcb1 libxdamage1 libxdmcp6
  libxext6 libxfixes3 libxi6 libxinerama1 libxkbfile1 libxml2 libxpm4
  libxrender1 libxshmfence1 libxt6 libxtst6 libxvidcore4 libxxf86vm1
  libzvbi-common libzvbi0 phonon phonon-backend-vlc qdbus qtchooser
  qtcore4-l10n quiterss ucf vlc vlc-data vlc-nox vlc-plugin-pulse x11-common
0 upgraded, 206 newly installed, 0 to remove and 3 not upgraded.
Need to get 63.4 MB of archives.
After this operation, 211 MB of additional disk space will be used.
Do you want to continue? [Y/n]

```

3.  I didn't mean to sound snobbish with the space for DietPi as hey -- people can install it anyway!  X is no concern as your installer handles this, so in the end -- 211 MB for a GTK app makes sense

So, dear sir -- say the word and I shall test and add accordingly!  Need to leverage RSS off dietpi.com/forums if we add this!

Cheers!

xenfomation, 2016-05-27:
My error in letting this one slip by!  I had tested this with MATE and will be adding the manual steps to this shortly followed by "code"! :)

xenfomation, 2016-05-28:
Successful & tested on v118 w/ MATE, GNUStep, and others -- adding code tonight!

[image]

xenfomation, 2016-05-29:
Tested and added to my fork.

Fourdee, 2016-06-04:
Additional implementation and testing for reference: https://github.com/Fourdee/DietPi/pull/364

Fourdee, 2016-06-08:
Documentation to finish: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=70#p2075

Fourdee, 2016-06-08:
Done.

closed 2016-06-08 by Fourdee