MichaIng/DietPi #197: DietPi-Automation | dietpi-update runs before network established and initially fails.

Fourdee, 2016-02-09:
> http://fuzon.co.uk/phpbb/viewtopic.php?f=11&t=304&p=1332#p1331

Fourdee, 2016-02-09:
Added the -k flag which will allow "insecure" certs. 

Curl seems to have an issue with verifying certs straight after systemD services are completed?
```
curl -k -L https://raw.githubusercontent.com/Fourdee/DietPi/"$GITBRANCH"/dietpi/server_version > "$FILEPATH_TEMP"/server_version
```

rhkean, 2016-02-11:
Tested and confirmed on Odroid XU4 and RPi2

Fourdee, 2016-02-11:
@rhkean 
Excellent :+1: . Thanks for letting me know.

As this fix cannot be applied by patching (the issue occurs prior to DietPi updating), I will need update all the DietPi image files to v109 for this issue to be resolved. 

The new image files will be available from [dietpi.com](url) when v109 is released.

rhkean, 2016-02-11:
let me know if you need me to test the new image.  I can test the XU4 and RPi2 images.

Fourdee, 2016-02-11:
@rhkean 
Thanks. I will try and get them completed today, will let you know when they are available.

Fourdee, 2016-02-11:
@rhkean 
Here is the updated XU4 image: http://dietpi.com/downloads/images/testing/
I tried the following automation template: http://dietpi.com/downloads/dietpi-automation_templates/DietPi-Core%20(DietPi%20as%20minimal%20image)/DietPi-Core%20-%20OpenSSH%20Server%20-%20DHCP%20(Fully%20automated).txt
Appears to be working fine :)

Fourdee, 2016-02-12:
@rhkean 
I've uploaded the remaning DietPi v109 images: 
[http://dietpi.com/downloads/images/testing/](url)

If you are able to test the ones you can with automation, i'd be grateful.

rhkean, 2016-02-12:
I can test the XU4, RPi (Jessie), & RPi (Wheezy).  I'll test them this weekend.
I don't have a C1 or OrangePi... sorry
Do you just support the RPi2?  or do you also support the B+ and/or A+?  If so, I can test those, too.

Fourdee, 2016-02-12:
@rhkean 
Thanks I appreciate it. I'll test the Orange Pi and C1.

All RPi's are supported (the only exclusion is the compute module).

Theres no need to test additional RPi's (B/A/Zero etc) as we are only checking the initial automation code (which doesn't change per device). 
A single test of Wheezy and one of Jessie on any RPi would be great.

Fourdee, 2016-02-12:
C1 & OrangePi Tested Ok using:
http://dietpi.com/downloads/dietpi-automation_templates/DietPi-Core%20(DietPi%20as%20minimal%20image)/DietPi-Core%20-%20OpenSSH%20Server%20-%20DHCP%20(Fully%20automated).txt

rhkean, 2016-02-13:
I think the Wheezy image has some problems... several "cannot stat /proc/...." messages during boot.
then, at the end of the bootup, just before login prompt:
---------
DietPi-Process Tool
Applying settings, please wait...

warning: bad ps syntax, perhaps a bogus '-'?
See http://gitorious.org/procps/procps/blobs/master/Documentation/FAQ
----
renices cron, dhcp client, and dropbear

rhkean, 2016-02-13:
ok... not sure why that last post blew up the text....  probably a "markdown" thing (really need to commit that stuff to memory)

ran the wheezy image with my dietpi.txt file.  mysql failed.  not sure what else, but I'm definitely going to have to give this image a thumbs-down.  will test the jessie image next

rhkean, 2016-02-13:
the jessie image is good

Fourdee, 2016-02-13:
@rhkean 
I'll debug the Wheezy image today. Thanks for testing and confirming Jessie.

> warning: bad ps syntax, perhaps a bogus '-'?

Yep, looks that way. Jessie is too kind and allows ```ps aux``` and ```ps -aux``` :)

rhkean, 2016-02-13:
oh that's not a kindness.... its a new "feature" of ps.... it now supports posix, bsd, and gnu style options... and they're different.  "-a" and "a" are 2 different options now... the man page specifically calls out the "ps aux" and "ps -aux" are different.   I think its rather obnoxious.

here's a quick link:  http://manpages.debian.org/cgi-bin/man.cgi?query=ps&apropos=0&sektion=0&manpath=Debian+8+jessie&format=html&locale=en

Fourdee, 2016-02-13:
@rhkean 

> dietpi.txt file. mysql failed

Was this via a webserver stack automation setting?

Fourdee, 2016-02-13:
> I think its rather obnoxious.

"Welcome to Linux" lol :)

rhkean, 2016-02-13:
probably.  I haven't looked at the specifics of the settings is a while.  I just created a patch file from my v106 dietpi.txt and applied it to the v109.  I can send you the file.  Actually you should already have it from the fuzon forum...
http://fuzon.co.uk/phpbb/download/file.php?id=224

Fourdee, 2016-02-13:
Thanks. I'll give it a try now.

Fourdee, 2016-02-13:
@rhkean 
Ok, installation went fine. No Mysql errors, its running fine.

I did find an error in your dietpi.txt:
```
AUTO_Locale=en_GB en_US
```
Causes:
```
-bash: export: `en_US.UTF-8': not a valid identifier
```

Resolution:
You should use only 1 language entry, eg: ```AUTO_Locale=en_US```

We do have a know issue with AUTO_Locale not applying. Not sure if its the same with Wheezy but I'll look into it after v109 is out. https://github.com/Fourdee/DietPi/issues/174

Fourdee, 2016-02-13:
> I think the Wheezy image has some problems... several "cannot stat /proc/...." messages during boot

I believe this has been here since the dawn of time on Wheezy. As far as I can tell, its not DietPi specific and it also occurs on the official Raspbian Wheezy image.

Appears RPi have removed Wheezy from the downloads page. Looks like they are phasing it out:
https://www.raspberrypi.org/downloads/raspbian/

Fourdee, 2016-02-13:
@rhkean 
As far as I can tell, everything is now resolved with automation and the new images. I will release v109 now.

If we experience any further issues related to this ticket, we can reopen and patch in v110.

closed 2016-02-13 by Fourdee