MichaIng/DietPi #200: DietPi-Software | DietPi-Cleaner - error message while Test run

k-plan, 2016-02-14:
To test it on Odroid C1 (Jessie: v.109):
Start DietPi-Cleaner, enable same cleaners (APT, Log) and start Test (Simulate the cleaning process, without modifying any data.)

Message (locales: en_GB.UTF-8):

```
Completed

/DietPi/dietpi/dietpi-cleaner: line 306: - : syntax error: operand expected (error token is "- ")
```

Message (locales: de_DE.UTF-8):

```
Completed

/DietPi/dietpi/dietpi-cleaner: Zeile 306: - : Syntax Fehler: Operator erwartet. (Fehlerverursachendes Zeichen ist \"- \").
```
After this, you are back on cmd (root@DietPi:~#)

Fourdee, 2016-02-14:
@k-plan 
Hi, thanks for the report.

I was unable to replicate this error using the C1 and your tests. Could you please send a ```dietpi-bugreport``` and provide the reference code? This will allow me to see your current setup and settings.

k-plan, 2016-02-14:
Hi Fourdee,
thanks for your quick reply. Have send a bugreport.
Reference CODE: 001e06201ba0-0

Fourdee, 2016-02-16:
@k-plan 
Apologies for the delay. I am still unable to replicate this error. 

I've modified the dietpi-cleaner script to allow us to debug. Could you run the following on your system:
```
cd ~
wget http://dietpi.com/downloads/tests/dietpi-cleaner
chmod +x dietpi-cleaner
./dietpi-cleaner
```
Then run the same tests you did last time.

When it fails on line 306. Please copy and paste the following debug sections:
```
Debug 1

ROOT_SPACE_USED_BEFORE: 400
ROOT_SPACE_USED_AFTER: 400
ROOT_SPACE_FREE_DELTA: Cleaner has not been run - 0

Debug 2

ROOT_SPACE_USED_BEFORE: 400
ROOT_SPACE_USED_AFTER: 400
ROOT_SPACE_FREE_DELTA: 0
```

k-plan, 2016-02-16:
Hi Fourdee,

here we go:
(Message (locales: en_GB.UTF-8):

```
Completed

Debug 1

ROOT_SPACE_USED_BEFORE:
ROOT_SPACE_USED_AFTER:
ROOT_SPACE_FREE_DELTA: Cleaner has not been run - 0
./dietpi-cleaner: line 310: - : syntax error: operand expected (error token is "- ")
```

Message (locales: de_DE.UTF-8):

```
Completed

Debug 1

ROOT_SPACE_USED_BEFORE:
ROOT_SPACE_USED_AFTER:
ROOT_SPACE_FREE_DELTA: Cleaner has not been run - 0
./dietpi-cleaner: Zeile 310: - : Syntax Fehler: Operator erwartet. (Fehlerverursachendes Zeichen ist \"- \").

```
Thanks for your help.

Fourdee, 2016-02-17:
@k-plan 
Thanks.

Ok, it looks like the issue is when obtaining the current filesystem usage, which is pulled from ```df```.

Could you please run the following commands and paste results for each:
```
df --block-size MB | awk '{ print $3 }'
```

and
```
df
```

And finally:
```
sed -n 4p /DietPi/dietpi/.hw_model
```

Thanks.

k-plan, 2016-02-17:
Hi Fourdee,
thanks for reply. Here we go:

```
:~# df --block-size MB | awk '{ print $3 }'
Benutzt
735MB
0MB
6MB
0MB
0MB
0MB
2MB
1MB
0MB
27MB
0MB
```

```
:~# df
Dateisystem                                  1K-Blöcke Benutzt Verfügbar Verw% Eingehängt auf
/dev/sdb1                                      3196552  717080   2297384   24% /
udev                                             10240       0     10240    0% /dev
tmpfs                                           194212    5484    188728    3% /run
tmpfs                                           485528       0    485528    0% /dev/shm
tmpfs                                             5120       0      5120    0% /run/lock
tmpfs                                           485528       0    485528    0% /sys/fs/cgroup
tmpfs                                            10240    1076      9164   11% /DietPi
tmpfs                                            20480      12     20468    1% /var/log
tmpfs                                           485528       0    485528    0% /tmp
/dev/mmcblk0p1                                  131428   25428    106000   20% /boot
curlftpfs#ftp://user:password@192.168.0.100/ 999999999       0 999999999    0% /mnt/ftp_client
```

```
:~# **sed -n 4p /DietPi/dietpi/.hw_model**
/dev/mmcblk0p2
```

Additional Infos:
```
:~# df -h
Dateisystem                                  Größe Benutzt Verf. Verw% Eingehängt auf
/dev/sdb1                                     3,1G    701M  2,2G   24% /
udev                                           10M       0   10M    0% /dev
tmpfs                                         190M    5,4M  185M    3% /run
tmpfs                                         475M       0  475M    0% /dev/shm
tmpfs                                         5,0M       0  5,0M    0% /run/lock
tmpfs                                         475M       0  475M    0% /sys/fs/cgroup
tmpfs                                          10M    1,1M  9,0M   11% /DietPi
tmpfs                                          20M     12K   20M    1% /var/log
tmpfs                                         475M       0  475M    0% /tmp
/dev/mmcblk0p1                                129M     25M  104M   20% /boot
curlftpfs#ftp://user:password@192.168.0.100/  954G       0  954G    0% /mnt/ftp_client
```

curlftpfs is my  mounted NAS Device (1TB)  without a Quota.  About 90 GB free space currently.

Fourdee, 2016-02-17:
@k-plan 
Thanks for the info.

The issue is caused by the following:
- ```/dev/sdb1``` appears to be your root filesystem path.
- DietPi expects the root filesystem path to be the second partition on ```/dev/mmcblk0 (/dev/mmcblk0p2)```
- As your root path is different from the expected, its failing to pull in the correct data, so it cant work out the current size, and hence break the script.

Just need a few more informations if you can please:
- Are you using an SD card or EMMC?
- Also, could you please attach/paste the following file ```/etc/fstab```

k-plan, 2016-02-18:
Hi Fourdee,

> /dev/sdb1 appears to be your root filesystem path.

This is correct. Device sdb1 is root filesystem path. It's a USB Stick / USB HDD.

> Are you using an SD card or EMMC?

oDroide C1+ is booting from a SD-Card ( /boot ).
root path is on USB Stick / USB HDD ( / ) - [http://www.czc.cz/teac-usb-hdd-stick-8gb/50290/produkt](url)

Desired information:

```
root@oDroide-C1:~# cat /etc/fstab
#Internal Drives---------------------------------------------------
proc            /proc           proc    defaults          0       0
/dev/mmcblk0p1  /boot           vfat    defaults,noatime  0       2
UUID=0677b3d1-9217-435b-8cde-4a0dfb9a0695        /               ext4    defaults,noatime,nodiratime     0       1
tmpfs                   /tmp                    tmpfs   defaults,noatime,nodev,nosuid,mode=1777  0 0
tmpfs                   /var/log                tmpfs   defaults,size=20m,noatime,nodev,nosuid,mode=1777  0 0
tmpfs                   /DietPi                 tmpfs   defaults,size=10m,noatime,nodev,nosuid,mode=1777  0 0

#External Drives---------------------------------------------------
# - Try and use only ext4 for USB drives
# - Faster performance than NTFS, espically on RPi v1
#/dev/sda1       /mnt/usb_1      ext4    defaults,noatime,nofail  0       0
#/dev/sda1       /mnt/usb_1      ntfs-3g    defaults,noatime,nofail       0       0

#Samba Client------------------------------------------------------
#/mnt/samba . Please use dietpi-config and the networking menu to setup this mount

#FTP Client Mount--------------------------------------------------
curlftpfs#ftp://user:password@192.168.0.100 /mnt/ftp_client fuse auto,allow_other,direct_io,transform_symlinks,user,uid=1000,nonempty,_netdev 0  0
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
```

```
root@oDroide-C1:~# blkid
/dev/mmcblk0: PTUUID="000cde21" PTTYPE="dos"
/dev/mmcblk0p1: SEC_TYPE="msdos" LABEL="BOOT" UUID="6E35-5356" TYPE="vfat" PARTUUID="000cde21-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="e139ce78-9841-40fe-8823-96a304a09859" TYPE="ext4" PARTUUID="000cde21-02"
/dev/sda1: LABEL="USB_8GB" UUID="65019e81-2d0b-0130-9bf7-0527ad7fbaa0" TYPE="ext4"
/dev/sdb1: UUID="0677b3d1-9217-435b-8cde-4a0dfb9a0695" TYPE="ext4" PARTUUID="17497643-01
```

```
root@oDroide-C1:~# fdisk -l

Disk /dev/mmcblk0: 29,3 GiB, 31440502784 bytes, 61407232 sectors
Disklabel type: dos

Device         Boot  Start      End  Sectors   Size Id Type
/dev/mmcblk0p1        3072   266239   263168 128,5M  6 FAT16
/dev/mmcblk0p2      266240 61407231 61140992  29,2G 83 Linux

Disk /dev/sda: 7,5 GiB, 8002732032 bytes, 15630336 sectors
Disklabel type: dos

Device     Boot Start      End  Sectors  Size Id Type
/dev/sda1           2 15630335 15630334  7,5G 83 Linux

Disk /dev/sdb: 7,5 GiB, 8002732032 bytes, 15630336 sectors
Disklabel type: dos

Device     Boot Start     End Sectors  Size Id Type
/dev/sdb1        2048 6630335 6628288  3,2G 83 Linux
```

Fourdee, 2016-02-18:
@k-plan 
Thanks for the info.

It looks like we can use ```df  grep '/$'``` to find the root filesystem, regardless of where its located:
```
root@DietPi-XU4:~# df -h | grep '/$'
/dev/mmcblk0p2   15G  1.1G   14G   8% /
```

I've updated the ```dietpi-cleaner``` script in the testing branch. If you would be able to test the updated code, i'd be grateful:
```
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/dietpi-cleaner -O /DietPi/dietpi/dietpi-cleaner
chmod +x /DietPi/dietpi/dietpi-cleaner
```

Then run a cleaner test as per previous and let me know if the error is resolved.

k-plan, 2016-02-19:
Hi Fourdee,

```
root@oDroide-C1:~# df | grep '/$'
/dev/sdb1        3196552  715944   2298520   24% /
```

Replaced org. dietpi-cleaner script with dietpi-cleaner script in the testing branch.

Start dietpi-cleaner test-run:

Result:
[image]

I dare say succsess. Thanks for your help.

Fourdee, 2016-02-19:
@k-plan 
Excellent! Thank you for your assistance, reporting the issue and running those tests :+1:

Fourdee, 2016-02-19:
Patched for next release (v110). 
Please reopen if required.

closed 2016-02-19 by Fourdee