MichaIng/DietPi #202: RPi | Realtime Kernel (Fully preemptible RT) installation option.

Fourdee, 2016-02-19:
Install RT kernel:
```
wget http://dietpi.com/downloads/binaries/rpi/DietPi-RPi-4.1.17-rt17.zip -O package.zip
unzip -o package.zip -d /
rm package.zip
```

Fourdee, 2016-02-19:
Add Kernel "hotswap" option to ```DietPi-Config```
+ Ultra lightweight kernel

k-plan, 2016-02-20:
Hi Dan,
will this option also be available on  Odroid-C1+ ?

As far I can see, it exist also a full preemptive kernel 3.10.75-rtxx on Odroid-C1 :
[https://blog.georgmill.de/2015/09/03/realtime-preemtive-kernel-sources-3-10-80-rt88-for-odroid-c1-on-github/](url)
[https://blog.georgmill.de/2015/05/27/full-preemptive-kernel-3-10-75-rt80-on-odroid-c1/](url)

Fourdee, 2016-02-20:
@k-plan 
I wasn't planning on it as the Odroid kernel we use is built and maintained by Meveric via apt-get upgrade. 

I've not compiled any odroid kernels yet, so I will do some research and look through the links you provided :+1: .

k-plan, 2016-02-20:
Hi Dan,
I personally see this at an outside estimate option for Kernel "hotswap" to DietPi-Config, not more at the moment.
I don't really need it, but notice in Software selection on DietPi, one main focus is audio/multimedia. So for this guys, it may be useful. Only suggest this option, because I do some research, in which case a full preemptive RT kernel is needed. Didn't know this possibility before you post it here. But i notice, it is  on Milestone v110 for now.

Prefer to see same bugfixes, better localization support and more productive Software as Cloudshell, Pi-Hole, ...

Fourdee, 2016-02-20:
> Prefer to see same bugfixes, better localization support and more productive Software as Cloudshell, Pi-Hole, ...

If you can link the Git tickets that contain the items you are refering to, I will look into their priority. If the ticket does not exist, please create one.

Fourdee, 2016-02-23:
Pushed back menu integration to focus on bug fixes for v110.

rhkean, 2016-04-04:
I compiled a 4.2 kernel for my XU4 this weekend, but it still needs some work.  

- The kernel doesn't report the hardware like meverick's 3.10 kernel does (affects dietpi's ability to detect the correct hardware) 
- the base configuration didn't include any cpu governors except 'performance'
- uas module (the reason I experimented in the 1st place) was not enabled
- kernel/initrd/uInitrd installation wasn't as smooth as it could be (and if you don't properly configure the boot.ini file, you better have another boot image available to fix it)

all in all, it booted and I got to see my 12TB of usb drives running on an SBC with UsbAttachedScsi speeds, but I didn't get the feeling that it was production ready yet.  (I also didn't see a C2 branch on tobetter's github page:  https://github.com/tobetter/linux )

Fourdee, 2016-04-05:
@rhkean 
Hardkernel has a 4.2-rc1 kernel for the XU4, is this the one you tried? https://github.com/hardkernel/linux/tree/odroidxu4-v4.2-rc1

I must admit, thats one benefit of the RPi, they are always developing for the latest kernel. Although, they probably have 20+ staff members working on it, and a bigger comminuty.

rhkean, 2016-04-05:
No, I haven't tried that one yet.  I used the one referenced on [Odroid dokuwiki](http://odroid.com/dokuwiki/doku.php?id=en:odroid-xu4)

it's found at [tobetter's github repository](https://github.com/tobetter/linux)

I found the hardkernel repository late last night.  The last check-in date on both repos are within a day of each other, so I'm betting they're pretty similar, but I'm going to do a file compare on them and try the hardkernel one today or tomorrow.  (preliminary scan shows that the hardkernel branch is also missing the odroid files in the src/linux/arch/arm/mach-exynos folder, though. (That's where the /proc/cpuinfo data comes from)

rhkean, 2016-04-05:
It's worth spending the time on, though...

Did you see my post under [General Discussion on Fuzon](http://fuzon.co.uk/phpbb/viewtopic.php?f=9&t=357)?

Fourdee, 2016-04-19:
@k-plan @rhkean 
Just doing some late spring cleaning. As the original item in this ticket was completed in Feb, will mark this as closed.

Rob, if you want to explore the 4.x kernel for XU4 with uas support. Please create a new ticket for this :+1: 

Unless we get asked by our users for a RT kernel for Odroid devices, there isn't much pointing doing it. Mainly due to the constant Odroid kernel updates on a daily/weekly basis that would render the RT kernel out of date, very quickly.

closed 2016-04-19 by Fourdee

rhkean, 2016-04-19:
I agree with your analysis.