MichaIng/DietPi #203: DietPi-Software | dietpi-cloudshell -  Scene Selection - Network Usage, CPU, Storage, Network Details (only locales change)

k-plan, 2016-02-20:
Hi Fourdee,
tested on on Odroid C1+ (Jessie: v.109) (locales: en_GB.UTF-8) (root filesystem is on USB device) :

- Install (apt-get install iperf) and start iperf (server) `root@oDroide-C1:~# iperf -s -f m -i 10`

- Start dietpi-cloudshell on Odroid.

- Start iperf (client) on a local network connected PC: `admin@PC$ iperf -c 172.16.10.55 -f m -i 2 -d -t 60`

- Do same more iperf speed test and watch output on dietpi-cloudshell 

Result, if more traffic when 9999 MB is generated:
[image]
 Is dietpi-cloudshell not able to print traffic in GB?

______________

- While iperf is running, watch Scene : CPU and `root@oDroide-C1:~# htop`

Result:
[image]
[image]
How come margin?

_____________

- Watch Scene : Storage / Memory:

Result:
[image]
[image]

root filesystem is on USB device ([https://github.com/Fourdee/DietPi/issues/200](url))

_______________

- Now change setting locales: en_GB.UTF-8 to locales: de_DE.UTF-8

-  Watch Scene : Network Details:

Result:

[image]

_______________

Be in doubt, if it is better to split this issue.
Thanks for your assistance.

cu
k-plan

Fourdee, 2016-02-21:
> [image]

Fourdee, 2016-02-21:
####Network Scene

> [image]
The values are pulled in directly from the network adapters stats:
```
NETWORK_USAGE_TOTAL_SENT=$(( $(cat /sys/class/net/$NETWORK_DETAILS_ADAPTER/statistics/tx_bytes) / 1000000 ))
NETWORK_USAGE_TOTAL_RECIEVED=$(( $(cat /sys/class/net/$NETWORK_DETAILS_ADAPTER/statistics/rx_bytes) / 1000000 ))
```
It looks like the values stored /sys/ are being reset, when they reach a certain value.

Fourdee, 2016-02-21:
## CPU Scene

> [image]
> [image]

Looks like we are missing a few processes in the total calculations.
```
#Current code
root@DietPi-XU4:~# ps xo %cpu | sed '1d' | sed 's/ //' | wc -l
129

#New code
root@DietPi-XU4:~# ps axo %cpu | sed '1d' | sed 's/ //' | wc -l
134
```

Fourdee, 2016-02-21:
## Storage Scene

> [image]
> [image]

Due to alternative rootFS mount.

Fourdee, 2016-02-21:
## Network Details Scene

> [image]

Fourdee, 2016-02-21:
@k-plan 
Thanks for the reports. I've separated them into sections. I will update each section whilst working on them.

k-plan, 2016-02-21:
**@ Storage Scene**

> Due to no swap?

Use DietPi default  swapfile: 
```
root@oDroid-C1:~# dphys-swapfile setup
want /var/swap=100MByte, checking existing: keeping it

root@oDroid-C1:~# ls -la /var/swap
-rw------- 1 root root 104857600 Feb 14 23:04 /var/swap
```

Fourdee, 2016-02-21:
@k-plan 

> Use DietPi default swapfile:

```
free -m -o

#total swapfile size
free -m -o  | grep -m1 'Swap: ' | awk '{print $2}'
```
0 = disabled.

k-plan, 2016-02-21:
@Fourdee 
```
root@oDroid-C1:~# free -m -o
             total       used       free     shared    buffers     cached
Mem:           948        146        802          6          6         34
Swap:            0          0          0

root@oDroid-C1:~# free -m -o  | grep -m1 'Swap: ' | awk '{print $2}'
0
```
Hmm, generating a new swapfile via dietpi-config:

```
root@oDroid-C1:~# dietpi-config

DietPi: Deleting existing swapfile
 - /var/swap

DietPi: Generating new swapfile
 - 105 MB
 - /var/swap
please wait...

want /var/swap=105MByte, generating swapfile ...

root@oDroid-C1:~# free -m -o
             total       used       free     shared    buffers     cached
Mem:           948        255        693          6          6        140
Swap:          104          0        104

root@oDroid-C1:~# free -m -o  | grep -m1 'Swap: ' | awk '{print $2}'
104
```

Hmm, let's do a reboot ...

```
root@oDroid-C1:~#reboot

root@oDroid-C1:~# free -m -o  | grep -m1 'Swap: ' | awk '{print $2}'
104

```

:0)  :+1:

Fourdee, 2016-02-21:
@k-plan 

> Hmm, display Memory RAM [BAR] doesn't match any more.

Can you run this command whilst in de_DE:
```
MEMORY_USED=105;MEMORY_TOTAL=948;echo -e "$(echo | awk "{print $MEMORY_USED / $MEMORY_TOTAL * 100}")"
```

Should be ```11.0759```

k-plan, 2016-02-21:
@Fourdee 

> Can you run this command whilst in de_DE:

```
root@oDroid-C1:~# MEMORY_USED=105;MEMORY_TOTAL=948;echo -e "$(echo | awk "{print $MEMORY_USED / $MEMORY_TOTAL * 100}")"
11.0759
```

Fourdee, 2016-02-21:
@k-plan 
Apologies, it would help if you had the updated script lol :)

Make sure the Memory usage scene is enabled, then run:
```
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/dietpi-cloudshell -O /DietPi/dietpi/dietpi-cloudshell
killall dietpi-cloudshell
dietpi-cloudshell 2
```

k-plan, 2016-02-21:
@Fourdee 

> Make sure the Memory usage scene is enabled, then run:

Done.

[image]

Fourdee, 2016-02-21:
## Memory Usage Scene
>[image]

DietPi-Cloudshell will now check for a active swap, before running calculations: https://github.com/Fourdee/DietPi/commit/2e8799066f8aaa0ef955cdbbea18ecb533853ed7

0% Ram graph is due to de_DE.UTF-8 breaking printf when using a float input:
```
#en_GB float
root@DietPi:~# MEMORY_PERCENT=11.12345;printf %0.f $MEMORY_PERCENT
11

#de_DE float
root@DietPi-XU4:~# MEMORY_PERCENT=11.12345;printf %0.f $MEMORY_PERCENT
-bash: printf: 11.0005352: Ungültige Zahl.

#de_DE int
root@DietPi-XU4:~# MEMORY_PERCENT=11;printf %0.f $MEMORY_PERCENT
11
```
Not sure how to resolve the above. Are printf command line options different in other locales?

k-plan, 2016-02-21:
@Fourdee 

> Not sure how to resolve the above. Are printf command line options different in other locales?

No, not printf command line options cause the problem, it's the decimal delimiter symbol for numbers.

locales: en_GB.UTF-8  =>>  " . "
 
locales: de_DE.UTF-8 =>>  " , "

```
root@oDroid-C1:~# MEMORY_PERCENT=11.12345;printf %0.f $MEMORY_PERCENT
-bash: printf: 11.12345: Ungültige Zahl.
```

```
root@oDroid-C1:~# MEMORY_PERCENT=11,12345;printf %0.f $MEMORY_PERCENT
11
```

k-plan, 2016-02-21:
###  Network Details Scene

[image]

@Fourdee 
> Most likely due to NETWORK_DETAILS_IP_INT=$(cat "$FP_TEMP" | grep -m1 'inet addr:'| cut -d: -f2 | awk '{ print $1}') expected value inet addr changes when in de locale.

> We can grab MAC with the following:
> `cat /sys/class/net/eth0/address`

Odroid C1+ use eth1 instead of eth0:
```
root@oDroid-C1:~# cat /sys/class/net/eth1/address
00:1e:06:30:07:2b
```

> We can find ip address using following ("inet " is same regardless of locale):
> `ifconfig eth0 | grep -m1 'inet '| cut -d: -f2 | awk '{ print $1}'`

```
root@oDroid-C1:~# ifconfig eth1 | grep -m1 'inet '| cut -d: -f2 | awk '{ print $1}'
192.168.0.111
```

If I can help you with another testing, let me know.

Fourdee, 2016-02-21:
@k-plan 

> No, not printf command line options cause the problem, it's the decimal delimiter symbol for numbers.
locales: en_GB.UTF-8 =>> " . "
locales: de_DE.UTF-8 =>> " , "

Interesting, but also really bizarre. I've never seen ```,``` being used as a decimal delimiter in any coding language.

Ok, so this will also affect all the ```Percent_To_Graph $(printf %.0f $MEMORY_SWAPERCENT)``` functions we use for the graphs. That is, if the input is a float.

I suppose we could try echoing the float and using cut to remove any values after and including the decimal place. This works on both en_GB and de_DE:
```
echo 12.1230 | cut -d. -f1
```
Means I'll need to change all references to the function, but it will resolve the current language issue.

k-plan, 2016-02-21:
@Fourdee 

> Interesting, but also really bizarre. I've never seen ` , `being used as a decimal delimiter in any coding language.

Correct, nothing to do with coding language, its a problem of numeral system (Decimal mark).

[](https://en.wikipedia.org/wiki/Decimal_mark#Influence_of_calculators_and_computers)

Didn't help, but it's not your fault.

Fourdee, 2016-02-21:
@k-plan 

> Correct, nothing to do with coding language, its a problem of numeral system (Decimal mark).
https://en.wikipedia.org/wiki/Decimal_mark#Influence_of_calculators_and_computers

Ah, thanks for the info. :+1: 

> Didn't help, but it's not your fault. Sorry, because you now have a lot of work.
Changing the function is a 5 minute job at most. Its just getting those 5 minutes to be available thats the issue lol.

I'll update you when the changes are completed. 
I will then provide you with the details to upgrade dietpi-cloudshell on your system. I would appreciate if you test and verify all the issues are fixed. I'll let you know when its ready.

Fourdee, 2016-02-22:
@k-plan 
I just need to resolve the network usage scene. ```Ifconfig``` RX/TX stats are also being reset, so that must pull data from ```/sys/class/net/$NETWORK_DETAILS_ADAPTER/statistics/tx_bytes```also.

I may need to use a 3rd party program that allows us to monitor network usage, without being reset.

k-plan, 2016-02-22:
@Fourdee 
Hmm, is [slurm](https://packages.debian.org/jessie/slurm) something you are looking for? 
And yes, Decimal mark is  a ` . `(dot) in slurm. ;)

[image]

`root@oDroid-C1:~# apt-get install slurm`

Speed is measured in `KB/s` not in `MBit/s`, a thing that bother me in slurm.
You also have a look at this [collection](http://www.cobin.de/binary.php) of usefull stuff.

Fourdee, 2016-02-22:
Tried slurm, nload and iftop.

None of these allow for single run, then exit. They must be run in the background, writing to a file that we would need to parse (not ideal).

netstat seems to be the best option, but we need to calculate MTU size (bytes) against total packets.
```
local mtu_size=$(netstat -N -i | grep "$NETWORK_DETAILS_ADAPTER" | awk '{print $2}')
NETWORK_USAGE_TOTAL_RECIEVED=$(( $(netstat -N -i | grep "$NETWORK_DETAILS_ADAPTER" | awk '{print $4}') * $mtu_size / 1024 / 1024 ))
NETWORK_USAGE_TOTAL_SENT=$(( $(netstat -N -i | grep "$NETWORK_DETAILS_ADAPTER" | awk '{print $8}') * $mtu_size / 1024 / 1024 ))
```
I've received over 20GB and the values have not reset :)

Fourdee, 2016-02-23:
@k-plan 
Ok, I think everything is now resolved, but I would like you to test on your system to be sure. I also added current network usage stats to the net usage scene.

Simply run the following code to update the scripts:
```
#Update network script
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/func/obtain_network_details -O /DietPi/dietpi/func/obtain_network_details
/DietPi/dietpi/func/obtain_network_details

#Update HW model script
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/dietpi-obtain_hw_model -O /DietPi/dietpi/dietpi-obtain_hw_model
/DietPi/dietpi/dietpi-obtain_hw_model

#Update DietPi-Cloudshell
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/dietpi-cloudshell -O /DietPi/dietpi/dietpi-cloudshell
```

Now you can run ```dietpi-cloudshell``` and "Start/Restart" to test the recent changes. Let me know if the issues are now resolved.

k-plan, 2016-02-24:
@Fourdee 

sorry for the delay, had to test v.5 a little bit.

> I've received over 20GB and the values have not reset :)

Hmm,  `(TOTAL)` works for me (without reboot), but  `(TODAY)` is reset on every launch of DietPi-Cloudshell . (_see screenshot below_). For me, it will be okay.

- If we choose in `Scenes` `Network Usage` without choosing `Network Details` at the same time, now starting DietPi-Cloudshell v.5, launch on terminal breaks and we are back in  dietpi-cloudshell menu.

- It's seem, we missing some CPU processes. DietPi-Cloudshell shows about 5-10% cpu load, htop shows in cpu (average) about 40%.
[image]
[image]

- I have notice, in same rar cases, if DietPi-Cloudshell launch on terminal, the prompt is back- coloured. Not sure, if it is wanted.
[image]

-  unesthetic: on first show of`Network Usage` Scene, there is a unplausible value in `(CURRENT)`. (Also a `c` in a additional line)
[image]
  Second run is okay. So it is only cosmetic.
  [image]

- Would be nice, if value in `(CURRENT)` will be switched automatically in MBit/s (if reasonable/posible) and
 value in `(TOTAL)` and `(TODAY)` will be switched automatically in MB / GB (if reasonable/posible).
[image]

Great work, you  be on the right track. Locales and doted dezimal is fix now. :+1: 
Love your v.5. Haven't test Pi-Hole so far.

Fourdee, 2016-02-25:
#### Network usage is now converted into human readable format. 
Also uses BC for float calculations.
[image]

#### > unesthetic: on first show ofNetwork Usage Scene, there is a unplausible value in (CURRENT).

Resolved by creating a init var which updates the current timestamp on first scene update, rather than first run of script.

#### CPU usage now uses BC for float calculations. 
I'am hoping this resolves the incorrect values you are receiving, if not, I will need to look into other methods of obtaining more accurate CPU usage stats.

#### > I have notice, in same rar cases, if DietPi-Cloudshell launch on terminal, the prompt is back- coloured. Not sure, if it is wanted.

```echo -e "$C_RESET"``` is now run before the prompt. This resets any active bash colours.

#### > If we choose in Scenes Network Usage without choosing Network Details at the same time, starting DietPi-Cloudshell v.5, the launch on terminal breaks and we are back in dietpi-cloudshell menu.

I was unable to replicate the above.

k-plan, 2016-02-25:
@Fourdee 

> I was unable to replicate the above.

- `root@oDroid-C1:~# dietpi-cloudshell`

-  choose wanted Scenes: for example `Network Usage: ` and  `CPU:` and hit `<OK>`
   [image]

-  choose `Start / Restart` and hit `<OK>`
  
   [image]

- hit `ENTER` key
   
   [image]
   [image]
    [image]
    [image]

Wihtout user interaction program breaks after introduction screen.

If you now start dietpi-cloudshell straight  again, the prompt will be bash coloured.
 [image]

Fourdee, 2016-02-25:
@k-plan 
Thanks for the info, however, i'am still unable to replicate :(
I've uploaded the new dietpi-cloudshell script. Could you please try with that:
```
#Update network script
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/func/obtain_network_details -O /DietPi/dietpi/func/obtain_network_details
/DietPi/dietpi/func/obtain_network_details

#Update HW model script
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/dietpi-obtain_hw_model -O /DietPi/dietpi/dietpi-obtain_hw_model
/DietPi/dietpi/dietpi-obtain_hw_model

#Update DietPi-Cloudshell
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/dietpi-cloudshell -O /DietPi/dietpi/dietpi-cloudshell
```

k-plan, 2016-02-25:
@Fourdee 
Done!

adjustments:
[image]

problem in menu display, but 1 sec intervall is cool ( :+1: )
[image]

Start - `Storage:` and `Memory:` Scene works fine, but:
[image]
[image]
[image]

---------------------------------------

Then I have done:
`root@oDroid-C1:~# apt-get install bc`
```
Die folgenden NEUEN Pakete werden installiert:
  bc
0 aktualisiert, 1 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
.....

bc (1.06.95-9) wird eingerichtet ...

```

- dietpi-cloudshell  >> Options : Update Rate : same display problem as before

- CPU load display is better (about 13% - htop cpu avg: about 40%)
  [image]

- in Network Usage (CURRENT) a different in value to iperf:
   
   [image]
  [image]

Fourdee, 2016-02-25:
@k-plan 
Thanks for reply.

#### > dietpi-cloudshell >> Options : Update Rate : same display problem as before

I assume you mean the formatting issue with characters for the scrolldown? 
Do you get this in other menus? Try ```dietpi-software``` , then select optimized or linux software.

#### > root@oDroid-C1:~# apt-get install bc

Yep my bad, I forgot to mention. This will be installed during the v110 update for all users: https://github.com/Fourdee/DietPi/blob/testing/CHANGELOG.txt#L11

#### > in Network Usage (CURRENT) a different in value to: root@oDroid-C1:~# iperf -s -f m -i 10

The current calculations are based on:
```packet count * MTU byte size = byte value``` If packets are fragmented, that would increase the results.

The program which provides this info, and is parsed by dietpi-cloudshell:
```
netstat -N -i
```

#### > CPU load display is better (about 13% - htop cpu avg: about 40%)

Bear in mind the ```dietpi-cloudshell``` cpu usage is based on current usage at the time the scene is updated. Its not an average.

k-plan, 2016-02-25:
@Fourdee 

> I assume you mean the formatting issue with characters for the scrolldown? 
yes.

> Do you get this in other menus? Try dietpi-software , then select optimized or linux software.
[image]

Double-Check this at home, with a different ssh client.

> Bear in mind the dietpi-cloudshell cpu usage is based on current usage at the time the scene is updated. Its not an average.
Yes. We have on Odroide C1+ four CPU. So which one? Yes, thats the problem. Use a RPi makes things easy. :) 
Will have at home a look at :100: CPU load.

> The current calculations are based on:

Will have a look on iperf. Will do some research,  if this software is using his own network stack.
Will do some test with ftp/NFS/Samba ...

Overall, great work Fourdee ! Verry nice :+1:

Fourdee, 2016-02-26:
@k-plan 

> Overall, great work Fourdee ! Verry nice :+1:
Deeply impressed

Many thanks to you for reporting the issues, testing, and providing feedback to help make dietpi-cloudshell better :) . :+1: 

I also fixed the missing leading zero's from ```bc``` in network usage scene:
[image]

Fourdee, 2016-02-26:
Will mark this as completed. Available in next release (v110).

Please reopen if required.

k-plan, 2016-02-26:
@Fourdee 

> The current calculations are based on:
> `packet count * MTU byte size = byte` value If packets are fragmented, that would increase the results.

`packet count * MTU byte size = byte` that's correct. 
BI values are calculated using Binary Units. (8 bit = 1 byte / 1024 byte = 1 kbyte)
Binary units  (binary prefixes) are prefered by operating systems. 

Network throughput speed / Bandwidth are calculated using the Internation System of Units (SI values )
Network throughput speed / Bandwidth calculated using SI Units in decimal prefixes. (1000 bit/s = 1 kbit/s / 1000 kbit/s = 1 Mbit/s) 
( Links: [1*](http://www.ibeast.com/content/tools/band-calc.asp) and  [2*](http://www.heise.de/netze/tools/bandbreitenrechner/) )
But MTU is in bytes, for example MTU=1500 bytes.

Take a look in your code ([line 781](https://github.com/Fourdee/DietPi/commit/f5159a3e3f1f49b2cf7c0461c0d14318c374a658#diff-9ccf5cb326d9008aaceab7f80ce4ffdbL730)): 

```
local now_sent_output=$( BYTE_PRINT_CONVERSION $NETWORK_USAGE_NOW_CURRENT_SENT )
local now_recieved_output=$( BYTE_PRINT_CONVERSION $NETWORK_USAGE_NOW_CURRENT_RECIEVED )
```
To get correct unit (Mbit/s) we needed additional:

```
	#Converts a bit into string in human readable format.
	## (for network data transmission rate (LAN, WLAN, ...)
	BIT_PRINT_CONVERSION(){

                local return_value=0
                local decimal_count=1

                #$1=bit value

                # - Kbit
                if (( $1 < 1048576 )); then
                        #return_value="$(( $1 * 8 / 1000 )) Kbit"
                        return_value="$(echo "scale=$decimal_count; $1 * 8 / 1000" | bc -l) Kbit"

                # - MBit
                elif (( $1 < 1073741824 )); then
                        #return_value="$(( $1 * 8 / 1000 / 1000 )) Mbit"
                        return_value="$(echo "scale=$decimal_count; $1  * 8 / 1000 / 1000" | bc -l) Mbit"

                # - GBit
                else
                        #return_value="$(( $1 * 8 / 1000 / 1000 / 1000 )) Gbit"
                        return_value="$(echo "scale=$decimal_count; $1 * 8 / 1000 / 1000 / 1000" | bc -l) Gbit"

                fi

                echo "$return_value"

	}

```

and have to change:

```
        #NETWORK USAGE
        Update_Scene_4(){

                #Update data
                Obtain_NETWORK_USAGE

                # - Convert usage values into human readable format. Run before clearing screen due to additional processing (delay)
                local total_sent_output=$( BYTE_PRINT_CONVERSION $NETWORK_USAGE_TOTAL_CURRENT_SENT )
                local total_recieved_output=$( BYTE_PRINT_CONVERSION $NETWORK_USAGE_TOTAL_CURRENT_RECIEVED )

                local today_sent_output=$( BYTE_PRINT_CONVERSION $NETWORK_USAGE_DAY_CURRENT_SENT )
                local today_recieved_output=$( BYTE_PRINT_CONVERSION $NETWORK_USAGE_DAY_CURRENT_RECIEVED )

                ##local now_sent_output=$( BYTE_PRINT_CONVERSION $NETWORK_USAGE_NOW_CURRENT_SENT )
                ##local now_recieved_output=$( BYTE_PRINT_CONVERSION $NETWORK_USAGE_NOW_CURRENT_RECIEVED )

                local now_sent_output=$( BIT_PRINT_CONVERSION $NETWORK_USAGE_NOW_CURRENT_SENT )
                local now_recieved_output=$( BIT_PRINT_CONVERSION $NETWORK_USAGE_NOW_CURRENT_RECIEVED )

```

[image]
[image]
[image]
[image]

So you can see, Incoming/Receved does now fit.
But Outgoing/Sent doesn't. Yes, this will be, because we calculated using fix MTU of 1500 bytes.
TCP Acknowledgment packets are containing clearly less bits. I have no solution for this at the moment.

--------

> I assume you mean the formatting issue with characters for the scrolldown? 

Checked, it is a ssh-Client display problem.

--------

> dietpi-cloudshell uses `ps -axo %cpu | sed '1d' | sed 's/ //'` to get a list of all processes cpu usage (%). As ps -axo %cpu values are relative to a per-core basis (eg: 1 process can show as 100%, but only 25% of a 4 core cpu is actually used), we add up the values and divided by cpu core count, to get overall usage across all cores.

Hmm, test it. Can not tell what is going on with `ps`

`htop`
cpu0 = 2%
cpu1 = 100%
cpu2 = 100%
cpu3 = 0%
= = = = = =
calculated load (by me): 202% / 4 = 50,5 %

cloudshell print: 6%

if I added all printed values of `ps -axo %cpu | sed '1d' | sed 's/ //'` = 18 %

k-plan, 2016-02-27:
> Closed.#203

Hi Dan,

have added some more informations after you closed  #203 .

Best Regards

k-plan

Fourdee, 2016-02-27:
@k-plan 

#### > To get correct unit (Mbit/s) we needed additional:

I've updated DietPi-Cloudshell to print current usage in bits, as per your recommendation and code (nice one!) :+1: 

#### > But Outgoing/Sent doesn't. Yes, this will be, because we calculated using fix MTU of 1500 bytes.
TCP Acknowledgment packets are containing clearly less bits. I have no solution for this at the moment.

Yep, your right, iam unsure also. Ideally, if the kernel wasn't clearing the stats ```/sys/class/net/$NETWORK_DETAILS_ADAPTER/statistics/tx_bytes``` we could parse those values for better accuracy (as it was previously).  I'll do some more digging to see if there is a way to prevent the kernel resetting those values, without having to compile a new kernel. 

> calculated load (by me): 202% / 4 = 50,5 %
cloudshell print out: 6%
if I added all printed values of ps -axo %cpu | sed '1d' | sed 's/ //' = 18 %

Not sure whats going on there. I'll run some tests on my C1 to see if my values have similar large differences.
If ```ps -axo %cpu``` is the cause of inconsistent results, i'll try parsing ```top``` instead. Either way, I want to get these values a little closer to ```htop```'s results.

#### If you want to test updated script:
```
#Update network script
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/func/obtain_network_details -O /DietPi/dietpi/func/obtain_network_details
/DietPi/dietpi/func/obtain_network_details

#Update HW model script
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/dietpi-obtain_hw_model -O /DietPi/dietpi/dietpi-obtain_hw_model
/DietPi/dietpi/dietpi-obtain_hw_model

#Update DietPi-Cloudshell
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/dietpi-cloudshell -O /DietPi/dietpi/dietpi-cloudshell
```

k-plan, 2016-02-27:
@Fourdee 

Tested, done. :+1: 

Yes, accuracy from `Network Usage` and `CPU`  leaves a lot to be desired. 
It is no measuring, it is parsing and calculation (by design). Sometimes, at high load, also a problem with timing and process parallelism.
More a guesstimate, but it's okay for now.

In german: " Hauptsache, die Richtung stimmt! " _[main point, to be on right way]_ :)

Fourdee, 2016-02-27:
@k-plan 

In reference to ```ps -axo %cpu```
http://www.sbras.ru/cgi-bin/www/unix_help/unix-man?ps+1
```
%cpu    The cpu utilization of the process; this is a decaying average
	     over up to a minute of previous (real) time.  Since the time base
	     over which this is computed varies (since processes may be very
	     young) it is possible for the sum of all %CPU fields to exceed
	     100%.
```
So if I read that right, its an average upto a minute of previous, not current usage.

> Yes, accuracy from Network Usage and CPU leaves a lot to be desired. 
It is no measuring, it is parsing and calculation (by design). Sometimes, at high load, also a problem with timing and process parallelism.
More a guesstimate, but it's okay for now.
It's work definitely better than before.

Yep, completely agree. I will continue working on improving the script. Will let you know if I come up with any good results.

Fourdee, 2016-02-27:
Now i remember why I didnt use ```top``` for cpu usage. It fails to print values in low screen resolutions:
[image]

```
echo "scale=1; 100 - $(top -b -n 1 | grep '%Cpu(s):' | awk '{print $8}')" | bc -l
```

closed 2016-02-29 by Fourdee