MichaIng/DietPi #204: Raspberry Pi 2 | Pydio | Impossible write into the AJXP_DATA_PATH folder

CedArctic, 2016-02-21:
Same issue that was reported here: http://fuzon.co.uk/phpbb/viewtopic.php?f=11&t=303#p1304
Still exists on the Raspberry Pi 2 image v109.

Fourdee, 2016-02-21:
@CedArctic 
Thanks for the report. I'll run a fresh RPIv2 install and test.

CedArctic, 2016-02-21:
Wow that was fast, thanks Fourdee!

Fourdee, 2016-02-21:
@CedArctic 

A fresh installation with dedicated USB drive worked fine.

Are you by any chance using a SD only installation (eg: you skipped the dedicated USB drive question)?

Fourdee, 2016-02-21:
I found an issue in the code for the SD installation of Pydio. So i'd imagine you are running SD card only. If you can confirm your running SD only installation, or not, i would be greatful.

CedArctic, 2016-02-22:
I am running dietpi off an SD but I answered yes to the prompt about using a HDD in the initial start up.  My HDD is formated as NTFS if it makes a difference.

Fourdee, 2016-02-22:
@CedArctic 

NTFS should be fine. We will check the permissions, just to be sure.

I'll need some extra information on your system, could you please run the following commands and paste the results for each:
```
ls -lha /mnt/usb_1
ls -lha /mnt/usb_1/pydio_data
ls -lha /var/www/pydio
```

CedArctic, 2016-02-22:
ls -lha /mnt/usb_1

root@DietPi:~# ls -lha /mnt/usb_1
total 528M
d--------- 1 root     root     8.0K Feb 21 15:38 .
drwxr-xr-x 5 root     root     4.0K May 12  2015 ..
drwxrwxrwx 1 root     root        0 Sep 29  2013 Acer Backup
drwxrwxrwx 1 root     root        0 Jul 16  2015 Alienware Backup
drwxrwxrwx 1 root     root        0 Jul 30  2015 Alienware Trainwreck Backup
drwxrwxrwx 1 root     root      12K Jun 27  2015 Assassin's Creed IV - Black Flag
drwxrwxrwx 1 root     root        0 Aug 16  2012 autorun
-rwxrwxrwx 1 root     root      124 Aug 29  2011 autorun.inf
drwx------ 1 root     root        0 Jul 16  2015 d8521b0763e05772c4
-rwxrwxrwx 2 root     root     9.2K Jul  2  2012 Declaration of Conformity.pdf
drwxrwxrwx 1 root     root        0 Apr  1  2014 Desktop Backup
drwxrwxrwx 1 root     root        0 Feb 21 14:40 downloads
drwxrwxrwx 1 root     root     4.0K Nov 13 08:59 Jimis Backup
drwxrwxrwx 1 root     root        0 Jul 31  2015 msdownld.tmp
drwxrwxrwx 1 root     root      60K Feb 21 14:40 Music
drwxrwxrwx 1 root     root        0 Aug 16  2012 NTI
drwxrwxrwx 1 root     root        0 Aug 16  2012 OpenSourceSoftwareInfo
drwxrwxrwx 1 root     root        0 Feb 21 15:11 owncloud
drwxrwx--- 1 root     root     4.0K Feb 21 15:00 owncloud_data
drwxrwxrwx 1 root     root        0 Feb 21 12:30 Pi
drwxrwxrwx 1 root     root        0 Feb 21 14:40 Pictures
drwxrwxrwx 1 root     root     4.0K Feb 14 08:02 PressPi
drwxrwxrwx 1 www-data www-data 4.0K Feb 21 15:38 pydio_data
drwxrwxrwx 1 root     root     4.0K Nov 13 08:46 $RECYCLE.BIN
-rwxrwxrwx 1 root     root      15M May 24  2012 Setup.exe
-rwxrwxrwx 1 root     root     512M Feb 21 14:11 .swapfile
drwx------ 1 root     root     4.0K Jul 30  2015 System Volume Information
drwxrwxrwx 1 root     root        0 Feb 21 14:40 Video
-rwxrwxrwx 1 root     root      75K Apr 23  2012 Warranty.pdf
d--------- 1 root     root     4.0K Jul 30  2015 Windows Backup

CedArctic, 2016-02-22:
ls -lha /mnt/usb_1/pydio_data

root@DietPi:~# ls -lha /mnt/usb_1/pydio_data
total 12K
drwxrwxrwx 1 www-data www-data 4.0K Feb 21 15:38 .
d--------- 1 root     root     8.0K Feb 21 15:38 ..
drwxrwxrwx 1 www-data www-data    0 Jan 19 11:52 cache
drwxrwxrwx 1 www-data www-data    0 Jan 19 11:52 files
drwxrwxrwx 1 www-data www-data    0 Jan 19 11:52 logs
drwxrwxrwx 1 www-data www-data    0 Jan 19 11:52 personal
drwxrwxrwx 1 www-data www-data    0 Jan 19 11:52 plugins
drwxrwxrwx 1 www-data www-data    0 Jan 19 11:52 public
drwxrwxrwx 1 www-data www-data    0 Feb 21 15:44 tmp

CedArctic, 2016-02-22:
ls -lha /var/www/pydio

root@DietPi:~# ls -lha /var/www/pydio
total 88K
drwxr-xr-x   6 www-data www-data 4.0K Feb 21 15:56 .
drwxr-xr-x   4 www-data www-data 4.0K Feb 21 15:54 ..
-rw-r--r--   1 www-data www-data 1.2K Jan 19 11:48 base.conf.php
-rw-r--r--   1 www-data www-data 9.8K Jan 19 11:48 cmd.php
drwxr-xr-x   3 www-data www-data 4.0K Jan 19 11:52 conf
-rw-r--r--   1 www-data www-data  964 Jan 19 11:48 content.php
drwxr-xr-x   5 www-data www-data 4.0K Jan 19 11:52 core
lrwxrwxrwx   1 www-data www-data   21 Feb 21 15:56 data -> /mnt/usb_1/pydio_data
-rw-r--r--   1 www-data www-data 4.6K Jan 19 11:48 dav.php
-rw-r--r--   1 www-data www-data 1.2K Jan 19 11:48 .htaccess
-rw-r--r--   1 www-data www-data 7.3K Jan 19 11:48 index.php
-rw-r--r--   1 www-data www-data 1014 Jan 19 11:48 index_shared.php
drwxr-xr-x   3 www-data www-data 4.0K Jan 19 11:52 phpunit
drwxr-xr-x 134 www-data www-data 4.0K Jan 19 11:52 plugins
-rw-r--r--   1 www-data www-data 1.6K Jan 19 11:48 publicLet.inc.php
-rw-r--r--   1 www-data www-data 3.2K Jan 19 11:48 rest.php
-rw-r--r--   1 www-data www-data 9.4K Jan 19 11:48 runTests.php

Fourdee, 2016-02-22:
@CedArctic 
Thanks. 

The permissions look fine, but just to be sure, lets check the /var/www folder aswell:
```
ls -lha /var/www
```

> Impossible write into the AJXP_DATA_PATH

Does this occur straight away when you access http://ip/pydio? Or were you able to access the configuration page before hand?

CedArctic, 2016-02-23:
It appears straight away.

CedArctic, 2016-02-23:
ls -lha /var/www

root@DietPi:~# ls -lha /var/www
total 88K
drwxr-xr-x  4 www-data www-data 4.0K Feb 21 15:54 .
drwxr-xr-x 15 root     root     4.0K Feb 21 18:11 ..
-rw-r--r--  1 www-data www-data  38K Feb 21 15:56 apc.php
-rw-r--r--  1 www-data www-data   35 Feb 21 14:40 index.html
-rw-r--r--  1 www-data www-data  23K Feb 21 15:56 opcache.php
-rw-r--r--  1 www-data www-data   20 Feb 21 15:56 phpinfo.php
drwxrwxr-x  5 www-data www-data 4.0K Feb 21 14:10 pihole
drwxr-xr-x  6 www-data www-data 4.0K Feb 21 15:56 pydio

Fourdee, 2016-02-23:
@CedArctic 
Thanks.

Just setting up a RPiv2 with NTFS usb drive testbed now to match your installation. Lets see what happens :)

Fourdee, 2016-02-23:
@CedArctic 
I did a fresh v109 image write and install of Pydio on a NTFS drive, using a RPiv2. Pydio is working fine.

I'll need to see your system settings. Please can you send a bug report: ```dietpi-bugreport```. No need to fill out the information. I only need the reference code that's generated.

CedArctic, 2016-02-24:
Hmm that sounds weird, since I installed pydio on the first start up and hadn't modified DietPi in any way.
Here's the code: b827eb3edd21-0

Fourdee, 2016-02-24:
@CedArctic 

Thanks. I've checked all your settings, and you are right, no modifications have been made. Everything seems to be in order.

I'am leaning towards the NTFS drive as the possible cause of the issue. Whilst checking my installation, i noticed that my NTFS drive has ```root:root``` permissions all the way through.

I know from past experience that you need to take ownership of NTFS drives for complete access/control. I'am curious to see if that is the cause.

Lets try setting root:root permissions on your USB drive:
```
chown -R root:root /mnt/usb_1
```

Now check ```root:root``` permissions have been applied to ```pydio_data```:
```
ls -lha /mnt/usb_1
```

If Pydio is still playing up, you may need to try other NTFS-3g mounting options in ```/etc/fstab```(http://www.tuxera.com/community/ntfs-3g-manual/). Either that or start again with EXT4 format during 1st run installation.

CedArctic, 2016-02-24:
Still the same issue strangely :/

Fourdee, 2016-02-24:
@CedArctic 
Hummm, this ones proving to be persistent lol.

As I have been unable to replicate the issue with a NTFS USB drive installation myself, i'am not sure what else we can do.

The only thing I can suggest is rewriting the DietPi image and choosing to format your USB drive to EXT4. I know its not ideal, but I tried this on v109 and Pydio worked without fault.

CedArctic, 2016-02-24:
Hmm that's 1TB of storage though :/ . Any way I can switch pydio to run off the SD?

Fourdee, 2016-02-25:
@CedArctic 

> Any way I can switch pydio to run off the SD?

Yes, but bear in mind any data uploaded/stored on Pydio will be located on your SD card (```/var/www/pydio/data``` I believe):
- ```dietpi-uninstall``` and remove PYDIO
- ```nano /DietPi/dietpi/.installed``` and change ```USBDRIVE 2``` to ```USBDRIVE 0```
- Reinstall Pydio in ```dietpi-software```

CedArctic, 2016-02-25:
Hmm but wouldn't that change the default disk for all applications?

CedArctic, 2016-02-25:
An additional question: Using the ext4 format won't cause issues with reading and writing through the samba server on a windows host?

Fourdee, 2016-02-25:
> An additional question: Using the ext4 format won't cause issues with reading and writing through the samba server on a windows host?

No issue. Samba server is compatible with EXT4.

> Hmm but wouldn't that change the default disk for all applications?

Only for new applications that you install with ```dietpi-software```. You can set ```USB_DRIVE 2``` once pydio is installed to switch back to USB install.

CedArctic, 2016-02-28:
I recently had to switch my SD cards and when I started my dietpi installation, the issue was gone oddly. Thanks for your support :) . I just have one last question, how do I go port forwarding with this so I can use it outside my LAN? Does it run on port 80? Do I have to change the Apache port?

Fourdee, 2016-02-28:
@CedArctic 

> I recently had to switch my SD cards and when I started my dietpi installation, the issue was gone oddly. 

Very strange, but good to hear its working. Could possibly be a corrupt SD card?

> UPDATE: Never mind sorted it out :P. Thanks once again for all this :)

Good stuff :)

I'll mark this as resolved and close the ticket. If the issue reoccurs, please reopen.

closed 2016-02-28 by Fourdee