MichaIng/DietPi #206: Single File PHP Image Gallery download link broken

rhkean, 2016-02-26:
Single File PHP Image Gallery has been rev'd from version 4.5.7 to 4.6.1 and the old version is no longer available for download at the coded link in dietpi-software (line 1487 of v109).

The new link is:  http://sye.dk/sfpg/Single_File_PHP_Gallery_4.6.1.zip

I've tested the script with the updated link and all works the same.

Fourdee, 2016-02-26:
@rhkean 
Great spot :+1: 
I'll mirror the 4.6.1 download on dietpi.com and update ```dietpi-software``` to link to it. This will prevent this occurring in the future.

Fourdee, 2016-02-26:
Now hosted on dietpi.com:
https://github.com/Fourdee/DietPi/commit/c58301736c31b1ab937a2a7f92ecd793587e1a57#diff-d92a6ee04e02fd2a2dc23d5bec3e6a98R1487

@rhkean 
https://github.com/Fourdee/DietPi/commit/c58301736c31b1ab937a2a7f92ecd793587e1a57#diff-a0493e27f48b50e18312b9f4508fc29dR17
:+1:

Fourdee, 2016-02-26:
Completed. Available in next release (v110)

closed 2016-02-26 by Fourdee