MichaIng/DietPi #207: DietPi-Software | Add Java installation option

starstuff, 2016-02-26:
Please add Java in Dietpi software because java is needed to run some apps. 

[](https://wiki.debian.org/Java/)

Fourdee, 2016-02-26:
@starstuff 

Thanks for the suggestion. In the mean time, instructions to install Oracle Java 8 is listed below.

On the Raspberry Pi, you can install Java with this command:
```
apt-get install oracle-java8-jdk
```

For non-rpi systems on Jessie:
```
echo "deb http://ppa.launchpad.net/webupd8team/java/ubuntu trusty main" > /etc/apt/sources.list.d/webupd8team-java.list
echo "deb-src http://ppa.launchpad.net/webupd8team/java/ubuntu trusty main" > /etc/apt/sources.list.d/webupd8team-java.list
apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys EEA14886
/DietPi/dietpi/dietpi-apt-get_update 1
apt-get update
apt-get install oracle-java8-installer
```

Fourdee, 2016-02-28:
note to self:
remove java installation code for ```SUBSONIC``` and set ```ORACLEJAVA=1```.

Fourdee, 2016-03-03:
Completed, v111

closed 2016-03-03 by Fourdee