MichaIng/DietPi #208: DietPi-Software | Add JDownloader installation option

starstuff, 2016-02-26:
Please add JDownloader in DietPi, Its an open-source download manager application that is easy to use and can even download youtube links. I use it to download videos i like on youtube. 

http://beta.jdownloader.org/ 

also, you can control JDownloader remotely using myjdownloader 

http://my.jdownloader.org/

ori64638, 2016-03-28:
+1 Very useful software

Fourdee, 2016-03-28:
Did a quick test with their install script, pre-reqs JVM. Will look into this further.
```
wget http://installer.jdownloader.org/jd_unix_0_9.sh -O install.sh
chmod +x install.sh
./install.sh
```

> No suitable Java Virtual Machine could be found on your system.
The version of the JVM must be at least 1.6 and at most 2.0.
Please define INSTALL4J_JAVA_HOME to point to a suitable JVM.

apotal12, 2016-04-16:
This worked for me on DietPi using LXDE Desktop:

Install Java:
> echo "deb http://ppa.launchpad.net/webupd8team/java/ubuntu trusty main" | tee /etc/apt/sources.list.d/webupd8team-java.list
echo "deb-src http://ppa.launchpad.net/webupd8team/java/ubuntu trusty main" | tee -a /etc/apt/sources.list.d/webupd8team-java.list
apt-key adv --keyserver keyserver.ubuntu.com --recv-keys EEA14886
apt-get update
apt-get install oracle-java8-installer

Install jDownloader:
> mkdir /root/jdownloader
cd /root/jdownloader
wget http://installer.jdownloader.org/JDownloader.jar
chown -R root:root /root/jdownloader
nano /root/Desktop/jDownloader2.desktop

Now just insert and save:
>[Desktop Entry]
Encoding=UTF-8
Name=jDownloader2
Comment=jDownloader2
Exec=bash -c "java -jar /root/jdownloader/JDownloader.jar"
Icon=/root/jdownloader/jDownloader.png
Type=Application
Categories=GTK;Utility;

Finally:
> chmod +x /root/Desktop/jDownloader2.desktop
wget -O/root/jdownloader/jDownloader.png http://jdownloader.org/_media/knowledge/wiki/jdownloader.png

Now you can start JDownloader using the LXDE Desktop Icon.

WolfganP, 2016-04-16:
Nice suggestion. I would like to have the chance of running it headless (no UI) and manage it thru the web remote at my.jdownloader.org (good & recent article of config options at https://board.jdownloader.org/showthread.php?t=67217)

Fourdee, 2016-04-16:
@apotal12 Thanks for the guide :+1: , I'll flag this for v117.

@WolfganP Excellent :+1: , I'll check this out aswell.

Fourdee, 2016-04-25:
#### Configuration issues:

I cant see a way to automate the users account details (encrypted):
```
root@DietPi:~# cat /etc/jdownloader/cfg/org.jdownloader.settings.AccountSettings.accounts.ejs
▒l$▒\▒HB▒
```

I also cant see a way to change the default download directory (```/root/Downloads```):
```
root@DietPi:~# cat /etc/jdownloader/cfg/org.jdownloader.settings.GeneralSettings.json
{
  "preferbouncycastlefortls" : false,
  "maxdownloadsperhostenabled" : false,
  "keepxoldlists" : 5,
  "useavailableaccounts" : true,
  "dupemanagerenabled" : true,
  "savelinkgrabberlistenabled" : true,
  "forcemirrordetectioncaseinsensitive" : true,
  "downloadspeedlimitrememberedenabled" : true,
  "downloadspeedlimitenabled" : false
```

Its not looking very "DietPi-able" at the moment. Lots of user input will be required.

Fourdee, 2016-04-26:
Sorry guys, going to have to turn this one down. The current JDownloader installation method isnt DietPi material at the moment as we cant configure default download directory, or automate the users login creds for my.jdownloader.com. 

I will post a script shortly that will allow users to manually install it, but we wont be featuring Jdownloader in ```dietpi-software``` at this time.

Fourdee, 2016-04-26:
Install Java:
```
cat << _EOF_ > /etc/apt/sources.list.d/webupd8team-java.list
deb http://ppa.launchpad.net/webupd8team/java/ubuntu trusty main
deb-src http://ppa.launchpad.net/webupd8team/java/ubuntu trusty main
_EOF_

apt-key adv --keyserver keyserver.ubuntu.com --recv-keys EEA14886
apt-get update
# - Accept the license: https://github.com/Fourdee/DietPi/issues/298
debconf-set-selections <<< "oracle-java8-installer shared/accepted-oracle-license-v1-1 boolean true"
apt-get install -y oracle-java8-installer
```

Install JDownloader:
```
mkdir -p /etc/jdownloader
wget http://installer.jdownloader.org/JDownloader.jar -O /etc/jdownloader/JDownloader.jar

# Update JDownloader
java -Djava.awt.headless=true -jar /etc/jdownloader/JDownloader.jar -norestart

# - USER input | goto https://my.jdownloader.org and register an account

#1st run, user enter http://my.jdownloader.com login details
java -Djava.awt.headless=true -jar /etc/jdownloader/JDownloader.jar

# - Press CTRL+C to exit.
```

Run JDownloader (headless)
```
java -Djava.awt.headless=true -jar /etc/jdownloader/JDownloader.jar &> /var/log/jdownloader.log &
```

closed 2016-04-26 by Fourdee

tillcash, 2017-10-31:
How to make **java -Djava.awt.headless=true -jar /etc/jdownloader/JDownloader.jar &> /var/log/jdownloader.log &** run on startup ?

Fourdee, 2017-11-05:
@tillcash 
Pop it in ```/etc/rc.local```, above the ```exit 0``` line.

MichaIng, 2021-09-01:
As it became quite popular on FeatHub, I'll reopen this request: https://feathub.com/MichaIng/DietPi/+103