MichaIng/DietPi #209: dietpi-software still prompts for input during some installs

rhkean, 2016-02-26:
While working on #205 , phpMyAdmin began prompting for database passwords.  This should not happen, given the debconf-set-selections code in dietpi-software.  Some research indicated that the automated apt-get install should include:  DEBIAN_FRONTEND=noninteractive

this change to dietpi-software appears to fix the problem... it also seems to speed up the install... (although I haven't timed it)
```
        AGI(){
                local string="$@"
                DEBIAN_FRONTEND=noninteractive apt-get install -y $string

                local result=$?
                if (( $result != 0 )); then
                        Error_AptGet_Failed
                fi
        }
```

Fourdee, 2016-02-27:
@rhkean 
It sounds like the debconf settings are not being applied. I just tried phpmyadmin install on Jessie, did not prompt for any debconf set details and seems to be working fine.

Are you running Wheezy?

rhkean, 2016-02-27:
no.  Jessie... RPi2 image 
written with the dietpi-writeimage script
only 2 mods.... #205 patch to dietpi-software and automated dietpi.txt

Fourdee, 2016-02-27:
Bizzare :)

Can you remember your software selections in dietpi-software before the installation started? May something else causing the issue.

rhkean, 2016-02-27:
I just reran the test.
I can send you the exact files.  email or post on Fuzon?

here are the dietpi.txt changes from default:
```
+Hostname=keanbean2.keanfamily.net

+AUTO_AutoLogin=1
+AUTO_SkipLicensePrompt=1
+AUTO_DietpiSoftware_SkipUpdateRebootPrompt=1
+AUTO_DietpiSoftware_SkipQuestions=1
+AUTO_DietpiSoftware_SkipUsbDrive=1
+AUTO_DietpiSoftware_IgnoreErrors=1
+AUTO_Install_Index=1

+AUTO_DietpiSoftware_Install_LINUXDASH=1

+AUTO_DietpiSoftware_Install_PHPSYSINFO=1
+AUTO_DietpiSoftware_Install_PHPIMAGEGALLERY=1

+AUTO_DietpiSoftware_Install_WEBSERVER_LLAP=1
+AUTO_DietpiSoftware_Install_WEBSERVER_MYADMINPHP=1

+AUTO_DietpiSoftware_Install_VIMFULL=1
+AUTO_DietpiSoftware_Install_SSHCLIENT=1
```
and the dietpi-software diff....
```
--- /var/tmp/mnt/boot/dietpi/dietpi-software    2016-02-11 15:42:14.000000000 -0500
+++ v110/dietpi-software        2016-02-27 12:48:50.196388889 -0500
@@ -479,6 +479,16 @@
                local info_3="MySql"
                local info_4="PHP"
                #--------------------------------------------------------------
+               # - Installs that require at least 1 webserver stack
+               # -   If none set, then enable LASP stack
+               if (( $LINUXDASH == 1 || $PHPSYSINFO == 1 || $PHPIMAGEGALLERY == 1 )); then
+                       if (( $WEBSERVER_LASP == 0 && $WEBSERVER_LAMP == 0 && $WEBSERVER_LAAP == 0 &&     \
+                             $WEBSERVER_LESP == 0 && $WEBSERVER_LEMP == 0 && $WEBSERVER_LEAP == 0 &&     \
+                             $WEBSERVER_LLSP == 0 && $WEBSERVER_LLMP == 0 && $WEBSERVER_LLAP == 0 )); then
+                               WEBSERVER_LASP=1
+                       fi
+               fi
+
                # Apply all "combo" installs that require webserver stack
 
                # - Installs that require WEBSERVER_LASP
@@ -488,14 +498,8 @@
                        WEBSERVER_LASP=1
                elif (( $RASPCONTROL == 1 )); then
                        WEBSERVER_LASP=1
-               elif (( $LINUXDASH == 1 )); then
-                       WEBSERVER_LASP=1
                elif (( $PIHOLE == 1 )); then
                        WEBSERVER_LASP=1
-               elif (( $PHPSYSINFO == 1 )); then
-                       WEBSERVER_LASP=1
-               elif (( $PHPIMAGEGALLERY == 1 )); then
-                       WEBSERVER_LASP=1
                fi
 
                # - Installs that require WEBSERVER_LAMP
@@ -755,7 +759,13 @@
                        echo -e "phpmyadmin      phpmyadmin/password-confirm     password dietpi" | debconf-set-selections
                        echo -e "phpmyadmin      phpmyadmin/setup-password       password dietpi" | debconf-set-selections
                        echo -e "phpmyadmin      phpmyadmin/mysql/app-pass       password dietpi" | debconf-set-selections
+                       if (( $WEBSERVER_APACHE > 0 )); then
                        echo -e "phpmyadmin      phpmyadmin/reconfigure-webserver        multiselect     apache2" | debconf-set-selections
+                       elif (( $WEBSERVER_LIGHTTPD > 0 )); then
+                               echo -e "phpmyadmin      phpmyadmin/reconfigure-webserver        multiselect     lighttpd" | debconf-set-selections
+                       else
+                               echo -e "phpmyadmin      phpmyadmin/reconfigure-webserver        multiselect     none" | debconf-set-selections
+                       fi
 
                        AGI phpmyadmin
 
@@ -1484,7 +1494,7 @@
                        Banner_Installing
 
                        #check source is online
-                       INSTALL_URL_ADDRESS='http://sye.dk/sfpg/Single_File_PHP_Gallery_4.5.7.zip'
+                       INSTALL_URL_ADDRESS='http://dietpi.com/downloads/binaries/all/Single_File_PHP_Gallery_4.6.1.zip'
                        /DietPi/dietpi/func/check_connection "$INSTALL_URL_ADDRESS"
 
                        #Install
@@ -2478,7 +2488,7 @@
                                # - Php-FPM confs
                                cp /DietPi/dietpi/conf/php-fpm.conf /etc/php5/fpm/php-fpm.conf
                                cp /DietPi/dietpi/conf/php-fpm_pool.conf /etc/php5/fpm/pool.d/www.conf
-                               sed -i '/cgi.fix_pathinfo=/c\cgi.fix_pathinfo=0' /etc/php5/fpm/php.ini
+                               sed -i '/cgi.fix_pathinfo=/c\cgi.fix_pathinfo=1' /etc/php5/fpm/php.ini
 
                                # - PHP5-fpm optimizations based on total cores
                                sed -i "/pm.max_children = /c\pm.max_children = $CPU_CORES_TOTAL" /etc/php5/fpm/pool.d/www.conf
```

rhkean, 2016-02-27:
reran tests:
image:  DietPi_v109_RPi-(Jessie).img (md5:  862b9c4c2fa1dd3edcdfbcd2e5d6ac09 )
dietpi.txt (with diff above)
dietpi-software (with diff above)
RESULT:  prompts for admin password

image:  DietPi_v109_RPi-(Jessie).img (md5:  862b9c4c2fa1dd3edcdfbcd2e5d6ac09 )
dietpi.txt (with diff above)
dietpi-software (with diff above + patch to AGI function)
RESULT:  all software installed + reboot and login to phpMyAdmin with u:phpmyadmin p:dietpi

Fourdee, 2016-02-28:
@rhkean 
Thanks for the info. 

> +AUTO_DietpiSoftware_Install_WEBSERVER_LLAP=1

Ah, so this is in regards to a lighttpd installation on the https://github.com/Fourdee/DietPi/issues/205 ticket. 
I thought you were referring to the v109 live LAMP install. I changed the ticket description to prevent myself getting confused again :)

I'll take a look using your updated code.

Fourdee, 2016-03-30:
@rhkean 
Excellent work on this :+1: 

I'll mark this as closed, will leave https://github.com/Fourdee/DietPi/issues/205 open for now whilst I do some testing.

closed 2016-03-30 by Fourdee