MichaIng/DietPi #210: DietPi-Software | Add emonCMS/emonPi installation option.

Fourdee, 2016-02-27:


R1CH-T, 2016-02-28:
As I mentioned in my forum post, a week ago (21/02/2016), [Open Energy Monitor (OEM)](http://openenergymonitor.org/) published the first release candidate for [a major operating system update and software upgrade](http://openenergymonitor.org/emon/node/12212). They have shifted to using Raspbian Light as their base OS and have added much new functionality.

The major new features they have added aim at "making the emonPi the heart of a smart energy optimised home" and focus on "how live energy monitoring data from the emonPi can easily be used to inform other services" [(source)](http://openenergymonitor.blogspot.co.uk/2015/12/openenergymonitor-emonpi-and-openhab.html).

DietPi includes several methods to have a go at some hardware hacking, including:

RPi Hardware Projects:
- RPi.GPIO - GPIO Interface library for RPi (python).
- WiringPi - GPIO Interface library for RPi (c).
- WebIOPi - Web interface to control RPi GPIO.
- I2c - Enables support for I2c based hardware.

It also has a Home Automation section:

Home Automation:
- Grasshopper - Web App to control Bticino MyHome.

However, [Node-RED](http://nodered.org/) is notable by it's absence and I believe that the [openHAB](http://www.openhab.org/features/supported-technologies.html) project would also be a good fit for DietPi.

The [emonCMS](https://github.com/emoncms/emoncms/blob/master/docs/RaspberryPi/readme.md) web interface should be fairly straightforward to include in DietPi, but the hardware control programs mentioned above might be a more complex proposition.

The [emonSD](http://openenergymonitor.org/emon/node/12212) image under development by OEM includes "NodeRED, LightWave RF MQTT OOK control and openHAB all pre installed and configured" as a software stack which should all work together out of the box.

As I mentioned in the forum post, rather than re-inventing the wheel, it should be straight-forward to create an installer or installers for these things, based on OEM's work.

Their project is hosted [here on GitHub](https://github.com/openenergymonitor) and the relevant parts, I believe are:

https://github.com/openenergymonitor/lightwaverf-pi
https://github.com/openenergymonitor/oem_openHab
https://github.com/openenergymonitor/oem_node-red
https://github.com/emoncms

The project's lead developers are Glyn Hudson and Trystan Lea.
Contact details:
https://openenergymonitor.org/emon/people
contact@openenergymonitor.zendesk.com
http://openenergymonitor.org/emon/user/10 https://github.com/glynhudson
http://openenergymonitor.org/emon/user/3 https://github.com/ TrystanLea trystan.lea@gmail.com

I hope that this information has been useful and that my suggestions might make useful additions to DietPi.

Let me know if you need anything,
Rich.

Fourdee, 2016-02-28:
@R1CH-T 

Many thanks for the detailed information, i appreciate it.

I'll update this ticket when work begins (hopefully after v110 is released).

R1CH-T, 2016-03-01:
I have just spent ages researching this further and added more links to the list above, however I found more relevant links, which I will post here. I'll leave the long list of links above, for future reference.

https://www.raspberrypi.org/forums/viewtopic.php?f=37&t=66946
https://github.com/openhab/openhab/wiki/Raspberry-Pi-RC-Switch-Binding

R1CH-T, 2016-03-03:
I posted on [OEM's forum about this ](https://openenergymonitor.org/emon/node/12212#comment-40083) and Glyn Hudson said:

> Yes RC switch binding should with with our hardware. Just select GPIO 18 to use the OOK Tx module. Installing by default may conflict with LWRF. Testing would be needed. Let me know if you get this working. 
> 
> DietPi looks very cool. This is the full image build guide I created while making the emonPi image. Emoncms can be installed as a normal web app but it's frequent disk writing could cause a normal SD to fail in a few months. This is why we have put a lot of effort into RO FS and low-write optimisation. See full image build guide: 
> 
> https://github.com/openenergymonitor/emonpi/blob/master/docs/SD-card-build.md

Fourdee, 2016-03-03:
@R1CH-T 
Thanks for the info. I've nugged this to v112 as v111 was mostly a dedicated release for RPiv3 support.

Fourdee, 2016-03-05:
Nugged to v113, apologies @R1CH-T . v112 is a hotfix release for RPi v3 wifi hotspot.

Fourdee, 2016-03-08:
@R1CH-T 
@R1CH-T 
Ok time to crack on :)

> The emonCMS web interface should be fairly straightforward to include in DietPi, but the hardware control programs mentioned above might be a more complex proposition.

Yep, this is being implemented 1st.

> I have just spent ages researching this further and added more links to the list above, however I found more relevant links, which I will post here. I'll leave the long list of links above, for future reference.

:+1: . Yep, there's enough reference and reading to keep me busy lol. I appreciate all the input and it will make this process much easier.

R1CH-T, 2016-03-08:
Thanks for looking into this. :)
A lot of those links in the long list came as the result of googling and re-finding stuff that I'd looked into ages ago.
It's certainly worth skimming over, but this stuff is not quite as valid as it was at the time it was posted and the concepts have since been absorbed into mega-projects like OpenHAB.
[OpenHAB's downfall might eventually happen due to it's being a little too all-embracing in it's scope](https://community.openhab.org/t/the-future-of-openhab/4048), but until then it is a really exciting project, cutting out much of the hard work in connecting stuff to the Pi. It seems to be the very best project of it's type out there.
It also seems to have a very vibrant community, so some users could be drawn here, too.

Fourdee, 2016-03-09:
Ok so far, i've created 3 new installation options:
Redis Server | https://github.com/Fourdee/DietPi/issues/57
EmonCMS
EmonHUB

Just need to code them in, then i'll read up on your research to find the next culprit to add :)

Fourdee, 2016-03-11:
@R1CH-T 
I've just read through: https://github.com/openenergymonitor/emonpi/blob/master/docs/SD-card-build.md , wish I read that first lol

I dont believe emulating all those installations on DietPi is beneficial. As DietPi installs the software when its required, all it would take is 1 of those programs to change a line of code (eg: a config file location) and it would throw out the whole installation.
The sheer amount of programs worries me and I dont have the time to maintain that many installations.

It also appears the EmonPi guys have really nailed the image with pre-setup features like read only file system, which DietPi is yet to implement.
So i believe instead of trying to emulate the EmonPi installation, I should focus on the single applications that users want (eg: emonCMS, emonHub) and gradually build from there.

I'd appreciate your thoughts on this.

Fourdee, 2016-03-11:
@R1CH-T 
Also, I do not currently own any of the hardware required, to test any of this software in a open energy monitor environment.
Do you, and, would you be willing to test/verify the software installations are working?

R1CH-T, 2016-03-13:
Thanks again for your efforts, Dan.
> Also, I do not currently own any of the hardware required, to test any of this software in a open 
> energy monitor environment.
> Do you, and, would you be willing to test/verify the software installations are working?
I'll be happy to test anything you need me to; just let me know.
> I've just read through: https://github.com/openenergymonitor/emonpi/blob/master/docs/SD-card-build.md , wish I read that first lol
Yeah, sorry about that. I wish I'd posted that link sooner and saved you some work.
> I don't believe emulating all those installations on DietPi is beneficial. As DietPi installs the software > when its required, all it would take is 1 of those programs to change a line of code and it could
> throw out the whole installation.
> The sheer amount of programs worries me and I dont have the time to maintain that many
> installations.
I'd never dream of asking you to do something that complex. The emonPi runs a bespoke Raspbian image just like DietPi does and they are designed for different purposes. Just having the ability to install the (emon)CMS software would be very useful, as you can enter the read and/or write API keys and gather, log and display sensor data from the main unit on a DietPi box and use it to trigger other functions on the DietPi box.
> It also appears the EmonPi guys have really nailed the image with pre-setup features like read
> only file system, which DietPi is yet to implement.
> So I believe instead of trying to emulate the complete EmonPi installation in DietPi, I should focus
> on the single applications that users want (eg: emonCMS, emonHub) and gradually build from
> there.
Yes, absolutely. The work they have carried out over there might be quite beneficial in pointing the way for creating the same functionality in DietPi. Also, the low-write file system seems to be a very good idea for prolonging the life of SD's or USB flash drives. As I cross-posted above:
> Emoncms can be installed as a normal web app but it's frequent disk writing could cause a normal
> SD to fail in a few months. This is why we have put a lot of effort into RO FS and low-write
> optimisation.

> I'd appreciate your thoughts on this, and, which software you would recommend to be added next
> (eg: openHAB).
Oh, definitely OpenHAB, as I believe that this would be a beneficial feature for users who want to be able to control all sorts of commercially available hardware from their DietPi box and as I said: Input data from emonCMS can be used to create trigger events for this stuff.
That's the particular use case I have in mind for myself, at least. I'm thinking getting one of the Raspberry Pi touch-screens to use as a lighting/mains-switch/whatever control panel. Maybe a web-kiosk type interface with the ability to flip between pages with switches, info screens etc.

Fourdee, 2016-03-19:
@R1CH-T 
My EmonPi arrived today (many thanks to you :+1: ) and made me to rethink the DietPi implementation. My goal is to allow for a single EmonPi installation option to make this as simple as possible for the user. I'll start with the basic features (cms/hub) so the user can check stats, graphs and sensor data, then build on it from there.

I just need to get my head around the device, the software used, configuration and user interface 1st, so a few days of testing the EmonPi is needed. Its going to be a mammoth job to implement all the software and configuration in DietPi, but it should be worth it.

I'am going to move this to v114 and disable the emonCMS and emonHUB installs for now. Allows me to get the v113 bug fixes out and focus on this afterwards. I know i've nudged this a few times to the next version, so I apologise, but I want the DietPi implementation to be "right" from the start.

R1CH-T, 2016-03-22:
> @R1CH-T
My EmonPi arrived today (many thanks to you :+1: ) and made me to rethink the DietPi implementation. My goal is to allow for a single EmonPi installation option to make this as simple as possible for the user. I'll start with the basic features (cms/hub) so the user can check stats, graphs and sensor data, then build on it from there.

Fantastic!
I Hope you get lots of use out of the device, find it as fascinating as I do and that some useful ideas come out of this. I think that there could be lots of useful additions to DietPI that can be helpful to  many users appearing out of this. I'm happy to be able to support two of my favourite projects at the same time.

> I just need to get my head around the device, the software used, configuration and user interface 1st, so at least a few solid days of testing the EmonPi is needed. Its going to be a mammoth job to implement all the software and configuration in DietPi, but it should be worth it.

I hope that I'm not sending you off at too much of a tangent, here. Don't let this get in the way of other feature improvements and additions requested by other users: Take your time.
Hopefully it can be implemented in such a way that it pulls in most of what it needs from the other project and requires minimal maintenance. Also, it would be great if it feeds back into OEM, too.

> I'am going to move this to v114 and disable the emonCMS and emonHUB installs for now. Allows me to get the v113 bug fixes out and focus on this afterwards. I know i've nudged this a few times to the next version, so I apologise, but I want the DietPi implementation to be "right" from the start.

Hey, no worries.
I'm just glad that you're fired up about this idea, too!

Fourdee, 2016-03-22:
@R1CH-T 

> I hope that I'm not sending you off at too much of a tangent, here. Don't let this get in the way of other feature improvements and additions requested by other users: Take your time.

Not at all. I always prioritize core bugs over everything else. However, I need to make sure i'am doing stuff that I enjoy sometimes, helps make it all worth while :)

> Fantastic!
I Hope you get lots of use out of the device, find it as fascinating as I do and that some useful ideas come out of this. I think that there could be some cool additions to DietPi as a result, that can be helpful to many users, appearing out of this. I'm happy to be able to support two of my favourite projects at the same time.

I love it, got it up and running yesterday. Now I can see when the misses leaves the TV on lol. She isnt too happy about how i've attached the EmonPi lol:
[image]

Its only temporary ;)

After having a good tinker and play with it:
The plan is to create a DietPi lightweight installation option that will aim at users who just want energy usage stats with the web interface (emoncms and emonhub). Once thats done, I'll look into the other areas (eg: RF) and add them later down the line.

WolfganP, 2016-03-22:
Interesting... but is that the full EmonPi or just the EmonTx module (http://shop.openenergymonitor.com/emontx-v3-electricity-monitoring-transmitter-unit-433mhz/)?
As per my understanding EmonTx may be ordered w/o wireless for a direct connection to an RPi, and that will be the cheaper solution for integration, right?

Fourdee, 2016-03-22:
@WolfganP

> Interesting... but is that the full EmonPi or just the EmonTx module 

Its just the EmonPi PCB, pulse sensor and clipon power sensor. The basic package + the pulse sensor:
http://cdn2.bigcommerce.com/server4400/98a75/product_images/optionset_rule_images/28_zoom_1429716170.jpg £59

> As per my understanding EmonTx may be ordered w/o wireless for a direct connection to an RPi, and that will be the cheaper solution for integration, right?

From my understanding, the EmonPi can receive the transmissions from a EmonTx unit. http://shop.openenergymonitor.com/emonpi-2/ . I think you could buy a RF receiver for your RPi and use to receive the EmonTx monitoring data £60?

To be honest, there's so many combinations and methods, i'am still trying to get my head round it all :) Was one of my biggest gripes placing the order, just confused me.

WolfganP, 2016-03-22:
@Fourdee Yep, I'm on the same situation. Trying to figure out what's the most basic set to allow power monitoring, wired, no fancy displays or cases :-)

R1CH-T, 2016-03-22:
@Fourdee @WolfganP 
I backed the Kickstarter and got the fancy-schmancy unit with the aluminium enclosure and all the solar-PV stuff, but it looks like you got the best bang for your bucks.
It looks like you got everything you need for your setup.
Just set up a DietPi box with the emonPi's API key and I think that it will read the feed data and this can then be used to set up graphs etc.

R1CH-T, 2016-03-22:
You can do all of this from the emonPi, but this is what I'm talking about.

R1CH-T, 2016-03-22:
The TX is an older concept and provides loads of connections for sensors, but is not at all necessary.

Fourdee, 2016-03-22:
> Trying to figure out what's the most basic set to allow power monitoring, wired, no fancy displays or cases :-)

As a tip, if your looking into purchasing the EmonPi:
You dont need any additional power supplies. 
This was another confusion when I ordered the unit. The EmonPi has a Mini USB (not micro) power socket, however, if you have a 2amp microUSB PSU for your Rpi, plug it into your RPi and it will power both the RPi and EmonPi board. So you only need 1 PSU for your RPi, nothing else.

The MiniUSB power is required for the EmonPi with the fully assembled case unit. Possibly to allow them to sell you additional "stuff", but not sure.

> I backed the Kickstarter and got the fancy-schmancy unit with the aluminium enclosure and all the solar-PV stuff, but it looks like you got the best bang for your bucks.

I seen the Kickstarter, did very well.
Yep, I have a habit of getting the most out of as little as possible :)

WolfganP, 2016-03-22:
> Yep, I have a habit of getting the most out of as little as possible :)

Same here. I'm not interested in local readings nor special radio link. I think I'm going for the emonTx... http://openenergymonitor.blogspot.com.ar/2015/04/emonpi-vs-emontx-v3-comparison.html

Fourdee, 2016-03-23:
@R1CH-T 
I've hit a brickwall with the EmonPi's data with the minimal EmonCMS/HUB install, seems way off. I've posted on the OEM forum, hopefully get some assistance on this:
https://openenergymonitor.org/emon/node/12384

Fourdee, 2016-03-23:
Turns out I needed to use the EmonPi EmonHub branch.

Just need to get the electric app working, then all good to go:
[image]

Fourdee, 2016-03-24:
@R1CH-T 
#### Turns out, we can send the data directly to emoncms.org. 
This means we dont need to install LAMP, emonCMS etc.
All we needed was a slightly modified EmonHub (emonpi variant) that wasn't restricted to work with ```/home/pi```, so I forked it: https://github.com/Fourdee/emonhub

#### So the install is ultra lightweight:
[image]

#### The only downside of the Emoncms.org upload is there will be some user setup required. 
Basically to obtain the API key and add it into emonhub.conf. But i'll document those steps.
```
APIKEY="ReplaceWithMyApiKey";sed -i "/apikey/c\        apikey = $APIKEY" /etc/emonhub/conf/emonhub.conf
service emonhub restart
```

#### Works on RPI 3. 
Although we needed to disable the BCM 43438 entirely to recover UART 0 that EmonPi uses. ```core_freq=250``` had no effect. So no Inbuilt WiFi or Bluetooth on the RPi 3.

@R1CH-T 
Any chance you would be able to test the install?

Fourdee, 2016-03-24:
Strange, seems i'am now being monitored by a moderator for my posts on OEM site lol:

> Your comment has been queued for moderation by site administrators and will be published after approval.

https://openenergymonitor.org/emon/node/12384

R1CH-T, 2016-03-24:
Great work Dan, thanks a lot for you efforts.

Let's go through this:
> I've hit a brickwall with the EmonPi's data with the minimal EmonCMS/HUB install, seems way off. I've posted on the OEM forum, hopefully get some assistance on this:
https://openenergymonitor.org/emon/node/12384

Please don't worry about not getting an immediate response; it's a small UK team, with a major update coming up and trying to implement big improvements right now. I think that you might have an idea of the situation they are in.

> Turns out I needed to use the EmonPi EmonHub branch.
Just need to get the electric app working:
[SCREENSHOT]

Yes, I see what you mean.

I didn't foresee how (or even if) you were going to go about this, but I could have told you that much if I had known your intentions and if I had the time to reply promptly and in-depth to your previous comments. Sorry, Dan. :(

As I said before; I got the full emonPi kit. It it simply worked out of the box and I was able to wire it all up, configure the software, set it on my LAN and also connect it to their online emonCMS site. I have also trouble-shot, bug-fixed and fed-back on the project, but I am more of a user than a hacker. Hopefully you are becoming more familiar with the setup now and probably understand more about how it works than I do, but there are wrinkles that I may be able to help you with.

It is quite complex and was probably a bigger task than I realised I had passed on, due to my lack of knowledge in this area and available time to work it out for myself. Sorry about that.

What I had in mind was that I could connect a remote Pi to my emonPi on the LAN (via the API) and control/monitor it from the other side of the house. I could then also use this second Pi to control hardware connected to itself. I could then control it via a touch-screen or even Mycroft or something.

Do you see what I mean?

Turn on the hot water immersion heater or charge the car if the solar panels are supplying enough juice, thus cutting the gas and electricity bills, or say 'Tea, Earl Grey, Hot' and have the toilet flush.

> Turns out, we can send the data directly to emoncms.org.
This means we dont need to install LAMP, emonCMS etc.

That's cool. No LAMP stack cuts down on the overhead.
I don't understand how it works exactly, but I knew about the API key functionality and know that it talks to remote a emonCMS instance. I think that this includes emoncms.org , but is not exclusive.

> All we needed was a slightly modified EmonHub (emonpi variant) that wasn't restricted to working with /home/pi, so I forked it: https://github.com/Fourdee/emonhub and modified it for root with /etc/emonhub install dir.
So the install is ultra, ultra, ultra lightweight :+1: :
[SCREENSHOT]

Ultra, ultra, ultra lightweight sounds great to me!
> The only downside of the Emoncms.org upload is there will be some user setup required.

How is this different to any other 'cloud' stuff. We can help, but it's unavoidable.

Basically to obtain the API key and add it into emonhub.conf, and, setup the feeds. But i'll document those steps.

APIKEY="ReplaceWithMyApiKey";sed -i "/apikey/c\        apikey = $APIKEY" /etc/emonhub/conf/emonhub.conf
service emonhub restart

Great as long as the upstream project maintains the variables. Par for the course, I suppose.

Works on RPI 3.

Although we needed to disable the BCM 43438 entirely to recover UART 0 that EmonPi uses. core_freq=250 had no effect. So no Inbuilt WiFi or Bluetooth on the RPi 3.

@R1CH-T
Any chance you would be able to test the install?

Fourdee, 2016-03-24:
@R1CH-T 

> It is quite complex and was probably a bigger task than I realised I had passed on, due to my lack of knowledge in this area and available time to work it out for myself. Sorry about that.

Not at all, no need to apologise, I love everything about the EmonPi! 
Feeds the geek inside of me and I'll actually use it after the DietPi implementation and testing is completed. Also really enjoyed working on it and the challenge of creating a minimal installation option for EmonPi as a energy usage device with an existing RPi.

> Yep.
I will flash DietPi to an SD card and put it in my emonPi for proper testing.
I have been alpha- and beta-testing the 'official' emonPi images.
Probably some time at night when the solar PV is off and the electricity usage is minimal - would be the best time.

Excellent. :+1: 
There's no rush as i still need to finish up some of the installation code and run last tests. I will also need to explain to you how to change your DietPi install to use testing code. Its only 1 line in dietpi.txt, however, I'll walk you through it when all is completed.

> (I am trying to collect some semi-reliable data for long term analysis, so uptime is a factor.)

Hehe, same here.
#### Heres my current setup using the DietPi installation:
- EmonPi PCB
- EmonPi Power usage clipon
- Raspberry Pi Zero (Finally found a use for it lol)
- Ecig power bank lol
[image]
[image]
[image]

> What I had in mind was that I could connect a remote Pi to my emonPi on the LAN (via the API) and control/monitor it from the other side of the house. I could then also use this second Pi to control [...]
Do you see what I mean?

I think so. For simplicity, you could leave the EmonPi as a dedicated unit. Then on the other RPi with touchscreen, install LXDE and use the webbrowser to view usage?
 

> Note to self:
Add apikey var to dietpi.txt, use for automation installation.

wthomson, 2016-03-25:
_Strange, tried to post this info on OEM site, seems i'am now being monitored by a moderator for my posts lol:_

    _Your comment has been queued for moderation by site administrators and will be published after approval._

Hi Fourdee,
I'm one of the Moderators at openenergy.org. I've set your user account so your posts will be visible immediately.

Regards,
Bill

R1CH-T, 2016-03-25:
@Fourdee 
> Not at all, no need to apologise, I love everything about the EmonPi!
Feeds the geek inside of me and I'll actually use it after the DietPi implementation and testing is completed. Also really enjoyed working on it and the challenge of creating a minimal installation option for EmonPi as a energy usage device with an existing RPi.

This is great! I'm enjoying watching you do it.
> Excellent. :+1:
There's no rush as i still need to finish up some of the installation code and run last tests. I will also need to explain to you how to change your DietPi install to use testing code. Its only 1 line in dietpi.txt, however, I'll walk you through it when all is completed.

I certainly may need a little bit of hand-holding on this, but I'm eager to learn, so thanks.
> Here's my current setup using the DietPi installation:

> - Raspberry Pi Zero (Finally found a use for it lol)

Wow! That's so cool! I've got one of those collecting dust, too.

> - Ecig power bank

Brilliant. I had to look that one up, but yeah, great idea - again.
I have a mate at work who has an e-cigarette with a massive battery pack on it: I mean; this thing could be used as a disco smoke machine!
Let me know how much battery-life you get out of that - it's intriguing.
An interesting thing I'd point out is: I have two of the [emonTH units](https://openenergymonitor.org/emon/modules/emonTH), which are each powered by an AA battery and report their voltage back to base. The OpenHAB module can actually be set up to send an email or other alert to let you know when the battery needs changing/charging. I haven't set this up because I can't remember the where I read this, but this might be useful for your situation, if it can be made to work on your base station.

> I think so. For simplicity, you could leave the EmonPi as a dedicated unit. Then on the other RPi with touchscreen, install LXDE desktop and use the webbrowser to view usage?

Perhaps LXDE or maybe a tiling window manager or even a web-browser running on the screen buffer or something like that. I'd need the touchscreen to work, though. I need to look into this.
I have a [Raspberry Pi Touchscreen](https://uk.rs-online.com/web/p/graphics-display-development-kits/8997466/) on order from RS and it is expected to be in stock on the 29th of March and be delivered on the 8th of April, so I guess I can look into this some time after that.

Fourdee, 2016-03-25:
@wthomson 

> Hi Dan,
I'm one of the Moderators at openenergy.org. I've changed your oem user account to make your posts visible immediatly after submitting them. i.e. you shouldn't see any more "queued for moderation" messages.

Hi Bill. Thank you for doing that, I appreciate it.

Fourdee, 2016-03-25:
@R1CH-T 

> I have a mate at work who has an e-cigarette with a hefty coil and a massive battery pack on it: I mean; this thing could be used as a disco smoke machine!
Let me know how much battery-life you get out of that - it's intriguing.

Lol. Yep. Portable smoke machines. Buts its too easy to increase your nicotine addiction with those devices.

Battery life, wasnt that great to be honest. Even when setting the RPi to powersave 700mhz and disabling HDMI output.
The Ecig was charged to 4.1 volts, After 3 hours, it was at 3.7 volts. So I think overall, around 6~ hours of use on its internal 2600mah battery.
I'am not sure how much power the EmonPi board uses, but even though I wasn't using the RF features, i think its still powered.

> The OpenHAB module can actually be set up to send an email or other alert to let you know when the batteries need changing/charging.
I haven't set this up because I can't remember the where I read this, but this might be useful for your situation, if it can be made to work on your base station.

Sounds impressive, however, i'am an electric/GPIO novice. I wouldn't even know where to start on how to obtain the current battery life into values lol. Something for the future :)

> Perhaps LXDE or maybe a tiling window manager or even a web-browser running on the screen buffer or something like that. I'd need the touchscreen to work, though. I need to look into this.

Sounds like a plan. I think you could even just run X11 on its own with a webbrowser, without the desktop. 
I'am yet to purchase the RPI touchscreen, so you'll have to let me know your impressions of it :)

Fourdee, 2016-03-25:
@R1CH-T 
Just wrapping up the installation code and documentation: http://fuzon.co.uk/phpbb/viewtopic.php?f=8&t=5&p=1525#p1525

I've added a menu interface that allows the user to enter their API key during installation. This will be automatically applied by DietPi.
[image]

Should have a test ready for later today and I'll provide the steps.

Fourdee, 2016-03-25:
@R1CH-T  (And anyone else who wants to test this)

- Download latest DietPi image: http://goo.gl/EsIvUp
- Write image to SD card
- Before unplugging the SD card, go to the boot partition (my computer) and open dietpi.txt in wordpad.
- At the very bottom, replace gitbranch to ```gitbranch=testing```
- Save the file
- Plug the SD card into your RPi and power on.

Now you can run through the normal DietPi setup process. When the dietpi-software menu appears:
- Select DietPi optimized software
- Scroll down and select EmonPi for installation
- Follow the onscreen instructions, then select Go start installation.

To setup your emoncms.org account: http://fuzon.co.uk/phpbb/viewtopic.php?f=8&t=5&p=1525#p1525 under "Setup Inputs and feeds for Power usage:"

R1CH-T, 2016-03-26:
@Fourdee 
Right. Lots to deal with here, so I'll jump right in and start with your first reply...
>Lol. Yep. Portable smoke machines. Buts its too easy to increase your nicotine addiction with those devices.
He uses a low strength mix, by all accounts, but I get the impression that he's probably wired all the time and is losing sleep because of it. Once he starts to get heart palpitations, he'll probably re-examine his situation. Harsh, I know, but at least he's not inhaling a cyanide and gunpowder mix, along with the nicotine. :|

> Battery life, wasn't that great to be honest. Even when setting the RPi to powersave 700mhz and disabling HDMI output.
The Ecig was charged to 4.1 volts, After 3 hours, it was at 3.7 volts. So I think overall, around 6~ hours of use on its internal 2600mah battery.
I'am not sure how much power the EmonPi board uses, but even though I wasn't using the RF features, I think its still drawing power.

2600mah seems pretty good for a 5v battery pack: It might be very useful for robotics, or strapping a PiZero with some sensors to a balloon or something, but maybe not this.
Yeah, the emonPi board is super simple, affordable open hardware, but is meant to be plugged in all the time and won't be super efficient (at least not at this stage in it's development).

> > The OpenHAB module can actually be set up to send an email or other alert to let you know when the batteries need changing/charging.
    I haven't set this up because I can't remember the where I read this, but this might be useful for your situation, if it can be made to work on your base station.

> Sounds impressive, I'll need to look into that.

It wasn't [an OpenHAB thing](http://openenergymonitor.blogspot.co.uk/2015/12/openenergymonitor-emonpi-and-openhab.html), after all. It turns out that it was a [demonstration Node-RED _Sketch_ (or whatever they're called), from OEM](http://openenergymonitor.blogspot.co.uk/2015/10/emonpi-nodered-and-mqtt.html).
I need to get my head around the interactions between MQTT feeds, NodeRED, OpenHAB, etc.
You can set up MQTT feeds in emonCMS for sure and feed them into these other modules in the EmonPi's software stack, but I simply haven't had the time required to hammer this all into my feeble brain.
If I want to stand any chance of realising any of the goals that I laid out earlier, I need to understand this better.

> Sounds like a plan. I think you could even just run X11 on its own with a webbrowser, without the desktop.
I'am yet to purchase the RPI touchscreen, so you'll have to let me know your impressions of it :)

I was being a little bit facetious and meant to say _framebuffer_, but even got that wrong. I actually had in mind running it on bare X-Server.
I set up a DietPi box on my telly, with Kodi a while back and at some point dropped out to the X-Server, so I think that you actually set this up yourself.
Without looking into the install script myself, could I pick your brains?
Would I be right in thinking that this is indeed the case and would you consider separating out the X-server install part into another install option and calling on it from the Kodi one?
That way, it could be pulled into other installers, like you do with some of the web stacks, etc.
Am I barking up the wrong tree here?

R1CH-T, 2016-03-26:
Following on from Glynn Hudson's reply to [your post on the OEM Forum](https://openenergymonitor.org/emon/node/12384), I didn't really understand what he was getting at:
> Awesome! Nice work, yes the emon-pi varient of emonHub is the more developed version of emonHub and is recommended. Let me know if you have any further questions. Running Emoncms locally on the Pi does [complicate things slightly](https://github.com/openenergymonitor/emonpi/blob/master/docs/SD-card-build.md). However. it's important to use to allow users the option of a stand-alone, fully open and non-cloud data logging option so users can fully 'own' their data. However, a lightweight option of just posting to emoncms.org seems like it's the best for your application. 
I like the Pi-zero attached directly to the emonPi :-) 
I notice in your Github discussion you mention using UART0 on the RasPi3 with the emonPi, that Wifi won't work. This is not true, only the BT needs to be disabled. The Wifi on the RasPi3 works great with the emonPi.
See: http://openenergymonitor.blogspot.com/2016/03/raspberry-pi-3.html

When he said
> Running Emoncms locally on the Pi does [complicate things slightly](https://github.com/openenergymonitor/emonpi/blob/master/docs/SD-card-build.md)

I get the point (if not the details), but I think that this is exactly what I'm asking for, so I am wondering if what I am looking for is doable.
Plus, I'm asking for the NodeRED and OpenHAB stuff (and the interoperability between them), on a remote (Diet)Pi, pulling data off or my existing emonPi. This needs sorting out before I can even think about touch-screens and stuff like that...
It would be wise to wait until most of the bugs have been squashed and the final release is out, before diving in, but I am concerned that my own understanding is so poor at this point.
Your thoughts?

R1CH-T, 2016-03-26:
> @R1CH-T (And anyone else who wants to test this)...

ASAP, I will dd the latest image to an SD card, enable testing, boot it onto a spare Pi, install EmonPi, transfer a copy of my config files and then swap the card into my EmonPi unit for testing. I can then re-attach the existing feeds in the config files to the inputs. I have done this before, so it shouldn't be a problem. It will also be a good test for the setup.
One thing I need to ask is this: Where shall I transfer the config files to and which ones do I need?

[image]

Fourdee, 2016-03-26:
@R1CH-T 

> Would I be right in thinking that this is indeed the case and would you consider separating out the X-server install part into another install option and calling on it from the Kodi one?

In DietPi-Software > Additional linux software, we have a installation option for Xserver. This will install the core Xserver/X11 software and set it up for you.
When you select Desktop/Kodi etc for installation, DietPi-Software will automatically install the Xserver option aswell.
Is this what you meant?

> I get the point (if not the details), but I think that this is exactly what I'm asking for, so I am wondering if what I am looking for is do-able at all.

At the moment, the DietPi installation does not install emonCMS (due to the issue with my electric app not working). So you would be unable to access the web interface on the local machine. 
If you use this minimal installation, all your emonPi data is sent to the cloud. You would then be required to use http://emoncms.org to access your usage stats. The upside is you can access this from anywhere in the world to view your stats.

> I get the point (if not the details), but I think that this is exactly what I'm asking for, so I am wondering if what I am looking for is do-able at all.

I plan to add emonCMS in the future (the installation code is in dietpi-software but disabled), but only if I can find a fix for the Myelectric App.

So I think for now, as you will require RF and all the other OEM installation features, the DietPi installation might not best for you.

> One thing I need to ask is this: Where shall I transfer the config files to and which ones do I need?

As far as I can tell, http://emoncms.org does not support backing up or restoring your data. So for the DietPi installation, currently, you will need to setup new feeds. Steps for this is here: http://fuzon.co.uk/phpbb/viewtopic.php?f=8&t=5&start=50#p1525

Fourdee, 2016-03-28:
@R1CH-T 

I'am going to mark this as completed for now. I've created two new tickets, these will be additional installation choices for the user (eg: modules) whilst we continue improving the DietPi implementation:

- EmonPi - Add optional installation for RF features. https://github.com/Fourdee/DietPi/issues/256
- EmonPi - Add optional installation for local EmonCMS on device (currently remote emoncms.org). https://github.com/Fourdee/DietPi/issues/257

closed 2016-03-28 by Fourdee

R1CH-T, 2016-03-29:
Sorry for dragging my feet on this.
I flashed this to an SD card a few days ago,  but I only just got the chance install emonHub and put it in my emonPi to test this out.
Happily, I can confirm that it worked immediately:
[image]
Great work!

Fourdee, 2016-03-29:
@R1CH-T 
No worries rich, I appreciate you testing the image/install and good to hear its all working. So many thanks :+1: 

Ordered a EmonTH node to help implement RF features (and have fun checking temp/humidity around the house lol). 
I will create another installation option for this, so the user can decide if they need RF or want to save some resources. Basically it will be a "addon" installation for the current EmonPi installation.

DietPi v114 was released yesterday which enables the EmonPi installation. I've updated the OEM post: https://openenergymonitor.org/emon/node/12384

PS. OEM are looking to enable the backup/restore features on Emoncms.org:
> We have an Emocncms backup module, it is not currently enabled on emoncms.org, but we hope to add it in the near future.
>

MichaIng, 2021-05-10:
Hey guys, just in case someone using EmonPi is subscribed to this issue:
- I'm switching our emonHub implementation back to the upstream branch, with years of new features and bugfixes. It meanwhile allows to use any config directory: https://github.com/MichaIng/DietPi/pull/4357
- I recognised that data can be sent via MQTT (Mosquitto) as well, additionally or as alternative to emonCMS. And we provide a Mosquitto install option as well. Shall we install Mosquitto as a dependency together with emonCMS or at least give users the choice when emonHub is selected for install?
- And I'm rethinking to add the emonCMS install option. There are other reasons why users might have a webserver installed already, so that it's no much overhead anymore. The regular disk writes should be solved via Redis RAM-based database, if I'm not mistaken, which is available via dietpi-software as well.

Last but not least, if someone is able to test the change (above linked pull request) with an actual emonPi, that would be great. Install process and service starts fine, but without the hardware, I cannot test data collection and emoncmg.org forwarding connection itself.