MichaIng/DietPi #211: DietPi-Software | Add ability to test Filesystem benchmarks on non-standard rootFS location.

k-plan, 2016-02-27:
@Fourdee 

have notice, in [v110](https://github.com/Fourdee/DietPi/issues/200#issuecomment-185931543) it will be assisted to run DietPi on non-standard rootFS location.
Will/would be nice, if `dietpi-config` -> `9 Tools` -  `Filesystem benchmarks` not only can/will test SD-Cards (/dev/mmcblk0)

[image]
[image]

Fourdee, 2016-02-27:
@k-plan 
Good spot :+1: 

Resolved by using the following code in dietpi-config:
```
benchmark_filesystem_freespace=$(( $(df -m | grep -m1 "$(sed -n 4p /DietPi/dietpi/.hw_model)" | awk '{print $4}') - $benchmark_filesystem_offset ))
```

The 4th line of ```/DietPi/dietpi/.hw_model```, which runs during boot, contains the rootFS device name. So we parse ```df``` and match with that value.
```/DietPi/dietpi/.hw_model``` is my way of having a "c++ class style system" for bash, using files :)

#### Would appreciate if you could test the fix, just to make sure:

If this is a new install, you will also need to update these:
```
#Update network script
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/func/obtain_network_details -O /DietPi/dietpi/func/obtain_network_details
/DietPi/dietpi/func/obtain_network_details

#Update HW model script
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/dietpi-obtain_hw_model -O /DietPi/dietpi/dietpi-obtain_hw_model
/DietPi/dietpi/dietpi-obtain_hw_model
```

Update/run DietPi-Config:
```
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/dietpi-config -O /DietPi/dietpi/dietpi-config
dietpi-config
```

k-plan, 2016-02-27:
@Fourdee 

> If this is a new install, you will also need to update these:
> ...
> Update/run DietPi-Config:

Done.

[image]

All 3 benchmarks run without a error message, but no result will print out in menu.

```
root@oDroid-C1:~# benchmark_filesystem_freespace=$(( $(df -m | grep -m1 "$(sed -n 4p /DietPi/dietpi/.hw_model)" | awk '{print $4}') - $benchmark_filesystem_offset ))
-bash: 1732 -  : Syntax Fehler: Operator erwartet. (Fehlerverursachendes Zeichen ist \"-  \").
```

```
root@oDroid-C1:~# cat /DietPi/dietpi/.hw_model
10
oDroid C1
3
/dev/sdb1
c743f338-108e-4344-9a7b-9c7f79f8298f

```
```
root@oDroid-C1:~# blkid
/dev/mmcblk0: PTUUID="000cde21" PTTYPE="dos"
/dev/mmcblk0p1: SEC_TYPE="msdos" LABEL="BOOT" UUID="6E35-5356" TYPE="vfat" PARTUUID="000cde21-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="e139ce78-9841-40fe-8823-96a304a09859" TYPE="ext4" PARTUUID="000cde21-02"
/dev/sda1: LABEL="USB_8GB" UUID="65019e81-2d0b-0130-9bf7-0527ad7fbaa0" TYPE="ext4"
/dev/sdb1: UUID="0677b3d1-9217-435b-8cde-4a0dfb9a0695" TYPE="ext4" PARTUUID="17497643-01"
```
```
root@oDroid-C1:~# df -h
Dateisystem                              Größe Benutzt Verf. Verw% Eingehängt auf
/dev/sdb1                                 3,1G    1,2G  1,7G   42% /
udev                                       10M       0   10M    0% /dev
tmpfs                                     190M    5,4M  185M    3% /run
tmpfs                                     475M       0  475M    0% /dev/shm
tmpfs                                     5,0M       0  5,0M    0% /run/lock
tmpfs                                     475M       0  475M    0% /sys/fs/cgroup
tmpfs                                      10M    1,2M  8,9M   12% /DietPi
tmpfs                                     475M     16K  475M    1% /tmp
/dev/mmcblk0p1                            129M     25M  104M   20% /boot
/dev/sda1                                 7,1G    637M  6,1G   10% /mnt/usb_1
```

Fourdee, 2016-02-27:
@k-plan 
Interesting, that line is a comment ```#Wifi Extras```. Looks like something managed to escape quotes and cause havoc on the scripts code.

Can you run these for me please, and let me know results (creates a 10mb file in /root/10mb_test):
```
dd bs=4K count=$(( 10 * 1024 / 4 )) if=/dev/zero of=/root/10mb_test conv=fdatasync 2>&1
```

```
dd bs=4K count=$(( 10 * 1024 / 4 )) if=/dev/zero of=/root/10mb_test conv=fdatasync 2>&1 | awk '/copied/ {print $8 " "  $9}'
```

k-plan, 2016-02-27:
@Fourdee 

```
root@oDroid-C1:~# dd bs=4K count=$(( 10 * 1024 / 4 )) if=/dev/zero of=/root/10mb_test conv=fdatasync 2>&1
2560+0 Datensätze ein
2560+0 Datensätze aus
10485760 Bytes (10 MB) kopiert, 0,912823 s, 11,5 MB/s
```

```
root@oDroid-C1:~# dd bs=4K count=$(( 10 * 1024 / 4 )) if=/dev/zero of=/root/10mb_test conv=fdatasync 2>&1 | awk '/copied/ {print $8 " "  $9}'
root@oDroid-C1:~#
```

Fourdee, 2016-02-27:
@k-plan 
Thanks,

As i suspected, ```awk``` and de_DE.UTF-8 issue :)

Can't use ```grep 'bytes'``` as en_GB is ```Bytes```. So left with:
```
dd bs=4K count=$(( 10 * 1024 / 4 )) if=/dev/zero of=/root/10mb_test conv=fdatasync 2>&1 | grep 'MB)' | awk '{print $8, $9}'
```

k-plan, 2016-02-27:
@Fourdee 

```
root@oDroid-C1:~# dd bs=4K count=$(( 10 * 1024 / 4 )) if=/dev/zero of=/root/10mb_test conv=fdatasync 2>&1 | grep 'MB)' | awk '{print $8, $9}'
11,2 MB/s
```

Fourdee, 2016-02-27:
@k-plan 
Thanks. Should now resolve the issue:
```
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/dietpi-config -O /DietPi/dietpi/dietpi-config
dietpi-config
```

k-plan, 2016-02-27:
@Fourdee 

Nice! :)

[image]

:+1: 

- Cosmetic: - may be now a suitable term for `SDcard` and `USBHDD` ? _(eg. root File System , USB Device ?)_ But for me, you can leave well enough alone.

Fourdee, 2016-02-28:
@k-plan 
Excellent, thanks for testing.

> Cosmetic: - may be a suitable term for SDcard and USBHDD now? (eg. root File System , USB Device ?) (idea: - or you can do 4 options? rootFS, SD-Card/emmc, USB Device, RAM?)

Good feedback, :+1: . I've changed the naming and descriptions based on your recommendations:
[image]

As for additional options:
99% of our users are using SD/EMMC, around 50% also use the USB HDD combination. I dont want to overcomplicate and possibly confuse our users with additional options at this time.

As the original issue has now been resolved, I'll mark this ticket as completed. Please reopen if required.

closed 2016-02-28 by Fourdee