MichaIng/DietPi #213: External Drive Does Not Mount

rafasamp, 2016-02-28:
I've been trying to figure this out for two days with no avail. I have an external USB drive that is being seen by DietPi but for some reason I can't mount it. From what we can see below it is indeed supposed to be mounted to /mnt/usb_1 however as you can see from the DF output all the data is being written on the local flash card.

Can anyone help me figure this out? I am at a complete loss.

fdisk -l output:
```
Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        8192   122879   114688   56M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      122880 60751871 60628992 28.9G 83 Linux

Disk /dev/sda: 1.8 TiB, 2000398931968 bytes, 3907029164 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x58f8f6e5

Device     Boot Start        End    Sectors  Size Id Type
/dev/sda1        2048 3907029163 3907027116  1.8T 83 Linux
```

**fstab**
```
proc            /proc           proc    defaults          0       0
/dev/mmcblk0p1  /boot           vfat    defaults,noatime  0       2
/dev/mmcblk0p2  /               ext4    defaults,noatime  0       1
tmpfs                   /tmp                    tmpfs   defaults,noatime,nodev,nosuid,mode=1777  0 0
tmpfs                   /var/log                tmpfs   defaults,size=20m,noatime,nodev,nosuid,mode=1777  0 0
tmpfs                   /DietPi                 tmpfs   defaults,size=10m,noatime,nodev,nosuid,mode=1777  0 0
/dev/sda1       /mnt/usb_1      ext4    defaults,noatime,nofail  0       0
```

**df output**
```
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       29772764 5647252  24109128  19% /
devtmpfs          494108       0    494108   0% /dev
tmpfs             498412       0    498412   0% /dev/shm
tmpfs             498412    6632    491780   2% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             498412       0    498412   0% /sys/fs/cgroup
tmpfs              10240    1052      9188  11% /DietPi
tmpfs              20480      52     20428   1% /var/log
tmpfs             498412      32    498380   1% /tmp
/dev/mmcblk0p1     57102   20708     36394  37% /boot
```

Fourdee, 2016-02-28:
@rafasamp 

Sounds like there are existing files in /mnt/usb_1 preventing the mount from occurring, or another issue.

Please paste results:
```
umount /mnt/usb_1
ls -lha /mnt/usb_1
```

rafasamp, 2016-02-28:
Aha! Apparently that was the issue. If anyone out there runs into the same problem, this is what I did:

`dd if=/dev/zero of=/dev/sda1 bs=512 count=1`

This apprently erased the faulty MBR that was in the drive and now df outputs a much happier:

```
Filesystem      1K-blocks    Used  Available Use% Mounted on
/dev/root        29772764 1015016   28741364   4% /
devtmpfs           494108       0     494108   0% /dev
tmpfs              498412       0     498412   0% /dev/shm
tmpfs              498412    6628     491784   2% /run
tmpfs                5120       4       5116   1% /run/lock
tmpfs              498412       0     498412   0% /sys/fs/cgroup
tmpfs               10240    1052       9188  11% /DietPi
tmpfs               20480     132      20348   1% /var/log
tmpfs              498412      44     498368   1% /tmp
/dev/mmcblk0p1      57102   20708      36394  37% /boot
/dev/sda1      1922728836  171356 1824865420   1% /mnt/usb_1

```

Thanks Fourdee, all I needed was a nudge ;)

rafasamp, 2016-02-28:
I'm reopening because of something funny I'm seeing. I am hoping my drive is not dying but this is what happens when I ran the external drive utility from dietpi:

```
umount: /dev/sda1: not mounted
1+0 records in
1+0 records out
512 bytes (512 B) copied, 0.00378651 s, 135 kB/s

Welcome to fdisk (util-linux 2.25.2).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Device does not contain a recognized partition table.
Created a new DOS disklabel with disk identifier 0xeefa648b.

Command (m for help): Created a new DOS disklabel with disk identifier 0xdee015d4.

Command (m for help): Partition type
   p   primary (0 primary, 0 extended, 4 free)
   e   extended (container for logical partitions)
Select (default p): Partition number (1-4, default 1): First sector (2048-3907029163, default 2048): Last sector, +sectors or +size{K,M,G,T,P} (2048-3907029163, default 3907029163):
Created a new partition 1 of type 'Linux' and of size 1.8 TiB.

Command (m for help): The partition table has been altered.
Calling ioctl() to re-read partition table.
/dev/sda: close device failed: Remote I/O error

umount: /dev/sda1: not mounted
mke2fs 1.42.12 (29-Aug-2014)
Creating filesystem with 488378389 4k blocks and 122101760 inodes
Filesystem UUID: 8c416bd0-0b7e-4a49-b533-8f181d6510ad
Superblock backups stored on blocks:
        32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208,
        4096000, 7962624, 11239424, 20480000, 23887872, 71663616, 78675968,
        102400000, 214990848

Allocating group tables: done
Writing inode tables: done
Creating journal (32768 blocks): done
Writing superblocks and filesystem accounting information: done

resize2fs 1.42.12 (29-Aug-2014)
The filesystem is already 488378389 (4k) blocks long.  Nothing to do!

Mount is denied because the NTFS volume is already exclusively opened.
The volume may be already mounted, or another software may use it which
could be identified for example by the help of the 'fuser' command.
```

What happens is, as soon as the utility finishes doing its thing, the drive mounts fine. However, it mounts it as read-only. When I reboot the system the drive disappears again.

```
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       29772764 1117452  28638928   4% /
devtmpfs          494108       0    494108   0% /dev
tmpfs             498412       0    498412   0% /dev/shm
tmpfs             498412    6632    491780   2% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             498412       0    498412   0% /sys/fs/cgroup
tmpfs              10240    1052      9188  11% /DietPi
tmpfs             498412      32    498380   1% /tmp
tmpfs              20480     140     20340   1% /var/log
/dev/mmcblk0p1     57102   20708     36394  37% /boot
```

Fourdee, 2016-02-28:
@rafasamp 

> dd if=/dev/zero of=/dev/sda1 bs=512 count=1

Yep, we use this during ext4 format in ```dietpi-external_drive_setup```: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-external_drive_setup#L166-L167

> /dev/sda: close device failed: Remote I/O error

Sounds like there may be a underlying hardware error on the drive. If there is a corruption of data on the drive, this would cause it to mount ext4 as read only.

Could also be power related, i'll need some more info:
Is your HDD a
- 2.5 or 3.5 inch
- SSD or platter
- How is it powered (eg: usb on RPi or dedicated PSU)

rafasamp, 2016-02-28:
It's a 3.5 Seagate External with it's own power supply. I was using it with Windows up until Fridaywith no problems at all. I wonder if there is something going on with the way it's getting formatted?

Fourdee, 2016-02-29:
@rafasamp 

> I wonder if there is something going on with the way it's getting formatted?

Try running through the format command steps manually, see if any of the commands return errors:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-external_drive_setup#L164-L177

rafasamp, 2016-02-29:
```
root@rafaelsampaio:~# umount /dev/sd[a-z][1-9]
umount: /dev/sda1: not mounted
```

```
root@rafaelsampaio:~# dd if=/dev/zero of=/dev/sda bs=512 count=1
1+0 records in
1+0 records out
512 bytes (512 B) copied, 0.00299307 s, 171 kB/s
```

```
root@rafaelsampaio:~# echo -e "o\nn\np\n1\n\n\nw" | fdisk /dev/sda

Welcome to fdisk (util-linux 2.25.2).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Device does not contain a recognized partition table.
Created a new DOS disklabel with disk identifier 0xe150b6ae.

Command (m for help): Created a new DOS disklabel with disk identifier 0x769f0617.

Command (m for help): Partition type
   p   primary (0 primary, 0 extended, 4 free)
   e   extended (container for logical partitions)
Select (default p): Partition number (1-4, default 1): First sector (2048-3907029163, default 2048): Last sector, +sectors or +size{K,M,G,T,P} (2048-3907029163, default 3907029163):
Created a new partition 1 of type 'Linux' and of size 1.8 TiB.

Command (m for help): The partition table has been altered.
Calling ioctl() to re-read partition table.
/dev/sda: close device failed: Remote I/O error
```

```
root@rafaelsampaio:~# echo -e "y" | mkfs.ext4 /dev/sda1
mke2fs 1.42.12 (29-Aug-2014)
Found a dos partition table in /dev/sda1
Proceed anyway? (y,n) Creating filesystem with 488378389 4k blocks and 122101760 inodes
Filesystem UUID: f02b8883-56aa-4b32-9510-2a7765a4e6a8
Superblock backups stored on blocks:
        32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208,
        4096000, 7962624, 11239424, 20480000, 23887872, 71663616, 78675968,
        102400000, 214990848

Allocating group tables: done
Writing inode tables: done
Creating journal (32768 blocks): done
Writing superblocks and filesystem accounting information: done
```

```
root@rafaelsampaio:~# resize2fs /dev/sda1
resize2fs 1.42.12 (29-Aug-2014)
The filesystem is already 488378389 (4k) blocks long.  Nothing to do!
```

```
root@rafaelsampaio:~# mount -a
root@rafaelsampaio:~# chown -R root:root /mnt/usb_1
```

Then when I run DF, the drive is there:

```
root@rafaelsampaio:~# df
Filesystem      1K-blocks    Used  Available Use% Mounted on
/dev/root        29772764 1015052   28741328   4% /
devtmpfs           494108       0     494108   0% /dev
tmpfs              498412       0     498412   0% /dev/shm
tmpfs              498412    6632     491780   2% /run
tmpfs                5120       4       5116   1% /run/lock
tmpfs              498412       0     498412   0% /sys/fs/cgroup
tmpfs               10240    1052       9188  11% /DietPi
tmpfs               20480     140      20340   1% /var/log
tmpfs              498412      32     498380   1% /tmp
/dev/mmcblk0p1      57102   20708      36394  37% /boot
/dev/sda1      1922728836   68956 1824967820   1% /mnt/usb_1
```

However, I can't do anything to it and after a reboot, it's gone again... so weird!
```
root@rafaelsampaio:~# cd /mnt/usb_1/
root@rafaelsampaio:/mnt/usb_1# mkdir test
mkdir: cannot create directory ‘test’: Read-only file system
```

Fourdee, 2016-02-29:
@rafasamp 
Is the drive read only straight after you ran the format?

rafasamp, 2016-02-29:
Yeah, I did all the steps in order before the reboot. It was read only right after the chown command

rafasamp, 2016-02-29:
So apparently there really is something wrong with the formatting.

```
[  193.315749] blk_update_request: critical target error, dev sda, sector 0
[  193.315883] JBD2: recovery failed
[  193.315942] EXT4-fs (sda1): error loading journal
```

I'm at a loss, the disk works great in Windows. Maybe I'll try NTFS and see how it goes...

rafasamp, 2016-02-29:
Well, I guess I have no choice but to stay in NTFS. It actually works! Go figure...

@Fourdee thanks for all the help, the r/w performance may not be the best under NTFS but for a music server I think it will be plenty. Hopefully on my next Pi build I can get a proper external drive that can support EXT4.

closed 2016-02-29 by rafasamp

Fourdee, 2016-02-29:
@rafasamp 
Yep, this is really bizarre, doesn't make sense lol. 

It might be worth testing your drive with EXT4 on the official Raspbian image, just to exclude anything with DietPi. I'am sure it would have the same results, but would be good to confirm.

Another option may be to try ```dd``` zero's to the whole drive, prior to formatting EXT4.

Either way, NTFS will work, but as you mentioned, lower performance due to increased CPU usage.