MichaIng/DietPi #214: DietPi-Update trough a proxy

mok-liee, 2016-02-29:
Hello, there is no option to Update DietPi trough a http proxy.

Fourdee, 2016-02-29:
@mok-liee 
Correct.

You will need to add your proxy settings at a linux level:
https://www.raspberrypi.org/forums/viewtopic.php?p=184490#p184490

mok-liee, 2016-02-29:
yes, i know. but this is only for "apt"
the dietpi-updater is launching a check on github using another tool, i think wget or something.

i have entered the export var and did apt-get update successfully but dietpi-update doesnt run.

or die i something wrong?

Fourdee, 2016-02-29:
@mok-liee 
Ah, i see.

DietPi-Update uses ```wget``` and ```curl```.

Try these for specifying proxy settings:
```wget``` add proxy: http://stackoverflow.com/questions/11211705/setting-proxy-in-wget
```curl``` add proxy: http://stackoverflow.com/questions/7559103/how-to-setup-curl-to-permanently-use-a-proxy

mok-liee, 2016-02-29:
oh, cool! very helpful.

maybe in the next version u can add an proxy setter in dietpi-config -> networking options

thanks!

Fourdee, 2016-02-29:
@mok-liee 

> maybe in the next version u can add an proxy setter in dietpi-config -> networking options

Added to the list :+1:

Fourdee, 2016-04-09:
> http://www.shellhacks.com/en/HowTo-Use-a-Proxy-on-the-Linux-Command-Line

Fourdee, 2016-04-09:
> note to self:
Coding is done for system-wide http/https/ftp export variables. Just need to test proxy settings are being used.
Commit https://github.com/Fourdee/DietPi/commit/2349756ddc902c40a55f8b5286131ea3ded07277

Fourdee, 2016-04-10:
Working fine:
```
root@DietPi:~# wget  http://speedtest.wdc01.softlayer.com/downloads/test500.zip
--2016-04-10 12:55:58--  http://speedtest.wdc01.softlayer.com/downloads/test500.zip
Connecting to 188.203.209.91:3128... connected.
Proxy request sent, awaiting response... 200 OK
Length: 524288000 (500M) [application/zip]
Saving to: ‘test500.zip’
```

Fourdee, 2016-04-10:
@mok-liee 
Any chance you would be willing to help test the proxy implementation? Just let me know and I'll give you the steps needed to install DietPi from the dev (testing) branch.

mok-liee, 2016-04-10:
no problem, in about 10 hours i have chance to test it.
looks very nice! great work

Fourdee, 2016-04-10:
@mok-liee 
Excellent. Here is how you can install the dev branch:

- Write DietPi image to SD
- Open /boot/dietpi.txt
- Change gitbranch to ```gitbranch=testing``` (its at the bottom)
- Save file, plug in and power on.

The proxy settings are in ```dietpi-config``` > Network Adapters > Proxy

mok-liee, 2016-04-11:
@Fourdee 
hi, very cool!

it works for me, but i has used another way.

the image on dietpi.com is at v111. I have downloaded and write this to my sd card.
Becuase i need to go through a proxy from the begining, i have downloaded the testing branch directly from guthub.

after extracting it to my sd card (v111) i have the new menu!

i have configured the proxy settings and after a reboot my dietpi go through my proxy.

wheyyy... very nice!!!

i have a few ideas for finishing this:

-  after enabling the proxy settings i think an modal for rebooting the device would be cool because its necessary to reboot.
- add a "proxy settings status" in the menu "Network Options: Adapter" 
- set default port to 8080, i think it is better (i commented the codeline already)
- maybe u can do a new final image for v116 an link it on dietpi.com

i hope my english is okay for u to understand and not to cheeky

greetings mok

Fourdee, 2016-04-11:
@mok-liee 
Thanks for testing and the great feedback :+1: 

#### I'll make the following changes based on your recommendations:
- Make a reboot request and remove the relogin prompt.
- Add proxy status (eg: on/off) in the previous submenu (network options: adapter)
- Default port 8080

I need to re-do all the DietPi images at some point. But if you can let me know which device/image you are using, i'll try and get that one done ASAP.

> i hope my english is okay for u to understand and not to cheeky

Not at all. All feedback is good feedback, helps make DietPi better :+1:

mok-liee, 2016-04-11:
i use the RPi Image, but its okay! this soloution i used yesterday worked for me. do not be held back by me.

thanks!

Fourdee, 2016-04-11:
@mok-liee 
When you copy the scripts from dev branch, make sure you do NOT copy ```/boot/dietpi/.version```. Otherwise, your system will not be updated correctly by the patch system.

mok-liee, 2016-04-11:
ahh... thanks for the note

Fourdee, 2016-04-12:
@mok-liee 

> - Make a reboot request and remove the relogin prompt.
- Add proxy status (eg: on/off) in the previous submenu (network options: adapter)
- Default port 8080

All done and thanks for testing this for us :+1: . Available in v116. Will set this to completed, please reopen as required.

closed 2016-04-12 by Fourdee