MichaIng/DietPi #217: Odroid C2 | Add support and optimizations in DietPi

Fourdee, 2016-03-01:
With the release of Pine64 and RPiv3, I currently lack funding at the moment for the C2 (£50~) purchase. Could be a month or two before that changes.

Gutted.

k-plan, 2016-03-01:
@Fourdee 

> I currently lack funding at the moment for the C2 (£50~) purchase. Could be a month or two before that changes.

Sorry, disagreeable to hear.
Will a small [PayPal donation](https://www.paypal.com/cgi-bin/webscr?cmd=_s-xclick&hosted_button_id=FPH645FBNC7K2) help, to get it faster?

- RPiv3: Nice, available, great provide support, main stream. Can understand, that you will support it quick. But for me, no significant improvement to Odroid C1+ at the moment.

- Pine64: Preoder Ship by May 2016, first not for everybody available. But realy nice device.

- Odroid C2: Will be dispatch in 7 days. Amazing  specifications and performance comparison

 	True for all: ARM 64bit is a very new platform and some system specific Linux softwares are not working stably at this moment.

Fourdee, 2016-03-01:
@k-plan 

> Will a small PayPal donation help, to get it faster?

That's very kind of you to offer. The C2 is currently being sold for £45 from a UK/EU supplier http://www.odroid.co.uk/hardkernel-odroid-c2-board. Any donations received would be put towards its purchase.

> RPiv3: Nice, available, great provide support, main stream. Can understand, that you will support it quick. But for me, no significant improvement to Odroid C1+ at the moment.

I completely agree. 
The RPiv3 is still miles away from the C1 in terms of performance. RPi still has eth running through the USB 2.0 bus. I was hoping the RPi v3 would get a USB3 bus upgrade, allowing 1gbit eth and less performance impact on system IO.
Also turns out the RPi v3 built in wifi runs through SDIO: https://ubuntu-mate.org/blog/ubuntu-mate-for-raspberry-pi-3/
No improvements to the SD card read/write speed either.

> Pine64: Preoder Ship by May 2016, first not for everybody available. But realy nice device.

Seems the RPi v3 release was very "timely" ;) Mine also arrives in May.

> Odroid C2: Will be dispatch in 7 days. Amazing specifications and performance comparison

Yep, its in a league of its own. The Odroid C1 is my favourite SBC of all time, the C2 may take its place :)

Fourdee, 2016-03-04:
@k-plan 
:+1: Thanks for helping with this, made the purchase a little easier :)
I haven't told the misses yet, wish me luck lol ;)
[image]

misury, 2016-03-05:
Rock on! I am so glad to know you are getting into the C2! Mine should be arriving early next week. Thank you for all your hard work and efforts friend.

k-plan, 2016-03-06:
@Fourdee 

> Thanks for helping with this, made the purchase easier on the wallet :)

I am pleased to see, it's make your decision easy. 
Hope for you, more DietPi user follow the lead, so that it is possible for you, to buy more upcoming hardware ...

rotational467, 2016-03-08:
I wish I had seen this a few days ago.  Pi-hole gave me a reason to pick up a Pi on sale, which led me to Diet Pi, and your forums led me to the C2 that will hopefully be delivered this week :)  DietPi is exactly the kind of image I was looking for for what I want to do, and I already have a couple of installs of it running.

I can't contribute with code at this time so donation gladly sent!

Fourdee, 2016-03-08:
@rotational467 
Thank you for your feedback and donation, I appreciate it! :+1: 

I'am still waiting for my C2 to arrive. It should hopefully be here before the weekend so I can crack on with adding a DietPi image for it. Cant wait, makes me feel like a child at xmas again lol.

Fourdee, 2016-03-08:
Just got an email from Odroid UK. They are out of stock and have no ETA on restock. Looks like the DietPi for Odroid C2 is going to be delayed.

http://forum.odroid.com/viewtopic.php?f=135&t=19340

rotational467, 2016-03-09:
My pleasure!  Glad to see that you were able to find another vendor who actually had some.  I'm in the US and ordered Monday morning from ameridroid, who as far as I can tell have them in stock but are warning about 1-2 days processing delay, so I'm hoping to see it by Saturday :)  I know what you mean about waiting for xmas!

Fourdee, 2016-03-14:
Odroid C2 arriving mid-week, cant wait.

k-plan, 2016-03-15:
@Fourdee 

wow, great news! Have you get a shipping or tracking notice from Diigiit Robotics Europe?

Shevraar, 2016-03-15:
first of all thank you @Fourdee for your work, I'm eagerly waiting for my odroid to come and I want to install DietPi :+1: @k-plan I ordered it from Diigiit Robotics Europe and they have shipped it on 12-03.
It should be arriving in these days. I got a shipping number as well.

Fourdee, 2016-03-15:
Arrived today and work has started :racehorse:

k-plan, 2016-03-16:
@Shevraar 
thanks for the info. :+1: 

@Fourdee 
I am very pleased to hear, that  C2 perfectly satisfied your expectations.

rotational467, 2016-03-16:
@Fourdee 

That's awesome news!  I got mine on Monday and yeah this thing is a real powerhouse even on microSD, until my eMMC gets here.  Enjoy!!! :+1:

Fourdee, 2016-03-16:
@k-plan 
So far its exceeding my expectations :)

@rotational467 
Ah nice one. EMMC makes a huge difference to filesystem IO and overall performance, its like swapping a platter HDD for a SSD.

Fourdee, 2016-03-17:
@k-plan @rotational467 @Shevraar @misury 
DietPi for Odroid C2 has been released under beta as we have a few issues to iron out with Kodi.

Download image and more info: http://forum.odroid.com/viewtopic.php?f=138&t=19948

Fourdee, 2016-03-17:
#### Workaround for sound in Kodi:
http://forum.odroid.com/viewtopic.php?f=138&t=19948&p=130983#p130983

rotational467, 2016-03-17:
@Fourdee 

Perfect timing, my eMMC module arrived today!  Thank you for the all of the hard work!

rotational467, 2016-03-18:
@Fourdee 

For some reason, I can't get a good HDMI signal under the current beta release.  I've got a 1680x1050 DVI monitor with a DVI -> HDMI adapter.  Have no problems with any of my RPi's with this setup or with Meveric's base image or the official odroid Ubuntu image (although I have to manually define my resolution in the config.txt to get a native output).  I don't think the either the RPi or the odroid is able to parse the EDID on this monitor.

The out of the box settings (1280x720p) give me no video signal at all.  Playing around with the resolution and hdmi flags in the boot.ini and config.txt can get me some output, but the image is full of artifacts and not at the specified resolution.  The image is corrupt in both the console and X.

Please let me know what info I can provide to help.  Note that I can do a remote X session over VNC at any resolution without a problem.

Fourdee, 2016-03-18:
Can you paste your '''/DietPi/boot.INI?''' Also, try commenting out the dvi flag in boot.INI.

Date: Thu, 17 Mar 2016 19:20:41 -0700

k-plan, 2016-03-18:
@Fourdee 
have also some trouble to get it started on C2 with a 4:3 PC-Monitor (native Display resolution: 1280x1024) and DVi-to-HDMI cable. Proved settings form C1+ didn't work.

Solution for me:
```
root@oDroid-C2:~# cat /DietPi/boot.ini

#Modifcations to /boot/* files, will not be saved when powered on.
#Please ensure you edit from the DietPi-RamDisk location /DietPi/*

ODROIDC2-UBOOT-CONFIG

#Resolution + Refresh rate
# setenv m "720p60hz"    ## P19 Monitor => signal on C1+ but NOT on C2
# setenv m "sxga"               ## (SXGA) work only on C1+ => P19 Monitor
setenv m "1280x1024p60hz"       ## (SXGA) work on C2  => P19 Monitor

# HDMI BPP Mode | 24bit to ensure Kodi and Xserver both work, whilst we wait for a fix from Odroid.
setenv m_bpp "24"

# HDMI DVI/VGA modes
# Uncomment only a single Line! The line with setenv written.
# At least one mode must be selected.
# setenv vout "dvi"     ## DVI-to-HDMI cable (SXGA) work on C2 and C1+  => P19 Monitor
setenv vout "vga"       ## DVI-to-HDMI cable (SXGA) work only on C2        => P19 Monitor

# Default Console Device Setting
setenv condev "console=ttyS0"

###########################################

# Boot Arguments
setenv bootargs "root=UUID=e139ce78-9841-40fe-8823-96a304a09859 rootwait rw ${condev} no_console_suspend hdmimode=${m} m_bpp=${m_bpp} vout=${vout} fsck.repair=yes net.ifnames=0"

# Booting

setenv loadaddr "0x11000000"
setenv dtb_loadaddr "0x1000000"
setenv initrd_loadaddr "0x13000000"

fatload mmc 0:1 ${initrd_loadaddr} uInitrd
fatload mmc 0:1 ${loadaddr} Image
fatload mmc 0:1 ${dtb_loadaddr} meson64_odroidc2.dtb

booti ${loadaddr} ${initrd_loadaddr} ${dtb_loadaddr}
```

@rotational467 

tried to edit `/DietPi/boot.ini` with `setenv m "1680x1050p60hz"` ? _(not `/boot/boot.ini`)_

`setenv vout "dvi"` or `setenv vout "vga"` makes no odds with my monitor and DVI-to-HDMI cable.

I didn't edit `config.txt`. I think, it's works on RPi only.

rotational467, 2016-03-18:
Hi guys,

Thank you for the suggestions.  I just flashed a clean image to start over.  Setting only `"setenv m 1680x1050p60hz"` gets me a corrupted image.  The key is the color depth (the one thing I didn't mess with last night of course)...setting `setenv m_bpp "32"` gives me a clean image at the correct resolution.  I'm not worried about Kodi so this is not an issue for me.  I'm sure this is just something goofy with this monitor, it's an older unit.

I know they're still working with the SoC vendor on the video driver so it's all good.  I do notice that you appear to have the hardware cursor working in the console which is awesome!

Fourdee, 2016-03-20:
Although DietPi for C2 is in beta, I will marked as completed here. Further bug reports should be created with a new Git ticket.

Please check the list of known issues, before creating a bug report:
http://forum.odroid.com/viewtopic.php?f=138&t=19948&sid=7f255a9875aa0043b60c29c69fd2a713

I will continue to improve and work on DietPi for C2. Patches and fixes will be applied during dietpi-update once available.

closed 2016-03-20 by Fourdee