MichaIng/DietPi #218: PHP APC Warning

rafasamp, 2016-03-01:
Hi!

When setting up a PHP instance to run Laravel or any other PHP framework we need to use composer and the CLI a lot. There is a bug currently whenever you run the CLI that states:

`PHP Warning:  PHP Startup: apc.shm_size now uses M/G suffixes, please update your ini files in Unknown on line 0`

To fix it, we need to edit /etc/php5/cli/conf.d/20-apcu.ini and add an M after the apc.shm_size memory limit so it looks like this:

```
extension=apcu.so
apc.shm_size=32M
apc.ttl=259200
```

I believe this is the place in the code that would need to be edited in order to add the M after the apc.shm_size variable:

https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L2493-L2496

I can do a pull request if you'd like, but I didn't have a chance to look at the contributor docs yet.

Thanks

Fourdee, 2016-03-01:
@rafasamp 
Good find. Is this a wheezy or jessie installation?

> I can do a pull request if you'd like, but I didn't have a chance to look at the contributor docs yet.

There are no contributor docs yet as I'am the only person who has ever worked on the code directly, but we can change that :)

You will need to fork the testing branch as this is the current development branch: https://github.com/Fourdee/DietPi/tree/testing

Make your changes and send a pull request for me to review.

rafasamp, 2016-03-01:
Hi @Fourdee this is a jessie installation. I will make the changes tonight and will send it in.

Thanks!

k-plan, 2016-03-01:
@Fourdee 
@rafasamp 

`PHP Warning: PHP Startup: apc.shm_size now uses M/G suffixes, please update your ini files in Unknown on line 0`

Can confirm this issue with composer / PHP on Apache2, Nginx or Lighttp. (needed for [volkszaehler](http://wiki.volkszaehler.org/doku.php/software/middleware/installation) )

```
#nano /etc/php5/cli/conf.d/20-apcu.ini 

apc.shm_size=10M

```
Yes, this will fix error message, when restart the webserver.

> Is this a wheezy or jessie installation?

Yes,  jessie installation on RPi or Odroid C1.

Fourdee, 2016-03-03:
@k-plan 
Thanks, legend :+1:

Fourdee, 2016-03-04:
@k-plan 
The following files are all symlinked to ```/etc/php5/mods-available/apcu.ini``` when PHP5 is installed:
```
root@DietPi:/# find / -name *apcu.ini
/etc/php5/cli/conf.d/20-apcu.ini
/etc/php5/cgi/conf.d/20-apcu.ini
/etc/php5/fpm/conf.d/20-apcu.ini
/etc/php5/apache2/conf.d/20-apcu.ini
root@DietPi:/# ls -lha /etc/php5/cli/conf.d/20-apcu.ini
lrwxrwxrwx 1 root root 29 Mar  4 15:24 /etc/php5/cli/conf.d/20-apcu.ini -> ../../mods-available/apcu.ini
root@DietPi:/# ls -lha /etc/php5/cgi/conf.d/20-apcu.ini
lrwxrwxrwx 1 root root 29 Mar  4 15:24 /etc/php5/cgi/conf.d/20-apcu.ini -> ../../mods-available/apcu.ini
root@DietPi:/# ls -lha /etc/php5/fpm/conf.d/20-apcu.ini
lrwxrwxrwx 1 root root 29 Mar  4 15:24 /etc/php5/fpm/conf.d/20-apcu.ini -> ../../mods-available/apcu.ini
root@DietPi:~# ls -lha /etc/php5/apache2/conf.d/20-apcu.ini
lrwxrwxrwx 1 root root 29 Mar  4 15:39 /etc/php5/apache2/conf.d/20-apcu.ini -> ../../mods-available/apcu.ini
```

I've updated DietPi-Software to write directly to  ```/etc/php5/mods-available/apcu.ini```, with ```target_php_cachesize+="M"```:
```
				#APCu
				php5enmod apcu

				# - apc.shm_size= requires M at end to prevent warning: https://github.com/Fourdee/DietPi/issues/218
				target_php_cachesize+="M"
				target_php_ini='/etc/php5/mods-available/apcu.ini'

				cat << _EOF_ > "$target_php_ini"
extension=apcu.so
apc.shm_size=$target_php_cachesize
apc.ttl=$target_apc_ttl
_EOF_
```

Resolved for v112.

closed 2016-03-05 by Fourdee