MichaIng/DietPi #219: OPi-PC not reset

openmy, 2016-03-03:
my OPi-PC not reset (auto power on affter off ), now my reset opi-pc  power on off by hand.

Fourdee, 2016-03-03:
@openmy 

Not sure I follow. Does the OPi automatically power on after a shutdown? Or, it wont reboot?

Reboot
```reboot```

Poweroff:
```poweroff```

openmy, 2016-03-03:
OPi  not automatically power on after a shutdown.
And if i tun power on off by hand opi update  work. but v110 error for my.
how do I get off the auto update....... help

Fourdee, 2016-03-03:
@openmy 
I think we might have an issue with the language barrier. Would you be able to video record the issue? This will help me understand much better allow me to assist you.

openmy, 2016-03-04:
my video test. the auto update error. OPi not automatically power on after a shutdown
https://drive.google.com/folderview?id=0B5dH7Uzc8K-FVGJ0Z2NETy14Y3c&usp=sharing

Fourdee, 2016-03-05:
@openmy Thanks for the videos. I haven't had the chance to look at these yet, my apologies, hopefully will be able to in the next few days.

Fourdee, 2016-03-08:
@openmy 

Ok, thanks for the videos and apologies for the delay. I see a few issues:
- /DietPi/dietpi/login | reported as not found
- dietpi-banner | hangs after login

I was unable to replicate any of these issues on a fresh OPi install. Its most likely hardware related and looks like filesystem corruption.

Please answer the following:
- SDcard make/model. Have you tried other cards?
- PSU make/model/amperage. Have you tried other PSu's?
- Are you modifying any files on in ```/boot``, after writing the DietPi image and before powering on?
- Image write method or tool.

openmy, 2016-03-08:
* SDcard make/model : Strontium 8gb  Class 4, 
  Have you tried other cards ? : Yes
* PSU make/model/amperage: Samsung Ep-Ta10jWE, 5,3V - 2A.
Have you tried other PSu's ? : Yes (tosiba Tsp-1 5v -2A , very good power)
* Are you modifying any files on in `/boot, after writing the DietPi image and before powering on? => NO (just download and use )
* Image write method or tool.:
SD Formatter
Win32DiskImager

ps: My OPi Test android os in orangepi.org perfect run.

openmy, 2016-03-09:
My updated using sd memory card class 10 and it works ( tosiba  32gb class 10 )....
Thank Fourdee supported me.

Fourdee, 2016-03-09:
@openmy 
Great to hear its working with another SD card :+1: 

I'll mark this ticket as closed. Please reopen if required.

closed 2016-03-09 by Fourdee