MichaIng/DietPi #223: Question Re: Automated image builder workflow, no hardware required

spiderkeys, 2016-03-07:
My organization has been working with the Beaglebone Black for years now and we are used to building our custom Debian image on a remote build server which primarily uses chroot as the method of modifying the base image and getting all of the packages installed and updated. I was curious if there is an implemented/existing workflow for getting a DietPi image fully built and updated without having to burn the base image to an SD card, power it up, make sure it is connected to the internet, and run the update process? I saw the automatic configuration feature, but that just seems to cut out the manual config steps, still requiring me to use a physical RPi to update the image.

Thanks for any info!

Fourdee, 2016-03-07:
> P.s. Awesome work on DietPi; the base resource usage and boot times are impressive!

Thank you. Its good to hear. Alot of hard work has gone into making that possible :)

> I was curious if there is an implemented/existing workflow for getting a DietPi image fully built and updated without having to burn the base image to an SD card, power it up, make sure it is connected to the internet, and run the update process?

Currently, DietPi is designed to be used by the end user. A hardware device is required to complete the DietPi setup and installation.

When I create a DietPi image from scratch, I use an existing image (eg: Raspbian Jessie) and modify it to become the DietPi image. I do not currently build or compile the images.

All I can offer is the notes I use when setting up a DietPi image. In theory you could use those steps when you compile and build a Jessie image. The notes are a little messy, so if you are unsure of any steps, let me know:
```
#-------------------------------------------------------------------------
//DietPi on a fresh install
#-------------------------------------------------------------------------

#Jessie , unified apt sources.
rm /etc/apt/sources.list.d/deb-multimedia.list
# ------- NOT RPI -------------------------
cat << _EOF_ > /etc/apt/sources.list
deb http://ftp.debian.org/debian jessie main contrib non-free
deb http://ftp.debian.org/debian jessie-updates main contrib non-free
deb http://security.debian.org jessie/updates main contrib non-free
deb http://ftp.debian.org/debian jessie-backports main contrib non-free
_EOF_
# ------- NOT RPI -------------------------

#Remove following Jessie
apt-get purge dhcpcd5 libsqlite* libxapian22 lua5.1 luajit netcat-* make makedev ncdu plymouth openresolv shared-mime-in* tcpd strace tasksel* wireless-* xdg-user-dirs triggerhappy python* v4l-utils traceroute xz-utils ucf xauth zlib1g-dev xml-core aptitude* avahi-daemon rsyslog logrotate man-db manpages vim vim-common vim-runtime vim-tiny mc mc-data
apt-get autoremove --purge -y

#install
echo -e "CONF_SWAPSIZE=0" > /etc/dphys-swapfile
apt-get install -y resolvconf bc dbus bzip2 psmisc bash-completion cron whiptail sudo ntp ntfs-3g dosfstools parted hdparm pciutils usbutils zip htop wput wget fake-hwclock dphys-swapfile curl unzip ca-certificates console-setup console-data console-common keyboard-configuration wireless-tools wireless-regdb crda --no-install-recommends

#firmware
apt-get install firmware-realtek firmware-ralink firmware-brcm80211 firmware-atheros -y --no-install-recommends

#DIETPI STUFF
#Create folders
mkdir /DietPi
mkdir -p /mnt/usb_1
mkdir -p /mnt/samba
mkdir -p /mnt/ftp_client
echo -e "Samba client can be installed and setup by DietPi-Config.\nSimply run: dietpi-config 8" > /mnt/samba/readme.txt
echo -e "FTP client mount can be installed and setup by DietPi-Config.\nSimply run: dietpi-config 8" > /mnt/ftp_client/readme.txt

/boot/dietpi/dietpi-logclear 2

#FSTAB
cp /boot/dietpi/conf/fstab /etc/fstab

#setup dietpi service
echo 1 > /boot/dietpi/.install_stage
cp /boot/dietpi/conf/dietpi-service /etc/init.d/dietpi-service
chmod +x /etc/init.d/dietpi-service
update-rc.d dietpi-service defaults 00 80
service dietpi-service start

#Cron jobs
cp /DietPi/dietpi/conf/cron.daily_dietpi /etc/cron.daily/dietpi
chmod +x /etc/cron.daily/dietpi
cp /DietPi/dietpi/conf/cron.hourly_dietpi /etc/cron.hourly/dietpi
chmod +x /etc/cron.hourly/dietpi

#Crontab
cat << _EOF_ > /etc/crontab
#Please use dietpi-cron to change cron start times
SHELL=/bin/sh
PATH=/usr/local/sbin:/usr/local/bin:/sbin:/bin:/usr/sbin:/usr/bin
[... 139 lines cut ...]
_EOF_
systemctl enable rc-local.service
systemctl daemon-reload

#Shutdown SSH/Dropbear before reboot
cat << _EOF_ > /etc/systemd/system/kill-ssh-user-sessions-before-network.service
[Unit]
Description=Shutdown all ssh sessions before network
DefaultDependencies=no
Before=network.target shutdown.target

[Service]
Type=oneshot
ExecStart=/usr/bin/killall sshd && /usr/bin/killall dropbear

[Install]
WantedBy=poweroff.target halt.target reboot.target
_EOF_
systemctl enable kill-ssh-user-sessions-before-network
systemctl daemon-reload

dpkg-reconfigure tzdata
dpkg-reconfigure keyboard-configuration #Keyboard must be plugged in for this to work!
dpkg-reconfigure locales
```

spiderkeys, 2016-03-07:
Appreciate it! If we end up writing a method for making automated builds, I'll be sure to send a pull request your way.

Fourdee, 2016-03-07:
@spiderkeys 
No worries :+1:

spiderkeys, 2016-04-11:
Just wanted to provide an update on this. Just today, the image engineer at RPI put the image build scripts up at: https://github.com/RPi-Distro/pi-gen. I'm currently working through them and modifying them to build a completely configured DietPi image (which still uses a pre-defined DietPi configuration to choose what to install and what settings to use). If you're interested, I will share my progress for all to use. May need help with some of the DietPi specifics here and there, but this will be extremely valuable for us, and I'm sure others might be interested in rolling custom images as well.

Fourdee, 2016-04-12:
@spiderkeys 
Thanks for the info :+1: 

I will give this a spin when I can.

> If you're interested, I will share my progress for all to use. May need help with some of the DietPi specifics here and there, but this will be extremely valuable for us, and I'm sure others might be interested in rolling custom images as well.

Sounds good, i'll help you with any DietPi stuff/queries, just pop them in this git ticket.

Fourdee, 2016-04-26:
I'll mark this as closed. Please reopen as required.

closed 2016-04-26 by Fourdee

spiderkeys, 2016-05-02:
@Fourdee 
I've been working through this, and have been having some issues, which I believe to be mostly related to the ambiguity of where the DietPi files need to be.

In bold above your notes you write: "assumes you have all the DietPi files and folders (excluding github) in /boot/*"

However, at various parts in your script you write things like:

```
cp /DietPi/dietpi/conf/cron.daily_dietpi /etc/cron.daily/dietpi
```
```
cat << _EOF_ > /etc/rc.local
#!/bin/sh -e
/DietPi/dietpi/dietpi-services start
/DietPi/dietpi/dietpi-banner 0
```
Which seem to indicate that the files should have been placed in /DietPi/*, except that conf/dietpi-service definitely expects things to be in /boot/dietpi/.

Could I get some clarification on the directory structures? Is /DietPi/ just a copy of the files in /boot/ or maybe its the other way around?

Here is a link to my script that sets up this stage of the image build. I had all of my DietPi files (from the github repo) stored in ./files/DietPi and had a Raspbian Lite image built up in ${ROOTFS_DIR} at the time of executing this script:

https://github.com/OpenROV-Dev/pi-gen/blob/dev/stage3/00-dietpi-setup/00-run.sh

Thanks

Fourdee, 2016-05-02:
@spiderkeys 
The files are physically located in ```/boot/```. During boot, the files are copied to ramdisk ```/DietPi/```, during shutdown, the files are copied back to ```/boot/```. ```dietpi-service``` completes this task.

During the DietPi image setup, ```dietpi-service``` is started: https://github.com/OpenROV-Dev/pi-gen/blob/dev/stage3/00-dietpi-setup/00-run.sh#L76. This allows us to continue the setup on the ramdisk ```/DietPi/```, and call files from it. 

Hope that makes sense, if not, let me know and I'll get some more caffeine and try again :+1:  :)

spiderkeys, 2016-05-02:
Alright, thanks. I'll make the appropriate changes, give it a try, and provide updates soon.

spiderkeys, 2016-05-02:
Another question I suppose I'll field in advance:

After I finish getting the initial setup complete, can you guide me in the right direction of what scripts I'll need to call to automatically execute the functionality contained in the DietPi automation process (http://dietpi.com/phpbb/viewtopic.php?f=8&t=273)? I assume that it gets kicked off at boot based on a flag set somewhere, but didn't immediately find where. Any potential gotchas that you might think of would be helpful too.

Thanks :smile:

spiderkeys, 2016-05-02:
Made the changes so that all of the files are copied to boot. New script reflected here:

https://github.com/OpenROV-Dev/pi-gen/blob/feat_dietpi/stage3/00-dietpi-setup/00-run.sh#L81-L95

For the above lines, I'm running into the following issues error output:

```bash
+ chroot /home/openrov/workspace/pi-gen/work/2016-05-02-raspbian/stage3/rootfs/ sh -e -
/boot/dietpi/dietpi-logclear: line 217: /DietPi/dietpi/dietpi-banner: No such file or directory
 Info:
 - Log file directory      | /var/log
 - Processed files         | 12
 - Excluded files          | 0
 - Cleared log files       | 0
 - Unsupported files       | 0
 - Deleted files           | 10
 - Space cleared           | 740 KB

+ cp /home/openrov/workspace/pi-gen/work/2016-05-02-raspbian/stage3/rootfs/boot/dietpi/conf/fstab /home/openrov/workspace/pi-gen/work/2016-05-02-raspbian/stage3/rootfs/etc/fstab
+ on_chroot sh -e -
+ mount
++ realpath /home/openrov/workspace/pi-gen/work/2016-05-02-raspbian/stage3/rootfs/proc
+ grep -q /home/openrov/workspace/pi-gen/work/2016-05-02-raspbian/stage3/rootfs/proc
+ mount
++ realpath /home/openrov/workspace/pi-gen/work/2016-05-02-raspbian/stage3/rootfs/dev
+ grep -q /home/openrov/workspace/pi-gen/work/2016-05-02-raspbian/stage3/rootfs/dev
+ mount
++ realpath /home/openrov/workspace/pi-gen/work/2016-05-02-raspbian/stage3/rootfs/dev/pts
+ grep -q /home/openrov/workspace/pi-gen/work/2016-05-02-raspbian/stage3/rootfs/dev/pts
+ mount
++ realpath /home/openrov/workspace/pi-gen/work/2016-05-02-raspbian/stage3/rootfs/sys
+ grep -q /home/openrov/workspace/pi-gen/work/2016-05-02-raspbian/stage3/rootfs/sys
+ chroot /home/openrov/workspace/pi-gen/work/2016-05-02-raspbian/stage3/rootfs/ sh -e -
DietPi-Ramdisk: Starting
/bin/cp: cannot stat '/boot/boot.ini': No such file or directory
DietPi-Ramdisk: Completed
DietPi-Ramlog: Starting
/bin/chown: invalid user: 'debian-tor:nogroup'
DietPi-Ramlog: Completed
/bin/df: cannot read table of mounted file systems: No such file or directory
Cannot open netlink socket: Address family not supported by protocol
 ───────────────────────────────────────
 DietPi     | 22:54 | Mon 05/02/16 
 ───────────────────────────────────────
 V117       | Unknown Device
 ───────────────────────────────────────
 IP Address | 
 ───────────────────────────────────────

 Created by: Daniel Knight
 Web:        DietPi.com
 Twitter:    twitter.com/dietpi_

 DietPi's web hosting is powered by MyVirtualServer.com

 DietPi is a free product and relies on your donations for funding.
 Donations:  http://goo.gl/pzISt9

 dietpi-launcher  = All the DietPi programs in one place.
 dietpi-config    = Feature rich configuration tool for your device.
 dietpi-software  = Select optimized software for installation.
 htop             = Resource monitor.
 cpu              = Shows CPU information and stats.

```
The logclearing errors seem pretty benign, presumably caused by it being called before dietpi-service sets up /DietPi/. However, the dietpi-service errors don't look so nice:

```
DietPi-Ramdisk: Starting
/bin/cp: cannot stat '/boot/boot.ini': No such file or directory
DietPi-Ramdisk: Completed
DietPi-Ramlog: Starting
/bin/chown: invalid user: 'debian-tor:nogroup'
DietPi-Ramlog: Completed
/bin/df: cannot read table of mounted file systems: No such file or directory
Cannot open netlink socket: Address family not supported by protocol
```

Are these failures expected and safely ignored?

Fourdee, 2016-05-16:
@spiderkeys Apologies for the delay on this and thanks for sharing your script.

These warnings can be safely ignored.
```
DietPi-Ramdisk: Starting
/bin/cp: cannot stat '/boot/boot.ini': No such file or directory
DietPi-Ramdisk: Completed
DietPi-Ramlog: Starting
/bin/chown: invalid user: 'debian-tor:nogroup'
DietPi-Ramlog: Completed
```

Intresting, ```df``` fails to read the mounts. Can you try running ```mount``` at this point, any results?
```
/bin/df: cannot read table of mounted file systems: No such file or directory
```

This is because you are running DietPi scripts prior to running the main DietPi service ```/etc/init.d/dietpi-service```. It can be safely ignored if your trying to clear the log directory prior to mounting the ramdisk.
```
/boot/dietpi/dietpi-logclear: line 217: /DietPi/dietpi/dietpi-banner: No such file or directory
 Info:
 - Log file directory      | /var/log
 - Processed files         | 12
 - Excluded files          | 0
 - Cleared log files       | 0
 - Unsupported files       | 0
 - Deleted files           | 10
 - Space cleared           | 740 KB
```