MichaIng/DietPi #224: Default UART0 baudrate set to 9600, where to change?

spiderkeys, 2016-03-07:
Normally, the baud rate of UART0 is set to 115200 for the RPi2. I've scoured through a bunch of settings and config files but can't find anywhere that would set it to 9600 out of the box. Is this normal behaviour? If so, is there a designated way for overriding with a different baudrate, rather than doing something hacky like changing it with stty in rc.local?

Fourdee, 2016-03-07:
@spiderkeys 
You will need to modify the ```/boot/config.txt``` file and look for the following entry which enables serial line and specifies baud rate:
```
console=ttyAMA0,115200
```
That works for me on Jessie. 
For wheezy, I believe there's another linux config file ```/etc/inittab``` that needs editing: http://www.raspberrypi-spy.co.uk/2013/12/free-your-raspberry-pi-serial-port/

Fourdee, 2016-03-07:
@spiderkeys 
Also, check your serial line client to ensure 115200 baud rate is also set.

spiderkeys, 2016-03-07:
@Fourdee 
Out of curiosity, adding that line doesn't enable the serial console (where you can login/listen to the boot) does it? I'm interfacing to an MCU on UART0 and wouldn't want direct all that chatter at the RPi.

Fourdee, 2016-03-07:
@spiderkeys 
I thought you were referring to a serial line console connection on TXD/RXD pins? This would allow you to basically "SSH without network", directly to the device. My apologies if you are referring to something else?

spiderkeys, 2016-03-07:
Sorry for the confusion. In my setup, I have an Arduino connected to the RPi on UART0, and I want to exchange data between the RPi and Arduino (the Rpi sends commands, the Arduino sends back status and sensor data). My Arduino's serial port is 115200 baud and for some reason the RPi is defaulting to a a baud of 9600. I don't want to turn on the serial console, though, because I have no intention of SSH'ing over the serial port, and the Arduino will be sending binary data which could potentially look like valid Linux commands and cause the RPi to freak out. I'm just trying to figure out what the best way to set the default baud rate of UART0 is while at the same time making sure the serial console functionality is disabled.

Fourdee, 2016-03-07:
@spiderkeys 
No worries, my fault as I didn't have my morning coffee :)

I honestly have little experience with GPIO/Serial usage. I've only dabbled with connecting external sound cards and serial line, nothing more.

Looking at this page: http://www.andremiller.net/content/raspberry-pi-and-arduino-via-gpio-uart , it would appear this user has enabled both ```console=ttyAMA0,9600``` and  ```kgdboc=ttyAMA0,9600```. 
He also mentions: 

> Arduino doesn’t work so well at the default baudrate of 115200, so I changed this on the Raspberry Pi to 9600:

At a punt, I would suggest try adding ```kgdboc=ttyAMA0,115200``` to ```/boot/cmdline.txt```. Failing that, I'am not really sure if I can offer any further assistance. It might be worth trying the official Raspbian Jessie image just to see if the same issue occurs, or if this is a DietPi related issue.

Also, are you running Wheezy or Jessie? ```cat /etc/*release```

spiderkeys, 2016-03-07:
I'm using Jessie, so I should be up to date. It's not a big deal, as I can write a workaround. (I'll probably just call stty -F /dev/ttyAMA0 115200 in a postboot script :smile:). Usually there is some driver setting/image config setting that is used to set the default rate for UART ports, so I was just hoping to find that. 

I found that in the normal Raspi distro, there is the following value you can set in /boot/config.txt:

```
init_uart_baud=115200
```
However, this is not present, nor does it seem to affect anything in the DietPi image. I'll have to do an install of the official Lite Image and see if they actually have it there or in the raspi-config tool, since I'm just going off of forum search results.

closed 2016-03-09 by spiderkeys