MichaIng/DietPi #226: cpufrequtils not having desired effects

spiderkeys, 2016-03-08:
Is there something in DietPi that would interfere with the normal functionality of cpufreq-set? I'm trying to manually underclock to 700Mhz on the fly without a reboot but the cpu frequency doesn't seem to respond to any changes I make with cpufreq-set, no matter which governor I select or which min-max frequency I set.

spiderkeys, 2016-03-08:
Example:

```bash
root@openrov:~/# cpufreq-set -d 700Mhz
root@openrov:~/# cpufreq-set -g powersave
root@openrov:~/# cpufreq-info
cpufrequtils 008: cpufreq-info (C) Dominik Brodowski 2004-2009
Report errors and bugs to cpufreq@vger.kernel.org, please.
analyzing CPU 0:
  driver: BCM2835 CPUFreq
  CPUs which run at the same hardware frequency: 0 1 2 3
  CPUs which need to have their frequency coordinated by software: 0 1 2 3
  maximum transition latency: 355 us.
  hardware limits: 600 MHz - 900 MHz
  available frequency steps: 600 MHz, 900 MHz
  available cpufreq governors: conservative, ondemand, userspace, powersave, performance
  current policy: frequency should be within 700 MHz and 900 MHz.
                  The governor "powersave" may decide which speed to use
                  within this range.
  current CPU frequency is 900 MHz (asserted by call to hardware).
  cpufreq stats: 600 MHz:28.18%, 900 MHz:71.82%  (4)
```
According to kernel.org docs:

```
2.2 Powersave
-------------

The CPUfreq governor "powersave" sets the CPU statically to the
lowest frequency within the borders of scaling_min_freq and
scaling_max_freq.
```

I am expecting to see the frequency get set to 700 Mhz here.

Fourdee, 2016-03-09:
@spiderkeys 

DietPi pulls all available cpu related options and applies all cpu settings via ```/sys/devices/system/cpu/cpuX/cpufreq``` with the following script, during boot and when you change cpu settings in ```dietpi-config``` > Performance options:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-cpu_set

> available frequency steps: 600 MHz, 900 MHz

This is why you are unable to set 700mhz. The frequency step is unavailable as reported by the kernel.

If you run ```dietpi-config``` and select Performance from the menu, you can set the CPU gov and frequency limits. The values listed are pulled directly from the kernel. So if the values dont exist, they are not supported.

[image]
[image]

spiderkeys, 2016-03-09:
Awesome, exactly what I needed to know. Thanks!

closed 2016-03-09 by spiderkeys