MichaIng/DietPi #228: DietPi-Software | Ampache install is out of date.

Zone22, 2016-03-10:
Hardware: DietPi on a raspberry pi first-generation.
Occurs when installing Ampache through DietPi-Software.
Ampache 3.8.0 is installed when 3.8.2 is available according to the Ampache web interface.

Fourdee, 2016-03-11:
@Zone22 
:+1: I'll get this updated to 3.8.2

Our Ampache install reads a pre-setup SQL DB during install, makes it easier for the user and installation is completely automatic, but it does mean we need to update the install now and then.

Fourdee, 2016-03-11:
3.8.2 requires:
[image]

Fourdee, 2016-03-11:
Composer requires Git install:
[image]

Fourdee, 2016-03-11:
Done. Ampache 3.8.2 install available in v113.

closed 2016-03-11 by Fourdee

Zone22, 2016-03-11:
Thank You! DietPi is amazing.