MichaIng/DietPi #229: DietPi-Config | Read only filesystem option.

Fourdee, 2016-03-11:


Fourdee, 2016-04-02:
As DietPi uses Ramlog, our SD writes are extremely low when compared with other images.

So a read-only filesystem will only benefit users who wish to prevent file modifications.

xenfomation, 2016-05-01:
Agreed -- read-only FS can be avoided with a USB drive if the idea is dodging entropy

axel-h, 2016-05-17:
One use case might be that an accidental shutdown of the system does not do any harm. I had a few cases with other distributions where power loss makes the SD card unusable and only a fsck on another machine can fix this. But it might also be a weakness of the file system that is used - should just pick a better one. Or it's a bug in the Odroid's SD usage. We'll see if I run into this with DietPi also.

trajano, 2017-02-08:
@axel-h I presume you tried `touch /forcefsck` and it doesn't really help with the file system corruption errors either like I did.

ParaplegicRacehorse, 2017-02-22:
Use case for write protection: thin clients and high availability fail-over server systems and anything else that might want or need to have most of its filesystem served to it over the network.

WolfganP, 2017-05-01:
I was investigating on this front as well to reduce SDcard wear, and found useful to install both **iotop** (apt-get install iotop) and **iostat** (apt-get install sysstat)
With the ramdisk implementation, it seems no much I/O activity is directed to the SDcard (mmcblk0 in my case) but still there is some, even in idle systems. Will keep investigating and share the results.

trajano, 2017-05-01:
One option that @ddurdle uses in his Pi installations is BTRFS and he makes a stable snapshot at one point which he can readily recover from.

ddurdle, 2017-05-01:
I highly recommend at least mounting the SD card in read-only mode during normal operation, at least for the OS component.  Break out user files to a separate partition or device.  **Highly recommend using overlays for this purpose (supported on kernel 3.18 or newer)**.  Secondly, I recommend leveraging BTRFS to allow for snapshots for easily backing up and for handling corruption if you refuse to go the read-only route.  Mount the booting OS as a subvolume snapshot.  If it badly gets corrupt, flip back to a snapshot.  I recommend not mounting the root BTRFS on a regular basis because if you corrupt the root BTRFS, you lose your snapshots.  It also comes in handy to at least backup snapshots periodically to a difference device.

I've kissed corrupt SD cards goodbye.

ddurdle, 2017-05-01:
I'll also add one other bit of warning.  Even if you are not writing to SD card regularly, simply having the partition mounted in read-write will eventually lead to corrupting the partition if undervolting or improper shutdown occurs.  I average 6 months before I get into a situation where I can't boot anymore and this is from very little or no actual writes to the partition (simply mounting it, not using it).

Leveraging overlays to capture all writes, I have yet corrupt a SD card.  I even have a few SD cards that are nearing their end-of-write life (they will be unbootable after 1-2 reboots in read-write mode), and they still run forever bootable and fully working when I redeploy.

Fourdee, 2017-05-07:
#### Notes:

Need some tweaks running ```/``` under RW
- Ramdisk:
```
root@DietPi:~# systemctl status dietpi-ramdisk -l
● dietpi-ramdisk.service - DietPi-RAMdisk
   Loaded: loaded (/etc/systemd/system/dietpi-ramdisk.service; enabled)
   Active: failed (Result: exit-code) since Sun 2017-05-07 12:42:44 BST; 51s ago
  Process: 554 ExecStart=/bin/bash -c /boot/dietpi/dietpi-ramdisk 0 &>> /etc/dietpi/logs/dietpi-ramdisk.log (code=exited, status=1/FAILURE)
  Process: 517 ExecStartPre=/bin/mkdir -p /etc/dietpi/logs (code=exited, status=0/SUCCESS)

May 07 12:42:44 DietPi bash[554]: /bin/bash: /etc/dietpi/logs/dietpi-ramdisk.log: Read-only file system
May 07 12:42:44 DietPi systemd[1]: dietpi-ramdisk.service: control process exited, code=exited status=1
May 07 12:42:44 DietPi systemd[1]: Failed to start DietPi-RAMdisk.
May 07 12:42:44 DietPi systemd[1]: Unit dietpi-ramdisk.service entered failed state.
```

Fourdee, 2017-05-07:
[image]

Fourdee, 2017-05-08:
Completed.

closed 2017-05-08 by Fourdee