MichaIng/DietPi #230: DietPi-Software | Add MPD installation option

Zone22, 2016-03-11:
Music Player Daemon (MPD) is a flexible, powerful, server-side application for playing music. Through plugins and libraries it can play a variety of sound files while being controlled by its network protocol.

http://www.musicpd.org/

MPD could be used in conjunction with Ampache for localplay.

Fourdee, 2016-03-11:
@Zone22 
MPD (and YMPD) can be installed by selecting HiFi from DietPi software. Although HiFi is a frontend webplayer, it will still install and config MPD. 

I will look into separating MPD into its own installation option, sometime in the future.

I've not tested it with Ampache myself. If you try it, please let me know, I will assist you to get it working if needed.

Zone22, 2016-03-11:
YMPD and Ampache in terms of compatibility coexist they are although not integrated. They share the same music directory. I am unable to connect to an Ampache MPD instance using the local IP and any port. Conversely I am able to connect to YMPD uusing a local IP and default port.  My setup is rather unusual because I'm using YMPD, Ampache, WiFi HotSpot, and ownclowd to manage music and data sync.

I'm creating in car jukebox/cloud. I'm utilizing ownclowd is the main syncing mechanism for pictures/music/data. Streaming/syncing the music collection that synced with ownclowd vvia compatible Ampache API app 'Just Player'. Everything's been successful with semireasonable performance on a first and raspberry pi. The last part is figuring out localplay I think I need to configure MPD to the owncloud music directory.

Fourdee, 2017-12-30:
MPD is now its own installation in dietpi-software.

closed 2017-12-30 by Fourdee