MichaIng/DietPi #231: DietPi-Software | Add Node.js installation option

bhaveshgohel, 2016-03-12:
Hello Dan,

This is issue is regarding installing Node.js from Application install option.
I know i can install Node.js using apt-get but it will install older version of Nodejs.
There is a nice bash script for installing latest version of Node.js in Linux:
https://github.com/taaem/nodejs-linux-installer 

Can we use that and add it in installation option? 

Thanks,
Bhavesh Gohel

Fourdee, 2016-03-12:
@bhaveshgohel 
Thanks for the request and links. Will come in very handy.

You'll have to excuse my ignorance, is node.js a javascript library for web development? I'am just trying to assign it a category.

bhaveshgohel, 2016-03-14:
Hello Dan,

Sorry for late reply,
Node.js is an open-source, cross-platform runtime environment for developing server-side Web applications. Although Node.js is not a JavaScript framework, many of its basic modules are written in JavaScript, and developers can write new modules in JavaScript. The runtime environment interprets JavaScript using Google's V8 JavaScript engine. 

Also See: [Is Node.js a framework?](http://programmers.stackexchange.com/questions/299152/is-node-js-a-framework) -- Yes an No. ¯\_(ツ)_/¯

Some Application that can be built on top of Node.js:
- Server-side application
- Client-side Web Application
- Cross Platform Native Application
- Terminal Application
- and many more..

It's All rounder. :smiley: 

According to there [website](https://nodejs.org/en/):
>Node.js® is a JavaScript runtime built on Chrome's V8 JavaScript engine.

So, Suggestion would be to create a new category name "Node.js" and add it as "Node.js - Javascript Runtime"

Fourdee, 2016-03-14:
@bhaveshgohel 

> So, Suggestion would be to create a new category name "Node.js" and add it as "Node.js - Javascript Runtime"

Excellent, sounds good to me and thanks for the additional information :+1: 

I'am not sure if its going in the DietPi or Linux software menu yet, will let you know when its completed.

spiderkeys, 2016-03-14:
Additional info, in case you are looking for a binary source: we've been using the armhf builds put out by Nodesource on both DietPi and our Debian build for BBB: https://github.com/nodesource/distributions

They seem to keep on top of things fairly well and have a wide range of tested builds for ARM platforms. They are compiled for Debian/Ubuntu, but I don't think that will cause any issues (we use Node extensively and haven't ran into any problems yet).

Fourdee, 2016-03-18:
@spiderkeys 
Excellent, thanks for sharing the binaries :+1: , will give them a spin.

Fourdee, 2016-03-18:
@spiderkeys 
Saved me about 4+ hours of manual compiling. Nice one :+1:

Node.js v5.x
Available in next release v113.

[image]

bhaveshgohel, 2016-03-19:
@Fourdee : Great :tada:   
It's best suited in both the current category name: Development / Programming and Shared Libraries. So any of one will be good enough. :+1: 

@spiderkeys Thank you for the additional info. :smiley:

Fourdee, 2016-03-20:
Available in v113

closed 2016-03-20 by Fourdee

bhaveshgohel, 2018-02-21:
Hello @Fourdee ,
Is it possible to add option to install LTS version rather than latest version of Node.js or keep both option?

Modification required in script:
Just need to replace link with http://nodejs.org/dist/latest-carbon/ in [here](https://github.com/taaem/nodejs-linux-installer/blob/master/node-install.sh#L13) and [here](https://github.com/taaem/nodejs-linux-installer/blob/master/node-install.sh#L51)

and in your script https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L7221 with "node-install-lts.sh"

Note: [Carbon]( https://github.com/nodejs/Release
) is Current LTS version(End-of-life: December 31, 2019)

Current PR for this changes - https://github.com/taaem/nodejs-linux-installer/pull/10

Thanks.
BG.

MichaIng, 2019-05-06:
@bhaveshgohel 
Many thanks for your suggestion. Please check my reply on your PR. Note that we are not part of the nodejs-linux-installer project, we just use it since it's a simple clean installer that does what we would do ourself otherwise.
So it's up to the project owner if/how he wants to allow non-latest version installs.

With DietPi-Software we will most likely not allow users to choose between Node.js versions. Instead we will simply install the latest version that is known to work with the related software title. In some cases we use the `n` module to install Node 8 where it is required.

What we are instead aiming for is to allow running different Node versions beside each other smoothly: https://github.com/MichaIng/DietPi/issues/2226
So this would allow to use multiple Node-based software titles with different maximum version supports to run smoothly beside each other, every title with the most current Node version it supports. The ARMv6 support drop is really a bumper here, since it makes this implementation a lot more complicated 🤔.

bhaveshgohel, 2019-05-06:
Using `n` module is also good option. :+1:

MichaIng, 2019-05-06:
@bhaveshgohel 
Jep, only issue is the overhead of having two Node versions installed (latest + 8). However simplest workaround currently.