MichaIng/DietPi #234: Static IP with Jessie & RPi3

wdawson001, 2016-03-14:
The new raspberry pi 3 defaults to WiFi and I think this is causing issues with the Diet-pi config when using a static IP. (I use mine for pi-hole). The install steps are not correctly assigning a Static IP

https://errorin10.wordpress.com/2016/03/07/raspberry-pi-configure-static-ip-with-etcnetworknterfaces-or-dhcpcd/

I tried that solution but I couldn't get my networking back up and running. Will try later with Wheesy to see if it has the same issue.

Fourdee, 2016-03-14:
@wdawson001 

> The new raspberry pi 3 defaults to WiFi and I think this is causing issues with the Diet-pi config when using a static IP

DietPi disables ```wlan0``` on all images by default. It must be enabled in ```dietpi-config``` or ```/etc/network/interfaces``` ```auto wlan0``` before it can be configured.

I think the issue may be in your ```/etc/network/interfaces file```. Could you please upload its contents, or paste it in your reply?

Fourdee, 2016-03-14:
@wdawson001 
I did a fresh PiHole install on a Pi 3, was unable to replicate the issue you are experiencing. 

For reference, heres the ```/etc/network/interfaces``` file it generated with static IP:
```
#/etc/network/interfaces
#Please use DietPi-Config to modify network settings.

# Local
auto lo
iface lo inet loopback

# Ethernet
auto eth0
iface eth0 inet static
address 192.168.0.104
netmask 255.255.255.0
gateway 192.168.0.1
dns-nameservers 8.8.8.8 8.8.4.4

# Wifi
#auto wlan0
iface wlan0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
wireless-essid FuzonWifi
wireless-key abcde12345
wireless-mode Managed
wireless-power off
wpa-ssid FuzonWifi
wpa-psk abcde12345
#dns-nameservers 8.8.8.8 8.8.4.4
```

wdawson001, 2016-03-14:
@Fourdee I'll do a fresh install tonight (in 4hrs) and see if it is replicated. The static ip issue was with over the Ethernet, not the wifi. 
Thanks for the quick response

Fourdee, 2016-03-14:
@wdawson001 

>  I'll do a fresh install tonight (in 4hrs) and see if it is replicated. The static ip issue was with over the Ethernet, not the wifi. 

No worries. 
If the issue reoccurs, please let me know and upload your ```/etc/network/interfaces``` so i can verify settings.

wdawson001, 2016-03-15:
Tried it again, failed again - if you select pi-hole and run through the auto setup that pi-hole puts you through, it failed for me. 

However, I setup the dietpi, with no software installs, and configured the network, then installed pi-hole and it worked great! 

(Good work on updating to the latest pi-hole!)

closed 2016-03-15 by wdawson001