MichaIng/DietPi #235: Calender and Contact Syncing with an additional Webclient

1qay1, 2016-03-14:
Hi there,

I really like DietPi´s lightweight approach and the easy installing methods for software packages.

Although the software package already provided is feature rich and with different options and choices there is currently a category not yet available: Calender and Contact Syncing with an additional Webinterface.

We all have a lot of devices (e.g. computers and smartphones) that have a calendar and adressbook. So it would be cool if all our devices with their calendars and adressbooks are synced. All this can be done with an RaspberryPi that runs a Webserver.

To make it easier to decide about this feature request and make it easier to implement I already did some "research" about working solutions. Fortunately these approaches perfectly fit with Dietpi´s lightweight approach.

Calender and Contact Server:

http://www.davical.org 

http://baikal-server.com

Webclient

https://www.inf-it.com/open-source/clients/infcloud/

Demo of the Webclient:

https://www.inf-it.com/infcloud/

What do you think?

Fourdee, 2016-03-15:
@1qay1 
Hi,

Many thanks for the feedback and software request.

I took a quick look at http://baikal-server.com and I'am edging towards it. It uses a MySql DB which will help with compatibility with our webserver stack selection. Davical uses PostgreSQL.

I'am currently wrapping up DietPi https://github.com/Fourdee/DietPi/milestones/v113 with Odroid C2 support. Once its completed, I'll look into your request in more detail.

1qay1, 2016-03-16:
Hi @Fourdee,

thank´s for thinking about my feature request.

Yes Baikal is using MySQl and SQLite DB. I tested it with DietPi works nearly out of the box; very easy to install (took me just a few minutes getting it up and running and I am surely no expert). Synchronisation reliable with iOS and Android clients tested.

Currently there is the version baikal-server.com (0.2.7) the newer and updated versions (0.4.0) are available under https://github.com/fruux/Baikal/releases. Both versions tested with Dietpi LAMB installation.

In combination with an additional Webclient (more Web-Frontend ) like https://www.inf-it.com/infcloud/) it will be very handy and comfortable to edit and display adressbooks and calendars.

I did some research and the Baikal project will evolve to Baikal2. Baikal2 brings Cal-Card DAV Server and an already integrated Web Frontend for displaying and editing adressbooks and calendar. More information and a demo version is available at https://github.com/fruux/Baikal2.

Therefore your edging towards Baikal seems to be good choice because it would evolve inline with the Dietpi releases towards a complete all-in package.

Like I already explained I think this would be a useful addition and complementation for the Dietpi package that´s currently missing and a lot of users will like and use this feature. Especially when it comes out of the box.

Yes I have the seen the next milestone v113 of Dietpi and I fully understand that´s currently a high priority where all efforts is necessary to get this finished. 

So I hope once it´s completed you may find some spare time to take a closer look in this feature request.

Fourdee, 2016-03-18:
@1qay1 

Decided to go for BaiKal 0.5 and InfCloud frontend.
BaiKal is done. InfCloud front end, aiming to get that completed tomorrow.
[image]

Fourdee, 2016-03-20:
@1qay1 
Running behind on InfCloud front end. As I'am trying to finish up v113, I will move it to v114 and a separate ticket:
https://github.com/Fourdee/DietPi/issues/242

Fourdee, 2016-03-20:
Note to self: Add usage details to dietpi.com/software

Fourdee, 2016-03-20:
BaiKal http://fuzon.co.uk/phpbb/viewtopic.php?f=8&t=5&start=50#p1502:

Available in v113

closed 2016-03-20 by Fourdee

1qay1, 2016-03-22:
Hi @Fourdee 

thanks for picking up my feature request and including Baikal DAV in DietPi Software list.
Hope this feature makes a lot of people happy and enjoy it added to DietPi.

Just a question: Where did you find the 0.5.0 version of Baikal? Benn following the project on github and thought 0.4.x is the latest release (https://github.com/fruux/Baikal/releases)

Fourdee, 2016-03-22:
@1qay1 

We actually install from the master branch (i'am assuming 0.4.2): https://github.com/fruux/Baikal. Which I would expect is the latest stable version. It could be possible they are using the master branch for active development, would need to ask.

Looks like their versioning system is a little different than expected, this is the version number they use to display in the BaiKal mainpage after install: https://github.com/fruux/Baikal/blob/master/Core/Distrib.php#L27

1qay1, 2016-03-23:
Hi @Fourdee

thanks for the clarification. Was just wondering. 

From my experience they might use "low numbers" (e.g. 0.4.2) but Baikal server is quite popular and in productive professional use since years and running stable.