MichaIng/DietPi #238: RPi-Monitor as optional remote monitoring tool

WolfganP, 2016-03-17:
Based on this article http://www.cnx-software.com/2016/03/17/rpi-monitor-is-a-web-based-remote-monitor-for-arm-development-boards-such-as-raspberry-pi-and-orange-pi/ , I found this project (https://github.com/XavierBerger/RPi-Monitor) may be interesting for integration to DietPi (as Raspcontrol was removed due to lack of project maintenance)

Fourdee, 2016-03-18:
@WolfganP 
Great find on this :+1: . Slick UI and great stats, love it.

Available in next release (v113) as a DietPi optimized installation for RPi (Jessie):
[image]

WolfganP, 2016-03-18:
Glad to help improving the distribution. Thanks for the quick integration!

k-plan, 2016-03-18:
@Fourdee 

> Available in next release (v113) as a DietPi optimized installation for RPi (Jessie):

Used it for long time on my RPis, it's a great piece of software. Can also run on oDroid C1 / C2 with same modifications:

[image]

:) Like it. And the Debian packaging is handy.

It's a pity, but XavierBerger is not working on it at the moment.  (https://github.com/XavierBerger/RPi-Monitor/issues/143#issuecomment-177316826) 
http://rpi-experiences.blogspot.fr/

If RPi only is in fokus, https://pi-control.de/ can also be a option. But he also come not to terms with his Version 2.0 (current: [Pi Control 1.3.3](http://willy-tech.de/pi-control-1-3-3/))

k-plan, 2016-03-19:
@Fourdee 
notice, that you will  integrate it in v.113. Verry nice.
https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L1826

Is there a reason, why you use `rpimonitor_2.9.1-1_all.deb` and not current [rpimonitor_2.10-1_all.deb](https://github.com/XavierBerger/RPi-Monitor-deb/blob/master/packages/rpimonitor_2.10-1_all.deb) ?

https://github.com/XavierBerger/RPi-Monitor-deb/blob/master/debian/changelog
http://rpi-experiences.blogspot.fr/2015/01/rpi-monitor-version-210-is-available.html

Fourdee, 2016-03-19:
@k-plan 
Yep, I didnt see the 2.10 in the middle lol: https://github.com/XavierBerger/RPi-Monitor-deb/tree/master/packages

I'll update the install to 2.10. :+1:  good spot.

k-plan, 2016-03-20:
@Fourdee 
have playing around (quick and very dirty) with some lines in  `/etc/rpimonitor/template/...` and it is really amazing on Odroid C1/C2. Not only on RPis as the name implies. 
Have updated the picture in posting above. We would be pleased to receive "_DietPi-Monitor-for-oDroid_" :smiley:

Fourdee, 2016-03-20:
@k-plan 

> have playing around (quick and very dirty) with some lines in /etc/rpimonitor/template/... and it is really amazing on Odroid C1/C2. Not only on RPis as the name implies. 

Very nice, love the DietPi version number in there :+1:  :)

Fourdee, 2016-03-20:
Note to self: Add usage details to dietpi.com/software

Fourdee, 2016-03-20:
http://fuzon.co.uk/phpbb/viewtopic.php?f=8&t=5&start=50#p1503

Available in v113

closed 2016-03-20 by Fourdee

k-plan, 2016-03-20:
@Fourdee 

> Available in v113

For now, it is only for all the RPi?
Is it possible, you missed to integrated  RPi-Monitor in `dietpi-software`  for Odroid C1/C2?
Can`t find it after update to v113 on C1/ C2 .... :weary:

BTW: 
```
root@Odroid-C2:~# cat /DietPi/dietpi.txt
...
AUTO_DietpiSoftware_Install_RASPCONTROL=0
....
```
:flushed: ... but no big deal.

Fourdee, 2016-03-20:
@k-plan 
Only available for RPi at the moment.

I know you modified the code for Odroid c2, but did it work before your changes? If so I can test all devices with RPi monitor and enable them for next release.

If your modified code is required for non-rpi devices, we would need to apply your modified code and test the following devices:
```
XU4
C1
C2
OrangePi
VM x64
```
I can test the all the devices you dont have. Just let me know which ones.

> AUTO_DietpiSoftware_Install_RASPCONTROL=0

Yep, its in there for Wheezy as some users are still downloading/install it.

k-plan, 2016-03-21:
@Fourdee 

> ... but did it work before your changes?

Yes run "out-of-the-box" on my C1+/C2

```
root@Odroid-C2:~#  wget https://github.com/XavierBerger/RPi-Monitor-deb/raw/master/packages/rpimonitor_2.10-1_all.deb

root@Odroid-C2:~#   dpkg -i rpimonitor_2.10-1_all.deb

root@Odroid-C2:~#  apt-get -f  install
```
et voilà - not perfect (_Voltage, Firmware, only cpu0 in focus_) but fundamental content:

[image]

With a little fine tuning in templates it can be handsome.

> XU4
> ~~C1~~
> ~~C2~~
> OrangePi
> VM x64

Sorry, only own a old iMac (late 2007), a few RPi and a [CubieTruck](http://cubieboard.org/tag/cubietruck/).
Don't  believe, it's a great help.
But I can send you the dirty hacked templates for C2 ...
:de: :)

k-plan, 2016-03-29:
Hi,
RPi-Monitor: Version 2.11 Beta is ready for test: [please read this before](http://rpi-experiences.blogspot.de/2015/06/rpi-monitor-version-211-beta-is-ready.html#more)

`:~#  wget https://github.com/XavierBerger/RPi-Monitor-deb/blob/devel/repo/rpimonitor_2.11-beta7_all.deb`

My Test result:
[image]

Same new and really cool features and nice layout.

Fourdee, 2016-03-29:
@k-plan 
Looks great :+1: , does this version support the Odroid straight off the bat? Or did you modify the code for this?

WolfganP, 2016-03-29:
It would be wonderful if anyone with knowledge of RPi-Monitor plugins can add one for proper RasPi shutdown (so headless systems can be completely controlled via web browsers), and add it to DietPi packages.

k-plan, 2016-03-30:
@Fourdee 

> does this version support the Odroid straight off the bat? Or did you modify the code for this?

Yes, like before. Run out-of-the-box. No code modding, only copy my old Odroid templates, to get some more supported infos and 4 CPUs. But it's easy, otherwise I was not able to do it.
You can doubtless do this better, as I can.

@WolfganP 

> It would be wonderful if anyone with knowledge of RPi-Monitor plugins can add one for proper RasPi shutdown (so headless systems can be completely controlled via web browsers),

Hmm, it's a problem with privilege of web services. There are no "plugins" in the broadest sense. But RPi Monitor will be able to start other programs, like [Shell In A Box](http://www.tecmint.com/shell-in-a-box-a-web-based-ssh-terminal-to-access-remote-linux-servers/). So you will be able, to execute `reboot` or `poweroff` and more in your browser.

Oder, wenn du ein wenig Deutsch lesen kannst, wäre das oben angeführte [Pi Control](https://willy-tech.de/pi-control/) die einfachere Alternative. Einen Webserver mit PHP über `dietpi-software ` installieren, dann den Rest wie auf der Seite erklärt. Installation ist sehr einfach. Fehlt etwas, wird danach gefragt und erklärt, was man tuen soll. Bei der Installation muss man seinen root Account mit Passwort angeben. Dann ist es auch möglich, Systemeinstellungen über das Webinterface zu ändern, das System zu rebooten oder auszuschalten. Ebenfalls mit der Android App. Funktioniert auch auf C1/C2.

WolfganP, 2016-08-22:
FYI, RPI_Monitor releases v2.11 into production https://github.com/XavierBerger/RPi-Monitor/releases/tag/v2.11

Fourdee, 2016-08-22:
@WolfganP 
Thanks for letting us know.

DietPi v130 will install 2.11, although, it appears to break Rich's USB drive additions: https://github.com/Fourdee/DietPi/issues/410

[image]