MichaIng/DietPi #242: DietPi-Software | InfCloud front end for BaiKal

Fourdee, 2016-03-20:
Continuation from https://github.com/Fourdee/DietPi/issues/235#issuecomment-198488772

1qay1, 2016-03-22:
Hi @Fourdee 

thanks again for adding the Infcloud Webclient as an frontend extension for the Baikal DAV Server.

As far as I know the initial configuration of Infcloud may be a little tricky. 

So it´s very helpful to take a closer look in these two documents:

https://www.inf-it.com/infcloud/readme.txt
https://www.inf-it.com/infcloud/config.js.txt

You will have to choose and edit the "globalNetworkCheckSettings" and for the Authentication it might be necessary to in the Infcloud config.js file. 

Additional it might be necessary to change the "fileglobalUseJqueryAuth" in the Infcloud config.js and change the Baikal Server settings from "digest" to "basic" in the admin interface.

It took me a while to sort the things out and this might be useful and saves you time...

Fourdee, 2016-03-26:
@1qay1 

> So it´s very helpful to take a closer look in these two documents:
https://www.inf-it.com/infcloud/readme.txt
https://www.inf-it.com/infcloud/config.js.txt
It took me a while to sort the things out and this might be useful and saves you time...

And then some lol. The config.js comments is a mess, going to take me at least a good few solid days from scratch to fully understand every option, setting and configuration.

Do you have a working config.js setup for BaiKal that I could use as a reference? This would vastly speed up the implementation.

1qay1, 2016-03-27:
Hi @Fourdee 

lol, yes I can fully understand. This was exactly the same what I thought about the configuration.

Fortunately there may be huge configuration options to fit all needs and demands but you have to change 2 settings (just a line) to get a working config and you do not need to touch the rest.

I will provide the needed help and hints. 

Currently I am traveling so I will get back to you soon.

Fourdee, 2016-03-28:
@1qay1 

> Fortunately there may be huge configuration options to fit all needs and demands but you have to change 2 settings (just a line) to get a working config and you do not need to touch the rest.

I will provide the needed help and hints.

Excellent :+1: appreciate it.
I'll try to to code in/automate the 2 settings during our installation.

I'll nudge this to v115, so no rush. I'am aiming to get v114 released today/tomorrow.

Fourdee, 2016-04-02:
@1qay1 
I'll set this to "no milestone" whilst we wait for your configuration file.

Fourdee, 2017-06-17:
Marking as closed due to no progress and lack of interest from our users in this software. If someone else wants to pick this up, please reopen.

closed 2017-06-17 by Fourdee