MichaIng/DietPi #244: DietPi-Software | Updated Redis server to use latest version (from source)

Fourdee, 2016-03-20:


Fourdee, 2016-03-28:
Was not required for our current EmonPi installation. 
Will look at this in the future when work starts on:
https://github.com/Fourdee/DietPi/issues/257
https://github.com/Fourdee/DietPi/issues/256

MichaIng, 2017-10-02:
As I was going through the old opened issues:

Did there still appear cases where an update of redis-server (and/or php(5)-redis) is necessary? Otherwise could be closed, especially since stretch will be hopefully soon released with new redis repo version.

In case, I don't like installation from source, since dependencies and updating need to be done manually/maintained by dietpi updates. So someone would need to regularly have an eye on possible updates of this software. Only advantage ist that possible bugs related to dietpi/other software can be securely tested before shipping the new version via dietpi-update.

As an alternative, if necessary/desired we could use backports from the official debian repo: https://packages.debian.org/de/stretch-backports/redis-server
As far as I tested with raspbian jessie that day, it worked quite stable for several things and of course the first used backport should be tested carefully on all systems.

closed 2017-12-30 by Fourdee