MichaIng/DietPi #245: Full Disk Encryption

chrisvella, 2016-03-22:
Would it be possible to add a full disk encryption option for installer? I understand it may cause some performance loss, but personally I feel its worth the piece of mind if my device was ever stolen.

Fourdee, 2016-03-25:
@chrisvella 
Thanks for the suggestion, i'll look into this when I can.

xenfomation, 2016-04-27:
@Fourdee --

While I haven't checked in anything YET, I wouldn't mind being assigned this issue as I have the very first PI 2+ where I discoverd DietPi.

Point is I am looking to:
- See about cipher hooks in Jessie's kernel
- See if complete partition, partial, or file-based encryption will make the least performance hit
- Run a random-based file utility to literally see how long it takes to turn a micro SD card into a brick
- Document, report back, and see if this should be a "SERIOUSLY COOL THING WITH SERIOUSLY LONG TERM AFFECTS DietPi didn't invent" ;-)

Hope all is well & Cheers!

Fourdee, 2016-04-27:
@xenfomation 

> I wouldn't mind being assigned this issue as I have the very first PI 2+ where I discoverd DietPi.

Done, dig in :+1:  :)

xenfomation, 2016-04-28:
Focusing on user home partition and root user partition w/ eCrypt FIRST.  System is prepped and testing will start @ 09:00 EST 27-APRIL-2016.  Simple test, simple metrics.

Links related to eCrypt and TrueCrypt (two test targets for possible enhancement satisfaction) to be provided shortly.

xenfomation, 2016-04-28:
a.  ecryptfs project page (actively maintained):
[https://launchpad.net/ecryptfs](https://launchpad.net/ecryptfs)

b.  ecryptfs how-to for everyone to see/consider (live system w/ encrypted directory has to leave a signature or clear text password config file, which is... not security savvy).
[https://www.howtoforge.com/tutorial/how-to-encrypt-directories-and-partitions-with-ecryptfs-on-debian/#-links](https://www.howtoforge.com/tutorial/how-to-encrypt-directories-and-partitions-with-ecryptfs-on-debian/#-links)

c.  currently not looking back @ wheezy or squeeze; just focusing on rollout, test, repeat, automation effort

xenfomation, 2016-05-01:
Tests so far have left my Pi unbootable post-install -- even following crypto option docs.  I am focusing now on eCrypt FS for directories I hold important data as it offers:

Directory Encryption
PARTITION (ie, USB drive) Encryption

Fourth test going on now -- thanks for the patience.  I have SD cards to burn and I did burn one out already -- no chance to recover data and the PI was practically unusable.

xenfomation, 2016-05-09:
@Fourdee @chrisvella --

Quick update.

In between other tasks, took a second round with this.  There are pros and cons to it, such as clear-text password in a text file for auto mount + encryption, user installation would need to be guided, etc.

It beats full drive encryption, works, but I will circle back to this as it may be better suited for:

- USB Drive Storage Only
- Unique home/paths outside of /root
- General directories for low-impact encryption, ie "not the mysql data directory!"

Cheers.

chrisvella, 2016-05-09:
I'm not sure if this is useful, but checkout how antergos manages fulldisk encryption. Personally I have found it very effective (on a lot more powerful machines).

xenfomation, 2016-05-09:
Will do as I am quite familiar with Antergos from a VM perspective!

xenfomation, 2016-05-28:
All,

I am still working on this, however with work on #271 I have to consider USB devices, as well.
Thanks for your patience and if anyone has had success with command-line installation of specific utilities, certainly copy and paste:

What packages were installed?
What configs were generated?
Which distro for SBC was used?

This request comes due to the age of this ticket and the information provided by @chrisvella, et al as I look to find a means and balance for this.

1989gironimo, 2017-11-25:
@xenfomation you won't complete this, won't you?

Fourdee, 2018-04-05:
Marking as closed due to:
- No activity and minimal interest from our user base.

closed 2018-04-05 by Fourdee