MichaIng/DietPi #246: dietpi software - add NFS server to option a SaMBa

stefaniniMM, 2016-03-22:
NFS server is better optimized for *nix sharing (all computers in home running linux, *BSD)

Fourdee, 2016-04-02:
@stefaniniMM 
Thanks for the request and apologies for the delay in this response. I'll setup this ticket accordingly.

xenfomation, 2016-05-02:
@Fourdee -- in initial testing, NFS4 is rather straight forward.  It is easy enough to create a default NFS4 Server/Client install with templates for /etc/exports, etc.

The questions I have -- to fully implement this for user needs -- would be if a sub-menu is warranted to ask:

Export path
NFS version
IP Exclusions
Etc

In the next 24 hours I can get a basic NFS4 server installer implemented to dietpi-software that creates:

/export/nfs4/
/etc/exports
/etc/nfs-common
/etc/nfs-kernel-common
Creates an NFSv4 service

Either way is fine with me as I do think NFS is a good thing for us YP/Old School/Penguin lovers.

Thanks!

xenfomation, 2016-05-09:
Oh, cool factor is -- just like Samba -- I have staggered out an NFS Client (common packages) and NFS Server (common + core).  The part next is to follow in footsteps to assist the user in configuring their NFS Server (if that is selected), etc.

So this is being worked on as an update to all!

xenfomation, 2016-05-12:
Options have been subbed in -- expanding upon NFS server install, such as with Samba.

Thanks.

xenfomation, 2016-05-17:
Delay to request has been to ensuring I reconfigure git scripts/git desktop to avoid collisions. 

Thanks, everyone.

Fourdee, 2016-09-09:
Notes to self:
- More testing of both server + client required
- Security: Add docs to explain how to limit users by IP address > http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=2821#p2821

Fourdee, 2016-09-09:
[image]

Fourdee, 2016-09-09:
Done.

closed 2016-09-09 by Fourdee

msdos, 2017-04-09:
I had some problems when using the NFS client in Dietpi. When trying to configure, giving my NFS server ip, the following message is shown (it's really fast):

```
mount.nfs4: Protocol not supported
```

After some help from these guys http://unix.stackexchange.com/a/205736 and http://dietpi.com/phpbb/viewtopic.php?f=11&t=791, I edited `/etc/fstab` and changed to 

```
192.168.1.1:/media/ /mnt/nfs_client nfs auto,vers=3,_netdev 0  0
```

And now it's working. Would it be the case to add an option in NFS config "Do not use nfs4" or something like that when configuring nfs? The default now is to use `nfs4`thus giving the error above.

Fourdee, 2017-04-10:
@msdos 

Thanks for the feedback and solution 👍 I'll get this information added in the script.